keirf/flashfloppy #161: [Big issue - no idea where to ask] Gotek as an external Amiga 500 floppy drive

Arawn-Davies, 2018-08-29:
Hi, I'm unsure where to open a new issue, so please forgive me if this is too big or whatever, but I've programmed my Gotek floppy emulator with the FlashFloppy firmware via the DfuSe DFU update tool and formatted my USB stick using FAT32 (not sure if that's a problem) and placed SELECTOR.ADF in the root directory. 

When I connect the Gotek to my A500 using one of these adapters I got off eBay [See Image](https://i.ebayimg.com/images/g/xiYAAOSwXshWrzO4/s-l300.jpg), and it just says "F-F" on the 7 segment display without much activity on the USB.

When I press the left button on the Gotek, nothing happens, however when I press the right button, the 7 segment display clears, and says "0", then "9" and then "28A", and goes back to the previous "F-F" message. As its set to be an external floppy drive, I've got my jumper set to S1 (S0 is for internal?) yet even on S0 I get the same thing. 

For the adapter, I've got the red wire on the floppy cable on the far side of the power cable, because when the red wire is on the near side, it won't get past the screen asking for the workbench disk - the floppy disk works, and is inserted, and I get the same thing on both S0 & S1

This might be an individual issue with my USB, Gotek, or the firmware that I flashed, or how it's configured via jumpers and cable orientation, but I've no clue. If someone does have any idea as to why, or a solution, that'd be much appreciated :)

keirf, 2018-08-29:
Two things.

1. Selector.adf is not the correct selector for FlashFloppy. You want autoboot.hfe and hxcsdfe.cfg which are shipped in the FlashFloppy download zip.

2. If the display stays at F-F when you insert a stick then flashfloppy didn't find a filesystem. Either it does not like the stick, or the way you formatted it. Worst case (unlikely) you are feeding the Gotek 12v instead of 5v and frying every USB stick you insert (in which case they no longer work in your PC either).

closed 2018-08-30 by keirf

rmanicardi, 2019-05-18:
Hi, how did you manage to solve the error -1 issue? I have the same and no clue on what to look at. Thanks!

keirf, 2019-05-19:
On amiga? What file selector version? It's displayed at bottom of screen.

rmanicardi, 2019-05-19:
Was 1.71, I’ve update to v5 and now it’s working fine!

LizardGTX, 2019-06-05:
Hi, i just got my Gotek drive and the FFManager is also v1.71 and an unknown error (-1) pops up as well. @rmanicardi how did you update it to version 5?
I downloaded the .zip "HXCFEUSB_HFE_beta_firmware" and exchanged the AUTOBOOT.HFE and the HXCSDFE.CFG but then the Gotek didn't show up anymore on workbench.

rmanicardi, 2019-06-06:
Hi @LizardGTX I've downloaded it from here:
https://github.com/keirf/HxC_FF_File_Selector/releases/download/v5-FF/HxC_Compat_Mode-v5-FF.zip
and used the AUTOBOOT.HFE and HXCSDFE.CSG provided in that zip

sandfern, 2022-02-11:
Thank you for the quick reply and sorry for my late reply. No, I'm using the v9, from the FF 3.30. The gotek came with the 3.29 firmware and I updated it to 3.30. Both shown this issue. I'll try later the older v5 (I'm not near the Amiga right now) and then I'll post the result here.