keirf/flashfloppy #162: 360k images from double sided 40 tracks disks are not working with host=msx

epolesi, 2018-08-29:
Please add support for {  9, 2, 84, 1, 2, 1, 0, _C(40), _R(300) } images to host msx.
When host = unspecified, it works, but when host = msx, it doesn't.
Thanks!

keirf, 2018-08-29:
The only exception applied to host=msx is 80 track single sided. If msx also supports 40 track double sided then the unfortunate problem is that both image types have the same file size. And file size is what flashfloppy uses to determine disk layout.

Perhaps msx could have layout determined from filesystem structures within the image file?

epolesi, 2018-08-30:
So, this is the case.
Both images (1 side 80 tracks and 2 sides 40 tracks) have the exact same size and, yes, the boot sector has a lot of information about the disk geometry (same as PC). The biggest difference between MSX and PC in terms of floppy disks is that MSX supports fewer formats, but all of them are PC compatible.
The curious thing about this issue is that setting host to "unspecified", all seems to work. This is why I asked to add a new format to the msx host list of supported formats.
Considering the "image size conflict" question, what could be a solution to use msx host with all its image variants?
Thanks

keirf, 2018-08-30:
What is the msx boot sector layout? If flashfloppy can detect a valid boot sector then it can pull the geometry from there.

keirf, 2018-09-03:
Any thoughts? Can you link to docs on the msx boot sector?

keirf, 2018-09-04:
Okay please can you try the attached firmware? Make sure you use `host=msx` in FF.CFG.

[ff_162_1.zip](https://github.com/keirf/FlashFloppy/files/2348234/ff_162_1.zip)

epolesi, 2018-09-05:
Please, hold this issue for some days.
Before you make any other changes, I would like to do some more tests, with other disk drive interface, but I will be able to do that only by Sep, 10. I'll come back then with more information.
Thank you.

keirf, 2018-09-06:
I tested the images and the firmware seemed to do (mostly) the right thing. There was one small bug, which is fixed in the firmware attached below. However, I suspect the problem was in your setup (as you seem to suspect too!).

[ff_162_2.zip](https://github.com/keirf/FlashFloppy/files/2356371/ff_162_2.zip)

epolesi, 2018-09-11:
I confirmed that the disk drive interface I've used only supports 3.5 floppy drives (80-track) and not 5.25 (40-track). Considering this limitation, and using the new firmware (ff_162_2.zip), I could successfully read/write/format/boot with both 80-track images (1 and 2 sides) using that interface.

On the other hand, I took another interface, which I know it supports 3.5 and 5.25 disk drives and could successfully test all 4 disk images (40 and 80 track,s, 1 and 2 sides).

There are other less used disk formats on MSX platform (ie, 8 sector per track disks) that I've not tested, but the boot sector follows the same structure, so I think we are good with them too if you are always looking for disk geometry there when host=msx.

In resume, I think the floppy emulation for host=msx is OK now.
Big thank you! You've been doing a really great job with FlashFloppy!

closed 2018-09-11 by keirf

keirf, 2018-09-11:
I do not wholly rely on the boot sector as I don't like to 100% trust it unless it's guaranteed every MSX disk, whether bootable or not, has a valid parameter block in the boot sector. Hence I only use it to disambiguate 320k and 360k images, with strict checks on values read from the boot sector. But anyway host=msx should now support all variants of 40- vs 80-track, 1- vs 2-sides, and 8- vs 9-spt.

commit 93e1df19 (2018-09-11): img, msx: Auto-detect 80T/SS vs 40T/DS images using boot sector.
Fixes #162