keirf/flashfloppy #163: constant usb activity ?

sandgoose, 2018-09-02:
finally got my china drive after weeks of waiting, flash process went smoothly and its working fine in basic mode (adf files dumped onto the root and selecting between them using the buttons)

but regardless of if the amiga is reading a disk or not the flash stick indicator light shows constant activity.

using v0.9.28a, tested 2 drives both show the activity on their internal lights even when the amiga and goteck access lights aren't on,

drives activity works as expected under windows/linux.

suggests anyone ?

Arawn-Davies, 2018-09-02:
Just managed to get my Gotek working after formatting under FAT16 with a different USB (first two didn't work, third one at least displayed 000) and I get this same thing, although after running FFManager it displayed unknown error (-1)

Are you able to boot into FFManager?

tenuke, 2018-09-02:
Constant usb-activity is how it currently works. This will change when Keir
implements buffer cache.

Arawn: Format the stick to FAT32 although I have no idea if this will fix
your issue. But try it out as it is what you should use anyways.

keirf, 2018-09-03:
If you want an unobtrusive drive that doesn't flash, try a SanDisk cruzer fit, or similar.

Always format the stick fat32. That's all that is tested and quite possibly autoboot.hfe only supports that.

keirf, 2018-09-03:
Actually I think Cruzer Fit does flash. Mine has gone missing so can't confirm, but the Ultra Fit certainly does. Oh well.

A definitely non-flashing stick is the Kingston DT50. It's cheap, too. But not as unobtrusive as a SanDisk Fit.

It will get fixed for all drives, eventually ;) It is likely months away though.

sandgoose, 2018-09-03:
@Arawn-Davies  maybe worth mentioning out of the handful of devices I tested I had 2 that didn't work, one was a really old/slow 4gb mini hard drive I believe is not usb 2, other was a usb floppy drive fat formatted with an adf file on it (lol weird I know but I thought this would be fun to test) I guess maybe try different sticks see if you have any luck?

@keirf thanks! good to know it is intended, I will see if I can dig out a stick without an indicator light, should have a few kicking around somewhere if not like you said they are relatively inexpensive.

closed 2018-09-12 by keirf