keirf/flashfloppy #165: Feature suggestion: Blank floppy images are saved to a copy instead of original file.

ewanuno, 2018-09-06:
currently when saving to floppy images, one must create a new empty blank image, then save to that image, then rename it on another computer.

if blank floppy disks were saved as new copies, this would save a lot of repetitive disk organisation

this would require a way of designating disks as "blank", for this i suggest a magic file name like image_name-FFBLANK_SAVE_AS_COPY.img
and resulting copies renamed
image_name-001-FFBLANK_COPY.

my use case is an ensoniq mirage sampling keyboard. i create a lot of disks and it would save me a load of housekeeping work. i suspect this would be similar for many other users who are creating data.

tenuke, 2018-09-06:
Cant you create as many blanks as you need an name them as you like?
Does that platform really need the blank disk to be named something
particular so its not possible?

ewanuno, 2018-09-07:
The idea is that we would only need one blank disk image which would then create all the fresh images we could even need, ending the need to remember which blank disks had been used and which were still blank.
It's definatly a convenience feature.
I currently have a folder full of blank images which I have to manually rename as I use them as you suggest.