keirf/flashfloppy #167: Single-sided .ST disk images don't work properly

sandord, 2018-09-13:
I have a few single-sided .ST disk images on a USB stick. None of them work though my Gotek running FlashFloppy (latest version). When I use a copy program to read from the disk, it starts throwing read errors from track 40 and upwards. I think that means that FlashFloppy isn't properly mapping the tracks.

keirf, 2018-09-14:
Please try the firmware attached below. It changes the ST-handling code quite a bit so please try with a few different images including the more common double-sided ones.

[ff_167_1.zip](https://github.com/keirf/FlashFloppy/files/2382852/ff_167_1.zip)

sandord, 2018-09-14:
Thanks, first try is a big success (Big Demo 😎). I will try more images and report back here.

sandord, 2018-09-14:
All single sided disks I've tried so far now work, although there were one or two games I had to try a second time because somehow they got stuck while loading.

keirf, 2018-09-15:
Thanks I will roll the fix into the next FF release.

closed 2018-09-15 by keirf