keirf/flashfloppy #168: After some minutes of normal activity the Gotek starts to quickly change the images contained in the USB pendrive

WiZkiD-DRM, 2018-09-15:
The issue occurred when I tried to add a rotary encoder without the 3 pull resistors. I followed the guide without any error, at first everything seemed to work correctly but suddenly the problem occurred while I was browsing the HFE images. If the gotek has just been turned on, everything works fine but after few minutes the images start to scroll by themselves very quickly as if one of the buttons is constantly pressed, after few more minutes the scrolling of the images slows down considerably but the issues is still there

keirf, 2018-09-15:
Does this problem occur with the rotary encoder disconnected?

WiZkiD-DRM, 2018-09-15:
yes, I have already disconnected the encoder and the issues still persist. I've noticed that more time passes with the Gotek turned on less the problem is critical

WiZkiD-DRM, 2018-09-15:
the only mod I've done is the OLED screen connected following your simple tutorial on the relative page :(
but thanks alot for the help

keirf, 2018-09-15:
The problem must be that one of the two original button pins to the microcontroller is getting pulled low after the Gotek has warmed up.

closed 2018-09-15 by keirf