keirf/flashfloppy #169: Image change occasionally results in FF reboot?

dl6lr, 2018-09-23:
Hi,

first of all thank you for the great software. I just tied a Gotek with FF 0.9.29a and OLED to an ancient GOULD K105D logic analyzer (instead of a TEAF FD-55B) and booted the software from it!
The USB stick currently has 3 raw disk images *.IMG with 2x40 tracks, 8 sectors of 512 byte each, resulting in 320K. This is the disk format the GOULD uses for its CP/M-86 like DOS.

What I found irritating is, that when switching the disk image with one of the keys, occasionally FF seems to restart with its Splash screen and then showing the first image. FF.CFG only contains:
 
display-type = oled-128x32-rotate-narrow

And the file IMAGE_A.CFG currently contains a strange combined file name that seems to be the longest file name of all images but starting with a shorter one:

DISZ80.IMG2.2_D6809_1.img

Contents of USB drive:

f:\>dir

 Datenträger in Laufwerk F: ist XMAS
 Volumeseriennummer: 2E17-991E

 Verzeichnis von f:\

22.09.2018  23:41           327.680 K105D_SOS_2.2_D6809_1.img
22.09.2018  23:55           327.680 DIS68000.IMG
22.09.2018  23:55           327.680 DISZ80.IMG
01.05.2017  00:00                25 IMAGE_A.CFG
23.09.2018  11:33                42 FF.CFG
               5 Datei(en),        983.107 Bytes
               0 Verzeichnis(se),  2.056.855.552 Bytes frei

Any hints?

Bernd

keirf, 2018-09-23:
Have you tried another USB drive?

dl6lr, 2018-09-23:
Not yet. I will give it a try.

dl6lr, 2018-09-24:
Tried another USB drive, that seems to work flawlessly (although this is much slower and older). Thanks for your hint!

closed 2018-09-24 by dl6lr