keirf/flashfloppy #171: Bug: Can't switch to LBA-Modus, if no Image-File is in Root-Directory

BeCube1, 2018-10-01:
The Display Shows ERROR34.
When i place one Image in the Root-Dir, the switch was immediately successful

keirf, 2018-10-01:
I can't promise to fix this one. The firmware is strongly bimodal at the moment and will only properly respond to the host when an image is mounted.

Also the clear error has been useful in diagnosing new users' confusion in how the device should be used. 

What behaviour do you precisely expect when an empty stick is inserted? Do you expect the stick to at least be FAT formatted?

keirf, 2018-10-01:
Another thing - - assuming you have images in subfolders - - is that FF does not support direct navigation into subfolders when using the original 888 display.

keirf, 2018-10-01:
Okay I understand. This is a bit of work... I will have to think about it. The workaround of a dummy image to bootstrap the Gotek is perhaps not so bad?

BeCube1, 2018-10-01:
thanks, i understand you :-)
Workaround: Hmm.... things goes a little bit more complicated, if you have only a 8Bit 4MHZ CPU and a 16KB ROM ;-)

EDIT too: 
The alternative usage model is: Direct FAT-Access with LBA-Modus. And start extracted Games directly.
The normal usage model is: 
Browse through the FAT16/32-Filesystem to an HFE-File in LBA-Modus, CD into it (04) and run the Game from the Image (in Floppy-Mode).  (The Gotek and HXC in LBA-Modus are very slow)
The ??? usage model is: 
Assign some Images to the Slots and select it with the hardware-keys on the HXC/Gotek.

keirf, 2018-10-01:
Re CMD-04 see my suggestion on that ticket

closed 2018-10-06 by keirf