keirf/flashfloppy #172: TI99 SSSD & DSSD image corruption

klrw111, 2018-10-02:
With the latest version and CFG file set to ti99,  the .dsk images read and execute properly.  When writing a few files everything appears to be ok, but copying a larger number of files approaching 360 sectors of disk space the image gets corrupted during the write.  Have also tried with .hfe and get the same result.

keirf, 2018-10-02:
Is this a regression (have you seen it work on an older firmware version)?

klrw111, 2018-10-02:
It is not a regression.  I tried on the earlier version posted on ATARIAGE and got the same issue.  Originally thought is was a Gotek hardware issue (since OLED shorted), but finally ordered a new Gotek and got the same issue with it.  I can post disk images if it helps.  The directory file seems to get trashed with outside data.  You can read part that is "emu.."

keirf, 2018-10-03:
Okay here is a firmware which I think fixes access to the second side of double-sided DSK images. But that wouldn't explain problems with HFE images, nor with single-sided (SSSD) images. Do other TI99 image types work? (ie. double- and high-density) Or are you unable to test those?

[ff_172_1.zip](https://github.com/keirf/FlashFloppy/files/2440544/ff_172_1.zip)

It is pretty unusual for things to go wrong with IMG/DSK files. I may need serial debugging (USB-serial dongle connected to a host PC while the Gotek is in use in a TI99) or some sample disk images may help but unsure.

If the Gotek is replaced by a real drive does that work fine?

klrw111, 2018-10-03:
I have a TI disk controller that is only capable of SSSD and DSSD.  The first trouble shooting step was to replace the Gotek with a real drive in the chain.  The real drive worked perfectly.  I will use a SSSD, DSSD, and a HFE image and test the revised firmware later today and let you know.  Thanks

keirf, 2018-10-03:
Ok let's wait and see. I'm suspicious I may need serial debugging for this one... :(

klrw111, 2018-10-03:
SSSD and HFE formats seem to be working, but DSDD fails consistently when writing files.  I tried just writing group of files with less than 360 sectors and still had issues.

keirf, 2018-10-04:
Please send me a dssd dsk image and the equivalent HFE image if you can

klrw111, 2018-10-04:
Here are the images. SSSD.dsk has files written by FF up to 360 sector.  DSSD.dsk is the corrupted image when sectors beyond 360 were written to a blank DSSD 720 sector TI image. HFEDSSD.hfe is the 720 sector disk written by FF.  The DSKA0001.HFE is a volume created and readable by TI tools for the PC for creating and writing TI disk images.  The SSSD.dsk image could be read by the PC software, but the DSSD.dsk and HFEDSSD.hfe could not be read by the sofware, although HFEDSD.hfe could be read and accessed as a disk via the Gotek.
[DSKA0001.HFE.zip](https://github.com/keirf/FlashFloppy/files/2447310/DSKA0001.HFE.zip)
[DSSD.dsk.zip](https://github.com/keirf/FlashFloppy/files/2447311/DSSD.dsk.zip)
[FF.CFG.zip](https://github.com/keirf/FlashFloppy/files/2447312/FF.CFG.zip)
[HFEDSSD.hfe.zip](https://github.com/keirf/FlashFloppy/files/2447313/HFEDSSD.hfe.zip)
[SSSD.dsk.zip](https://github.com/keirf/FlashFloppy/files/2447315/SSSD.dsk.zip)

keirf, 2018-10-05:
Okay I had a quick look and realised the side-2 track offsets in DSK files was still messed up. I have fixed that, please try the following firmware:

[ff_172_2.zip](https://github.com/keirf/FlashFloppy/files/2450046/ff_172_2.zip)

If it works I'm also interested whether performance is comparable with a real drive. Thanks!

klrw111, 2018-10-08:
I had a family emergency this weekend and was unable to much.  Hope to get to do more this week.  Thanks

keirf, 2018-10-09:
Fixed in v0.9.30a. Please open new tickets if further testing shows other problems or performance issues compared with real floppy disks.

closed 2018-10-09 by keirf