keirf/flashfloppy #173: HD select signal should be from controller -> disk, not from disk -> controller

Invincing, 2018-10-17:
I believe FlashFloppy configures pin 2 as an output from the drive if it's configured as HD select signal.

HD select should be input to the drive from the floppy controller, not output from the drive. This information is available e.g. in:
http://bitsavers.informatik.uni-stuttgart.de/pdf/mitsubishi/floppy/MF355/UGD-0489A_MF355B_Specifications_Sep86.pdf (page 22) and http://www.hermannseib.com/documents/floppy.pdf .

The second document describes this signal (DENSEL) in the following way:
_DENSEL selects double density or high density and at the same time if reduced write current should be used as those are coupled. For this reason, this pin is sometimes also called low current. [Level assignment is still unknown. It is said that the assignment is inverted between 51⁄4" and 31⁄2" floppies.] Modern ‘‘intelligent’’ 31⁄2" HD drives do not depend on this signal but recognize the data rate from the floppy disk._

If both the disk drive and the controller are driving the same signal, this can result in very strange electrical issues in the system. I've connected a Gotek with FlashFloppy to Akai S1100 sampler (using Akai S950 settings, as they should be compatible). The disk behaviour was very unstable with regular read errors, but occasionally disk images were working. Changing the pin02 state to "nc" (Unused/No Connection) resolved all the stability issues, as Gotek was not driving the signal anymore. (Since FlashFloppy can detect whether to use HD or DD mode based on the disk image file, it doesn't really need the HD select signal.)

keirf, 2018-10-18:
Hi, FF only outputs density on pin 2 when `interface =ibmpc-hdout`. The default action for pin 2 on the much more common `interface =ibmpc` is already `nc`.

There certainly systems that do require the output on pin 2, such as dome Ensoniq synths, and they are the only ones I currently explicitly recommend that interface setting for.

In light of this, is there any action you consider necessary for this ticket?

Invincing, 2018-10-18:
I think that what this only needs is a documentation change and some naming changes. The `dens` setting is not documented well enough (I had to read the source code to find out what it actually does, i.e. that it's referred internally as `outp_hden`). Based on the documents linked above and the various disk controller data sheets I originally assumed that the `dens` is the same signal as DENSEL in the documents. It's not - it's going to the opposite direction.

In addition to making the data direction more obvious in the documentation, it would probably make sense to improve the naming as well. As some pins can serve both as inputs and outputs, and this may vary depending on the system, maybe adding `in` or `out` as part of the configuration parameter would reduce the chances of confusion.

I understand that pin 2 is in some cases used as output from the floppy disk for HD density signal. It seems that in many disk drives the input/output HD select mode can be [chosen with jumpers](http://matthieu.benoit.free.fr/pdf/skettes.pdf). Based on [Akai S950 schematics](http://floppy.shugart.free.fr/materiels/akai/s950/manuel/akai_s950_schematics.pdf) this seems to be the case with that sampler as well - pin 2 from floppy interface named "density" is routed to one input of 74HC257P MUX IC. So the direction is definitely from disk to controller, and `akai-s950` configuration is probably correct in this respect. Maybe a separate configuration `akai-s1100` with values `pin02 = nc`and `pin34 = rdy` could be added? (Akai S1000 is probably compatible with akai-s950, at least it uses the same disk controller [NEC uPD72066C](https://datasheet.datasheetarchive.com/originals/distributors/Datasheets-112/DSAP0046987.pdf) whereas S1100 uses [NEC uPD72068GF](https://pdf.datasheet.live/52eb8039/necel.com/UPD72068GF-3B9.pdf)).

Based on the floppy controller datasheets I've read (e.g. [NEC uPD72068GF](https://pdf.datasheet.live/52eb8039/necel.com/UPD72068GF-3B9.pdf), [Intel 82077AA](http://www.buchty.net/casio/files/82077.pdf), [National PC8477B](http://ohlandl.ipv7.net/floppy/DS011332.pdf)), the DENSEL signal actually tells the drive to use higher data rate for HD floppies. Would FlashFloppy be able to gain faster transfer rates if it took advantage of this signal? Or does it already do this by detecting that the current image is a HD floppy?

Simone2018, 2018-10-18:
I think that in the IBM-PC interface the HD or SD is choosen from image dimension... (720kB or 1440kB). It would be useful to override this with an option if possible, like:
Density=High
or better:
Rate=300 [kb/s]

thank you

keirf, 2018-10-25:
I have added further documentation to the FF.CFG page on wiki, and to the example FF.CFG shipped with FlashFloppy. Please take a look, I think it clarifies that pins 2 and 34 are always outputs (drive-to-host), or not connected.

closed 2018-10-29 by keirf

Invincing, 2018-10-31:
Also, please add Akai S1100 specific config to
https://github.com/keirf/FlashFloppy/wiki/Host-Platforms#akai-synthesisers

pin02 = nc
pin34 = rdy

You could also add specific language to FF.CFG page mentioning that on some systems pins 2 and 34 may be inputs to the floppy drive, and in that case they should be set to "nc".