keirf/flashfloppy #175: Amiga File Selector will not regognise FF Gotek in external enclosure

glvas, 2018-10-30:
Great job on the firmware.

Works fine when gotek is installed internally. However, in my case, when installed in an external floppy enclosure as DF1:, AUTOBOOT.HFE boots fine, but the drive is not regognised by the File Selector. File Selector shows "ERROR: HxC Drive not detected!"

Drive was fine as DF1: when previously flashed with the Cortex firmware. Also File Selector from HxC firmware will regognise the drive just fine, but will, naturally, give an invalid HXCSDFE.CFG file error later.

Thanks.

TzOk83, 2018-10-30:
I'm afraid the HxC File Selector works only with DF0, yet on Amiga there is an easy mod to make external FDD DF0, and internal DF1. Anyways I strongly recommend you installing an OLED display + rotary encoder and forget about AUTOBOOT.

keirf, 2018-10-30:
Your df0 is probably not an original amiga drive and has a ready signal hack I guess. Have you tried latest autoboot.hfe shipped in the FlashFloppy v0.10 release?

keirf, 2018-11-02:
Any update? I will close this ticket soon.

glvas, 2018-11-02:
No change, File Selector does not work when Gotek is configured as df1:. No difference with latest firmware and autoboot.hfe.

I am using an accelerated, original Commodore Amiga 1200 with SD HD, the original Chinon internal drive, and the external enclosure where the Gotek is installed, is from a Cumana CAX 354 floppy. I am using the drive as an install media jukebox and do not care to install it internally or hack a drive switch, it works fine for me as a df1:.

I opted to use it in native mode with just the 3-digit segment display and the images dumped in the top folder of the USB drive, so I am fine as is. Maybe some else minds.

Thanks.

glvas, 2018-11-03:
Sorry for the delay. I downloaded systest. Nice little program.

When doing the read test on df1:, where the Gotek is installed, it reports "No READY signal - PC or ESCOM drive?". The drive is recognised fine and shows an ID of ffffffff. Please note, as I said, that Cortex firmware File Selector worked fine with this configuration, and also an HxC firmware Gotek drive (since mated to an Amstrad CPC) was also recognised fine in this same configuration by that File Selector.

Thanks.

keirf, 2018-11-04:
The FlashFloppy version of the file selector, compared with HxC original, detects a Gotek by sampling the READY signal immediately after turning on the MOTOR signal. Real drives are not ready that fast, hence this distinguishes Gotek from real drive. HxC by contrast tries stepping the heads too fast, which is another thing real drives can't do.

Your problem is that the READY signal is not propagated from the Gotek drive. Clearly READY is generated okay in the enclosure's PCB and transmitted to the Amiga, since READY (when MOTOR is off) is used to identify external drives (that's the ID ffffffff in systest).

Possible problems:
1. Gotek pin 34 is not working;
2. Enclosure PCB is not passing through pin 34 properly;
3. The ribbon between Gotek and Enclosure PCB is damaged.

Option (3) is by far the most likely. You can test this with a multimeter, and the fix would be to replace that short ribbon.

Of course if you don't care about using the file selector, and you're really only using DF1 for AmigaDOS stuff, you could just leave it. AmigaDOS does not care about the READY signal except for drive identification (which you have tested is working).

closed 2018-11-07 by keirf

glvas, 2018-11-08:
Just an update. After changing the connectors on the short ribbon cable connecting the floppy to the enclosure, the gotek is now recognised just fine as a df1:. So you were right, it was your 3rd option, the ribbon cable.

Many thanks.