keirf/flashfloppy #177: 720Kb emulation for IBM-PC

zeux69, 2018-11-01:
Sorry, but I couldn't understand.  Can I emulate IBM-PC 720Kb floppy on this firmware? I found in FF.CFG drive emulation interface - ibmpc only. Is it enough to set this mode? My equipment support IBM-PC 720Kb floppy only. Thanks.

keirf, 2018-11-01:
FlashFloppy supports 720k IMG files out of the box, no configuration. If your system also has an IBM compatible ribbon interface (rather than Shugart) then you need a jumper on JC.

closed 2018-11-02 by keirf

zeux69, 2018-11-07:
OK, thanks a lot. It works! One more question. 
Can upgraded SFRM1M44 work on 720Kb drive?

keirf, 2018-11-07:
Yes the specific model of Gotek you get doesn't matter, they're all the same hardware just differ in original firmware (which you are overwriting with FlashFloppy).