keirf/flashfloppy #178: DSK with strange protection doesn't work

Turk182es, 2018-11-01:
Hi guys,

The game Le Manoir de Mortevielle doesn't load. It ssem to have a strange protection that the USB Floppy cannot resolve.
Protection : KBI, 19sectors, KBI MASTER
DSK file : http://www.cpc-power.com/index.php?page=detail&onglet=dumps&num=1349

is there any solution ?

TomDDG, 2018-11-01:
Is this due to the GAPS protection? https://github.com/keirf/FlashFloppy/wiki/Host-Platforms#amstrad-cpc

Turk182es, 2018-11-01:
I don't think so because a lot of GAPS DSK works perfectly

TomDDG, 2018-11-01:
Have you tried running that script on the disk?

Turk182es, 2018-11-01:
Which Script ?

TomDDG, 2018-11-01:
The one in the link I provided

keirf, 2018-11-02:
This title is much more likely to work once run through the fixup script mentioned in TomDDG's link above. How do I know?
1. The script tells you how many sectors it fixed up!
2. There is also a edsk-dump script in scripts/edsk.py which flags interesting info about each sector. This disk raises a number of "Pre-Sync???" warnings, which indicates it needs fixing up.

closed 2018-11-02 by keirf