keirf/flashfloppy #179: Single density version of "SD HxC Floppy Emulator Direct Access mode"

richard-broadhurst, 2018-11-04:
Please add a single density version of "SD HxC Floppy Emulator Direct Access mode".
I think you suggested using track 254 and leaving the rest the same.

keirf, 2018-11-04:
Yes sorry I had totally forgotten about this. I will work on it and post progress (and firmware to test) here on this ticket.

keirf, 2018-11-04:
Question: I assume a 512-byte sector size would be okay? 256-byte is more normal for FM formats, but I don't think 512-byte is precluded. It makes things simpler if the logical sector format and overall protocol is identical to MFM (except for essential track-framing differences between FM and MFM). Only difference will be defaults to 5*512-bytes sectors (FM) vs 9*512-byte sectors (MFM).

Simone2018, 2018-11-05:
I think Direct Access Mode is already in "single density" if you intend MFM DD (250 kBit/s), on track 255 or I'm wrong?

keirf, 2018-11-06:
Okay here is a firmware for you to test. The source is in Github branch `fm_da`.

[ff_179_1.zip](https://github.com/keirf/FlashFloppy/files/2554094/ff_179_1.zip)

richard-broadhurst, 2018-11-06:
I think the beeb OS only supports 256 byte sectors without driving all the different controllers directly, but I see how far I get.

keirf, 2018-11-07:
If you need changes just let me know. It's mostly easy to switch to 256b sectors. What were you planning to use the Direct Access mode for by the way? I don't think you're writing an image selector, right?

richard-broadhurst, 2018-11-07:
The primary use is to select disk images.

richard-broadhurst, 2018-11-07:
Well, I can read the status, which gives the "HxCFEDA" string and "FF-v0.1".
I don't know if writing the control data is working, doing a NOP seems to work, but it might just be being ignored.
The command is supposed to have a checksum, but I can't find any details of it.
I had assumed that command 4 was all I needed, but it doesn't seem to do anything, including giving an error.
I don't know if this is because my checksum (=0) is invalid, or because 512 byte sector writes don't work (on the BBC, but they should), or maybe it requires the config where files have a sequential number or who knows!
The "Menu" disc would be built on the PC, so I don't need to be able to scan the stick.

keirf, 2018-11-07:
Command 4 isn't implemented! Why not try a set lba command? You could check it worked by seeing if contents of sector 2 changes (that's part of the window into lba space)?

Ps I can implement command 4. Or a variant with semantics of our choice.

richard-broadhurst, 2018-11-07:
seems to read different sectors, got to go now, but I'll try setting a new LBA base tomorrow.

keirf, 2018-11-09:
Okay the single-density FM access is now in master branch.

I will implement CMD_SELECT_IMAGE (cmd 4) and notify you here when that's done. That simply takes a 16-bit slot number/index. Very simple.

richard-broadhurst, 2018-11-25:
Will this work with indexed mode?
DSKA9999.ssd

keirf, 2018-11-26:
DSKA9999.ssd won't work because only slots 000-999 are supported. However that means that DSKA0000.ssd up to DSKA0999.ssd will work.

keirf, 2018-11-26:
I will get on with CMD_SELECT_IMAGE soon but I wanted to rework some FF internals at the same time to give more flexibility to direct-access mode.

richard-broadhurst, 2018-11-26:
No problem, I think I am ready at my end, 2897 games/apps over 792 single sided disk (ssd) images indexed in 25KB with a menu system, just need to test with CMD_SELECT_IMAGE.
I am grateful for all the effort that you put in for us.

keirf, 2018-11-27:
CMD_SELECT_IMAGE is implemented in the latest FF version (v0.12).

What mode are you running FF in for your tool, is it indexed mode (DSKA0000 etc)? I have supported CMD_SELECT_IMAGE for all navigation modes, but I might restrict it to Indexed and Autoboot-Selector modes as it doesn't make a great deal of sense for direct-navigation mode.

richard-broadhurst, 2018-11-28:
Thanks, Indexed mode as it made the most sense, but I was thinking of supporting the other modes so as not to rename the files although this would be very fragile.

keirf, 2018-11-28:
Perhaps for Direct-Navigation mode it will make more sense to provide a different command, eg allow you to specify a full filename (or even full path, perhaps via a sequence of chdir() type invocations). Could provide a FS-type interface... opendir, getnext, chdir, openfile, etc

keirf, 2018-11-28:
Assuming you don't want to go straight at the block layer and do FAT munging yourself. I'm thinking... not ;)

keirf, 2018-11-28:
Anyway let me know how CMD_SELECT_IMAGE goes for you at the mo. I didn't really actually *test* it...

richard-broadhurst, 2018-11-28:
It's not going well, I can't update to 0.12 instead of UPD, I get EJE followed by rdo.
The same thing happens on several sticks, ones that were fine for going to 0.11.

keirf, 2018-11-28:
Noone else has reported this so it's probably you, and doubly so if you did not update the bootloader to 0.11 (if you're not sure, then you didn't). Are you definitely holding both buttons at power on?

richard-broadhurst, 2018-11-28:
I didn't update the bootloader to 0.11, I'll go and RTM!

richard-broadhurst, 2018-11-28:
Other GOTEK updated fine, so I can continue with that, although the first one is custom fitted to my machine!

keirf, 2018-11-28:
It makes no sense really as obvs the buttons work. Perhaps you will need to reprogram that Gotek the longwinded way (serial or DFU)

keirf, 2018-11-28:
Yes sounds weird. I did test direct access with the amiga selector before releasing 0.12 so it should all basically still work even using FM SD.

richard-broadhurst, 2018-11-29:
I have reflashed both GOTEKs using serial USB to v0.12
The disc selection is working in my test app.
One minor thing is that the display (7seg) doesn't update until the new disc is accessed.
This doesn't matter, I mention it so that others don't think the feature isn't working.

kgl2001, 2018-11-30:
I'm not able to get this working correctly on my BBC, but I suspect I'm doing something wrong. I am able to read the status / control sector, but I'm not able to select an image using the CMD_SELECT_IMAGE command. More details here:

https://stardot.org.uk/forums/viewtopic.php?f=16&t=13718&start=150#p221860

Any thoughts on where I might be going wrong?

keirf, 2018-11-30:
As someone else pointed out, you should be writing a 512-byte sector. Because you write a 256-byte sector the CRC will be wrong and FlashFloppy rejects your write.

kgl2001, 2018-11-30:
Yes. That's fixed it. Thank you.

richard-broadhurst, 2018-12-01:
All working, but I might have to use native mode and tell FF that I have an OLED display to get past the 511 files in the root folder issue or find a better way to pack 4000 apps on to 511 discs instead of 800+.

closed 2018-12-01 by richard-broadhurst

keirf, 2018-12-01:
What's the 511 images in root folder issue?

keirf, 2018-12-01:
Bear in mind I may well disable CMD_SELECT_IMAGE for native mode.

kgl2001, 2018-12-01:
I actually quite like using it in native mode. I like the fact that I can change the disk from the host, and still see a meaningful filename in on the OLED screen (albeit, after accessing the disk). What's the issue with keeping CMD_SELECT_IMAGE active in native mode?

keirf, 2018-12-01:
Numbered slots don't make much sense in native mode. Would be better to allow you to read directory entries and specify image to load by file name? I suppose could keep cmd 4 as well...

keirf, 2018-12-01:
Oh yes, embarrassingly I forgot about that hard limit :)

Well, perhaps the better fix for you would be if I let you put the indexed images in a sub-folder.
For example, `nav-mode=indexed-sub` in FF.CFG and place DSKA*.* in subfolder IMAGES?

keirf, 2018-12-02:
Or I could implement a CMD_CHDIR so you can change the image directory yourself. This could be first of a suite of FS-related commands.

richard-broadhurst, 2018-12-02:
I have everything working with ~400 double sided images, so have some breathing space that should last another 37 years.
The CMD_CHDIR would seem like the best option.
Would it make sense to use image DSKA0nnn mode if available and if not, load the nnn'th image?
I'm not sure how this would interact with index/non mode.

keirf, 2018-12-07:
FYI Introducing FM support actually broke sector writes for MFM *and* FM. I'm surprised you didn't see corruption problems. Anyway you should upgrade to (and test) fw v0.14.
0.12 and 0.13 are bad.

sonicthewedge, 2018-12-07:
For the Amiga?

-----

Dr. Ed Young 
M.Math. M.Sc. Ph.D. FACS

keirf, 2018-12-07:
@sonicthewedge No, the Amiga selector uses a non-standard protocol which explicitly identifies the sector it's writing. So should have been fine. Nonetheless it is worth testing again with v0.14.

cfarrar, 2018-12-08:
Sometimes shitaki mushroom happens.

Sent via the Samsung Galaxy Note® 3, an AT&T 4G LTE smartphone

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/keirf/FlashFloppy/issues/179#issuecomment-445216703",
"url": "https://github.com/keirf/FlashFloppy/issues/179#issuecomment-445216703",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [keirf/FlashFloppy] Single density version of \"SD HxC Floppy Emulator Direct Access mode\" (#179)",
"sections": [
{
"text": "",
"activityTitle": "**Keir Fraser**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@keirf",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"keirf/FlashFloppy\",\n\"issueId\": 179,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/keirf/FlashFloppy/issues/179#issuecomment-445216703"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 406125136\n}"
}
],
"themeColor": "26292E"
}
]

sonicthewedge, 2018-12-08:
@cfarrar absolutely.

FW0.14 seems to have done the trick for the USB stick.  I am still working on the Pi Zero solution - having a fight with HDMI now for some reason.

An aside, directory names with "." in them do not work in the FM.  When you try to navigate to them, you are returned to the directory above.  Odd.  Did not happen before.  I had:

workbench/v1.3 and had to change to workbench/v1_2

keirf, 2018-12-08:
This problem with dots in names is within the amiga autoboot selector is it?

sonicthewedge, 2018-12-08:
The File Manager when you navigate to assign an image to a slot.

keirf, 2018-12-08:
I'll have a look but I'm sure the behaviour can't have changed. I don't make significant changes to the file selector.

sonicthewedge, 2018-12-08:
Could well be me - I have had all manner of issues as you know!

Thank you for your help - much appreciated.

sonicthewedge, 2018-12-10:
So, rebuilt USB drive with Pi Zero.  Copied Amiga ADF images and FW FF-v0.14 and FlashFloppy/HxC Selector v1.74-ff freshly downloaded from Github with the example FF.CFG:

"ERROR: LBA Change Test Failed ! Write Issue ?"

sonicthewedge, 2018-12-10:
Got it to work with FW FF-v0.9.6a and HxC File Selector v1.70-ff.  Not the foggiest idea why or how.

sonicthewedge, 2018-12-10:
Seems the ERROR has gone but the File Selector does not write to the drive.  Selecting "Save" attempts a write, the LCD displays "F-F" briefly then nothing is saved after rebooting.

keirf, 2018-12-10:
This is with your Pi Zero as USB stick?

If the LCD displays F-F then it has lost connectivity to the USB stick -- it thinks it has been pulled out.

sonicthewedge, 2018-12-10:
That is correct. I have had the Zero USB Gadget running successfully previously as I have mentioned. It works very effectively for remotely adding image files etc. to my Amiga without having to physically swap USB devices.  I pretty much followed this: https://www.raspberrypi.org/magpi/pi-zero-w-smart-usb-flash-drive/

I neglected the watcher configuration as the Amiga is almost always hard-reset and I wanted to avoid thrashing the SD card.

I spoke to JF De Nero about this too and they had embarked on a similar project: http://hxc2001.free.fr/floppy_drive_emulator/

It is pretty useful (when it worked!). I would be happy to share my approach with you and the community if I can resolve this issue.  There are some physical modifications required to the USB cable to remove the power supply.

keirf, 2018-12-10:
You'd probably find it useful to make a debug build of FlashFloppy and find out in the serial log what happens when the USB disconnects. You may also need to add extra logging to the USB stack.

sonicthewedge, 2018-12-10:
Absolutely - you suggested this once before. I am in the process of working out how to do this.  Any clues?

Thanks again for your help.

sonicthewedge, 2018-12-10:
A consideration when creating the filesystem with 42GB of space:

sudo dd bs=1M if=/dev/zero of=/piusb.bin count=43008
sudo mkdosfs /piusb.bin -F 32 -I

The file system Samba share shows as NTFS on Windoze.

Perhaps there is a problem with drives over 32GB?

Reference: 
https://manpages.debian.org/wheezy/dosfstools/mkdosfs.8.en.html

sonicthewedge, 2018-12-11:
So, frustratingly, the following setting needed to be reordered and appended on the Pi:

sudo modprobe g_mass_storage file=/piusb.bin **removable=1 ro=0 stall=0**

I have no idea when or why this was required to change but was the cause of the write problems.

In other news, I got FlashFloppy firmware to debug build but I encountered a problem building HxC_FF_File_Selector:

m68k-amigaos-strip -s FFManager
make bootblock.bin
make[1]: Entering directory '/home/pi/dev/HxC_FF_File_Selector/amiga'
m68k-amigaos-objcopy -O binary bootblock.o bootblock.bin
m68k-amigaos-objcopy:bootblock.bin: invalid bfd target
Makefile:29: recipe for target 'bootblock.bin' failed
make[1]: *** [bootblock.bin] Error 1
make[1]: Leaving directory '/home/pi/dev/HxC_FF_File_Selector/amiga'
Makefile:18: recipe for target 'AUTOBOOT.HFE' failed
make: *** [AUTOBOOT.HFE] Error 2

keirf, 2018-12-13:
I will try upgrading to latest tool chain and see if I can repro the build failure.

sonicthewedge, 2018-12-24:
Did you have any luck with this?

Are you suggesting that the amiga-gcc version is too recent?  That its
dependencies are too recent too?  Do you know which versions work?

Warm regards.

-- 

*Dr Ed Young, FACS*

Executive Director // Young Consulting

youngconsulting.com.au

*p:* +61 (0) 409 414 294 // *e:* ed@youngconsulting.com.au // *s: *
andrewedwardyoung // *l: *au.linkedin.com/in/younged

*GiveWell* > givewell.org > Real Change For Your Dollar

Melbourne // AUSTRALIA

ABN: 98 121 918 969

Pronouns: xe, xem, xeir, xeirs and xemself (xemselves)

 The information contained in this email message may be confidential. If
you are not the intended recipient, any use of, interference with,
disclosure or copying of this material is unauthorised and prohibited. If
you have received this message in error, please notify me by reply email
and then delete the message.

keirf, 2018-12-24:
I didn't have time, and then I forgot. Could you please open a new ticket for this amiga-gcc build issue?

keirf, 2019-01-22:
FYI I just implemented CMD_SELECT_NAME which allows to select a new image by name.

[18 more replies]