keirf/flashfloppy #180: Provide API over serial to control basic functions

z0m8ied0g, 2018-11-05:
Create a binary API to perform basic functions via serial.

The initial API function required are...

- Mount a disk image by number.
Also might be useful to have another API command to do it by filename.

- Check if a valid USB stick is present.
Simple check if the USB stick has been inserted and is readable.

- Fetch names of disk images in the currently selected directory on the USB stick.
Return the filenames of disk images in currently selected directory. May be useful to have pagination to return a set no of filenames per request.

- Change directory on USB.
Select the current directory.

- Create empty disk image.
Create an empty disk image. Will have to select the disk type when creating (DD / HD etc) and filename to use.

- Display message on screen.
Would be useful to be able to display a custom message on the OLED screen and disable the normal ones.

See facebook discussion for more details...

https://www.facebook.com/groups/flashfloppy/1049673525192880/?comment_id=1051745661652333&reply_comment_id=1059585644201668&notif_id=1541335194232414&notif_t=group_comment_reply

penfold42, 2019-04-23:
Is anyone actively working on this ?

I have a need for an API to control this so will likely do something over the next few weeks

penfold42, 2019-09-02:
Has anyone been working on this ?

I have mostly finished the controlling side and need to start working on the FF side.

At a minimum I’ll start implementing button/rotary controls over serial and firmware and api version commands.

keirf, 2019-09-03:
Probably best make a start and point us at a branch to take a look at. Which uart will this connect to? One possible issue is the main uart dma channels are already used elsewhere.

keirf, 2019-09-03:
Another question: does serial interface respond only while in navigation mode, or respond always but dump into navigation mode if not there already, or is some subset expected always to respond even during drive emulation mode?

z0m8ied0g, 2019-09-03:
What happens currently when you press a key on the front. I take it it waits for the drive emulation to get to some point before navigating on the menu or ejecting the disk?

z0m8ied0g, 2019-09-03:
You could probably get away with only allowing commands while a disk image isn't loaded then having an eject command (or just send anything) that ejects the current image and drops back to navigation mode. Then you'd only have to service the drive emulation to set the disk change flag while responding to serial commands.

keirf, 2019-10-09:
It would be nice to be able to resync to the command stream if bytes are lost or corrupted. Eg. send bytes x=0x00-0x7f direct, send x=0x80-0xff as { 0x80, x-0x80 }. Then we have 0x81-0xff for out-of-band signalling (eg. command/response start/end) allowing easier error recovery. Just putting this down while I think about it.

JanHolbo, 2020-02-05:
RE #225 How hard/easy would it be to port the FlashFloppy code to an ESP32? Pros could be control by BT or WiFi, update storage by WiFi, enabling the FlashFloppy hardware to be buried deep internally in the retro machine as access to storage and control buttons no longer is needed.

keirf, 2020-02-05:
Interesting question. I suppose the most obvious thing that seems to be missing is STM32's super nice timer peripherals with very flexible I/O. They are very critical in FlashFloppy for reading/writing floppy data.

peter-mount, 2021-03-08:
Has there been any progress on this?

One thing I do with my GoTek's is to attach a Raspberry PI 0W to it in Gadget mode instead of a flash drive. This is useful for me as I can then copy new images (usually during software development) to the PI over WiFi and it's then available without having to unplug anything avoiding wear & tear on multiple usb sockets etc as well as making development a lot faster.

That said one thing I have been thinking was if there was a way of controlling the GoTek from the attached PI, so I could do the disk selection/changing from the command line as well as possibly list the available disks (as seen by the gotek) from there - disk images are already visible as they are in a FAT partition on the PI.

As the PI has a uart available to it so it makes sense that if a uart connection was available then this would be pretty simple to implement on the PI.

I'm commenting on this now as I've been following #445 but from what I'm reading right now that seems to be duplicating what's in this issue.

keirf, 2021-03-09:
@peter-mount Sidestepping the serial API non-progress slightly, one option for changing disk would be:
1. Eject
2. Rewrite `IMAGE_A.CFG` on the FAT partition (or `INIT_A.CFG`, if `image-on-startup=init` in FF.CFG on the FAT partition)
3. Mount

Not as quick as a serial API perhaps, but pretty easy to knock up in Pi-land, and can then see if it's fast enough for your needs?

Diepzeevogel, 2023-04-07:
Have there been any updates on this? I would love to be able to interface with flashfloppy from another device to push new floppies and select them automatically.
I was also thinking of using an ESP32 for this in one of these old Yamaha Disklavier machines to make it into a sort of jukebox.

peter-mount, 2023-04-08:
I don't think there's been any movement since I asked 2 years ago. That said I have been using the method Kierf suggested above since then with no issues.

keirf, 2023-04-08:
Is it interesting to have the serial interface fully accessible while a disk is mounted, or can entering serial command mode (at least temporarily) eject the mounted disk.

z0m8ied0g, 2023-05-11:
I'd be happy with that, if the disk doesn't respond until I'm out of serial mode then that's fine. 
Initially just being able to change the disk image would be great.