keirf/flashfloppy #181: Sam Coupe Error 'Missing Disk'

weirdocollector, 2018-11-06:
Hi!

I've connected a Flash Floppy Gotek to my Sam Coupe using the Quazar Floppy Interface [https://www.samcoupe.com/hardfdc.htm](url), which is based on a 1772 controller, and it's impossible to read any .dsk image due to the 'Missing Disk' error, as the disk insertion appears to not be recognized by the Sam.

I've used a physical floppy with the same Quazar interface, and it works well.

Any idea on why this happens?

Thanks!

keirf, 2018-11-06:
Needs a jumper at JC (rear of Gotek) possibly?

weirdocollector, 2018-11-06:
I've tried putting a jumper on JC with the same result: Missing disk (Error 55)

keirf, 2018-11-06:
It will almost certainly be a jumpering issue all the same. Some combination of jumper at S0 vs S1, and jumper at JC or not.

Are you also confident you have a good disk image mounted in the Gotek?

weirdocollector, 2018-11-06:
Yes, the image is OK. I've tried it with a SAM emulator and another SD card solution for the same micro.

weirdocollector, 2018-11-06:
Yay! Jumper on S1 (no JC) did the trick! Case closed.

closed 2018-11-06 by weirdocollector