keirf/flashfloppy #182: H8D image format

robdaemon, 2018-11-07:
I'd like to use a Gotek with FlashFloppy in my Heathkit H88/H89. It looks like most of the disk images out there are .H8D files, built by a tool called ImageDisk / IMD.

The Heathkit can have both hard sectored and soft sectored floppy drives, single density and double density.

For a reference as to the disk images I've found, see: http://sebhc.lesbird.com/software.html

I'm guessing it's not easy to support based on the previous TD0/IMD issue, but if it's necessary to convert the images I can do that too.

Thanks for this awesome firmware!

keirf, 2018-11-18:
Have you tried renaming to .IMD and then converting to .HFE using the HxC software (Windows, Mac) or my own Disk-Utilities (Linux command line)?
This assuming the format is actually IMD, which seems likely if the images were created using ImageDisk.

keirf, 2019-02-04:
Any update on this?

orion3311, 2019-02-25:
Found this comment searching for FlashFloppy and H8D. This page may be a little more helpful with the format, and on the page is a dos program for running analysis on H8D images.

http://www.retrotechnology.com/restore/h8d_util.html

Hoping this works at some point, just aquired a Heathkit H89 and hard-sectored floppies are hard to find.

orion3311, 2019-02-25:
Actually to add to this...this page (another page on the OP's original link) describes the image format exactly.

http://sebhc.lesbird.com/h8d.html

keirf, 2019-02-25:
Okay the image files are straightforward sector dumps. However there is no documentation on the actual physical framing and encoding of these sectors on a real disk -- and I get the impression that (for hard sectored disks at least) the protocol is specific to Heathkit.

I would need documentation at that level, and I don't think it exists.

orion3311, 2019-02-27:
The thing with hard-sectored disks is that they use a hole in the floppy to denote the beginning of a sector. I found this which is a basic description: https://en.wikipedia.org/wiki/Hard_sectoring

Then I found this which is much better and hopefully means this is a fairly simple setup:
-----------------------
Having sector holes now allowed to have simple, 5 bit wide counter and comparator logic to 'find' a sector for reading or writing. The counter was (prepared to) reset whenever two pulses came in shorter than expected time (the index hole was placed between the holes for the last and first sector, so only one detector was needed) and incremented with every following pulse. Whenever the value was the same as looked for, comparator issued a pulse and the controller read whatever was coming along (and stopped by the next pulse in case of no data), or blurt out its data. When writing, no checks were made; when reading, sector headers were only checked after having read the whole sector to double check if it's the right track and sector.

Long story short, a hard-sectored disk can be read without without looking for index and sector markers, as everything is on the track - it only needs a logic fast enough to make the decision to keep a sector when track/sector number is the one it's looking for. Similar writing can be done by using appropriate timers.
-----------------------
That was from here (about halfway down the page): https://retrocomputing.stackexchange.com/a/4491

In the case of Heathkit...the disks are "10 sector" (I believe that means 10 sectors per track)

Hope this helps!

keirf, 2019-06-18:
The hard sectoring is the least of it -- the question is what the raw track bitstream looks like. This is not documented. It could be reverse engineered from a raw dump, if someone could interface a drive to Kryoflux or Supercard Pro and generate one?

menkelis, 2020-03-13:
And this has a very good description of a H17 emulator.
Has a explanation of generating index pulses. 

http://www.lesbird.com/sebhc/project8080/design_h17.html

snhirsch, 2021-02-06:
Hi, Keir.  If you are interested in adding support for Heath hard-sector format I can get you all the documentation you need regarding low-level format on the media.  It has been picked apart thoroughly by the Heath 8-bit hobby community.