keirf/flashfloppy #184: Amiga 500 - Boot Selector Won't Load

shan2752, 2018-11-22:
Hi,

I installed a df0 selector and have a new external gotek drive with FF.  I updated to the latest version of FF (0.11) but the boot selector wont load.  It reads the autoboot.hfe file but then nothing happens.  The drive has S0 jumpered (tried changing to S1 but wouldn't recognize drive entirely).  Aany ideas what the issue could be?  I am running kickstart 1.3.  Thanks!

keirf, 2018-11-22:
What if you use a USB stick which contains nothing except one bootable ADF image - does that boot?

keirf, 2018-11-22:
(I'm unclear whether autoboot doesn't load at all, or loads but then fails to initialise the Gotek)

shan2752, 2018-11-22:
Hi,

I tried a different USB stick with just an adf file and it doesn't load.  I bought a gotek with LED display and sound mod from AmigaStore.  It acts as if I am loading a non-bootable disk on a normal amiga floppy.  Accesses a few tracks then goes back to standard insert disk boot screen.  Any ideas?  Thanks for your help!

keirf, 2018-11-26:
What are you using to interface to the Amiga? Perhaps the enclosure/adapter is not working correctly? Perhaps try installing the Gotek temporarily inside your Amiga and see if it works there.

shan2752, 2018-11-29:
I tried that as well and still nothing.  I'm going to try what keirf recommended and report back when I can get to doing it.  Thanks!

sonicthewedge, 2018-12-03:
I had this problem with an external FDD.  It turned out the adaptor in the enclosure was shorting out on the case.  I stuck some rubber stand-offs in there and it started working again - exactly the same symptoms as you describe shan2752.  Hope this helps!

I still have a problem with it however.  I can get a game to boot if I rename it to AUTOBOOT.HFE but with the usual up-to-date (FW0.13 and FM1.74), I get a "d-A" notification on the LCD and an "LBA Change Test Failed" error.  Any clues?

keirf, 2018-12-03:
What if you revert firmware to 0.12 or even 0.11? You can do this via the normal update mechanism.

You can also download and test older file managers from releases section of the hxc_ff_file_selector repository too.

keirf, 2018-12-03:
Please try downgrading the file selector a version or two.

sonicthewedge, 2018-12-03:
Now on 0.11/1.71 and I get the LBA error and "000" on the LCD.

keirf, 2018-12-03:
The most valuable information is exactly which version (of fw or fs) introduced the problem. So try backing off further, even down to 0.9.x series of the firmware.

sonicthewedge, 2018-12-03:
0.9.31a/1.70 LBA error and "000".  There was a F02 error when I first tried it.  Booted in FS once the second time of asking.

keirf, 2018-12-03:
It seems the problem is not really version related? Perhaps something went bad in your hardware setup? Perhaps try running Systest (ADF download available in my Amiga-Stuff repository) and run the floppy write test.

sonicthewedge, 2018-12-03:
Could possibly be.  I have it working from another USB stick.  What I am trying to get to work again is a Pi Zero acting as a USB gadget.  It was working previously and I have rebuilt the entire thing to try to get it to work again without any luck.

Do you know what the "ERROR: LBA Change Test Failed : Write Issue ?" and "d-A" LCD errors are?  I cannot find references to them anywhere.

keirf, 2018-12-03:
Maybe try formatting the offending stick.

LBA write error means that the host tried writing a command to the Gotek but it failed. This gets retried a few times with latest file selector so it sounds like a bad wdata/wgate signal issue, if reads work.

D-A is the new message displayed when in the file selector. It means Direct Access. It's not an error.

sonicthewedge, 2018-12-03:
Are there any guidelines for the capacity, sectors, file format etc. for the USB stick?  As I mentioned, I am emulating a USB stick with a Pi Zero with Raspbian.  This might be causing the write problems though I have had it working before in a very similar way so I do know it is possible.

keirf, 2018-12-07:
FlashFloppy is not very fussy about the USB stick really. If you're mimicking a bog-standard stick you should be fine.

sonicthewedge, 2018-12-07:
Interesting - just had the LBA write error on the USB stick that was working.

Probably need to start over because this is not going well and I don’t seem to be able to get back to any working state.

Might be something more serious on the Amiga?

-----

Dr. Ed Young 
M.Math. M.Sc. Ph.D. FACS

keirf, 2018-12-07:
If writes aren't working properly then you could see weird errors. That can be as simple as a bad floppy cable. Perhaps do a debug build of FlashFloppy and get serial debug log when running your test setup.

closed 2019-01-07 by keirf

shan2752, 2019-03-23:
Hi,

I narrowed down the issue to the DF0: floppy drive cable.  When the floppy drive cable is unplugged, the Gotek will boot as DF0: with my ACA500 Plus.  Once I connect the real DF0: floppy drive cable, the Gotek will not boot.  Floppy disk games work fine so it doesn’t appear to be a cable issue.  Any ideas?  I’m truly stumped here.  Thanks!

tomse, 2019-03-23:
Sounds like you're trying to have both a gotek drive and a real floppy on the same cable.. i.e. a PC cable, which obviously doesn't work with amigas..

but you'd probably need to describe your setup before coming to conclusions.