keirf/flashfloppy #186: Error e34 On Gotek display bt using on PC

fran62, 2018-11-25:
Hello everyone,

My Gotek is flashed with the lastest 0.11 firmware. On my amiga, it works perfectly (with jumper on s0 + HXCSDFE.GFX + AUTOBOOT.HFE + FF.CFG + adf games placed on the usb stick)

but on my PC i have the **Error 34** as soon as the computer boot (before i moved jumper on S1 + JC, usb stick is formated in fat 32 and i use the usb floppy manager 1.40i  tools to create 100 blank  1.44mb image as Phil from Philcomputerlab explain on youtube.

If you know where is the issue please it would be nice !

Thank you

keirf, 2018-11-25:
The advice for use on PC is probably for original Gotek firmware, not FlashFloppy. All you need for PC is a formatted USB stick containing IMG files (these are eg 1.44mb files for HD PC floppy images).

fran62, 2018-11-25:
thank you for your answer, do you know where i can find this original firmware please ?

keirf, 2018-11-25:
It's installed on the Gotek from factory. When you overwrite it with FlashFloppy you cannot get it back. But FlashFloppy is a better more flexible firmware. You don't need the original firmware.

fran62, 2018-11-25:
ok but it's  strange that i get this error 34

keirf, 2018-11-26:
Not really since the stick is set up wrong. I suggest deleting all files from the USB stick and place on it instead files with IMG suffix. You should be able to find some online to test out.

fran62, 2018-11-26:
I tried several usb stick and created  100 blank images with usb floppy manager v1.40i or vitual image software  and at each time i plug one on the gotek the displays goes to F-F to E34 error.

when i plugg back the usb stick on the main pc, i see that all 100 partition have been deleted and instead a file has been created : IMAGE_A.CFG

keirf, 2018-11-26:
Stop using usb floppy manager 1.40i and instead place .IMG files directly on the stick, as I have previously said.

keirf, 2018-11-26:
ie you use it just as you did on Amiga, but instead of placing ADF files, you place IMG files.

fran62, 2018-11-26:
Success, it works with image. Do you how assign an image to a specific slot ?

keirf, 2018-11-26:
Add the following line to FF.CFG:
`nav-mode = indexed`

Now name your IMG files DSKA0000.IMG, DSKA0001.IMG, DSKA0002.IMG, and so on. It does not matter if there are gaps in your numbering: FlashFloppy will skip to the next slot with a correctly-named image file as you navigate with the buttons.

fran62, 2018-11-26:
thank you. 

Anyway, do i have to create each image manually, ?

keirf, 2018-11-26:
Copy your working IMG file to a new slot (eg. DSKA0001.IMG). Now boot your retro PC, select slot 001 and format it ("FORMAT A:" or whatever). Now you have a formatted/empty IMG file you can store away and then use for populating any further slots you need.

fran62, 2018-11-26:
Everything is ok now, thank your help keirf :)

closed 2018-11-26 by keirf

shrusee34, 2021-02-18:
Good afternoon, I have the same problem, but I want to connect the floppy to my computer. Tell me what am I doing wrong?

keirf, 2021-02-18:
Probably the stick is not formatted FAT32