keirf/flashfloppy #189: Amstrad NC200 correct disk image format

soydemadrid, 2018-12-20:
Hi I hope you can help.

I've managed to add a Gotek FlashFloppy drive to my Amstrad NC200 portable.

I know it uses DOS compatible 720kb disk but not sure what file formats or disk images to try.

I dragged a DOS .hfe disk to the usb to try and ask the machine to format it. But the NC200 just reports "No Floppy In Drive"...

I wondered if you knew anything I could try to get it to format or what to try - or if there is anyway you can please add NC200 support to FlashFloppy. Or perhaps I just need some kind of .cfg file to get it working?

Many thanks for any help.

keirf, 2018-12-20:
HFE will work but 720kB IMG is even better. Your problem is likely the interface not set up right. Have you tried adding a jumper at position JC? Another thing to try is moving the select jumper between S0 and S1.

soydemadrid, 2018-12-20:
Hi thanks for the fast reply.

Ok I've just tried a jumper on JC as well as S0 and then a jumper on JC as well as S1, then just S1 without JC and also S0 without JC...

All of them it just says "No floppy disk in drive".

It does look like it is trying though. When I turn on the machine it says F-F - then 000 with no green light on. When I go to the format disk option on the computer the green light flashes a few times at length before the "No floppy inserted" error comes up. It seems like the gotek drive is at least trying!

Thanks again for any help.

keirf, 2018-12-20:
How did you physically interface to the nc200? Because that portable doesn't use a regular 34 pin IDC right? Your problem likely lies in that interface. The drive select jumper setting that made the green led flash is the correct one.

soydemadrid, 2018-12-20:
Hi thanks - well I used one of these types of pcb adaptors:

https://www.picclickimg.com/00/s/ODAwWDgwMA==/z/XfYAAOSwBRFaNS4Z/$/34-Pin-Floppy-Interface-to-26-pin-FFC-_57.jpg

which I thought should work ok if they're being sold as adapters to go from gotek 34pin to 26pin?

keirf, 2018-12-20:
There's more than one 26 pin pinout I'm afraid. Would need more detail or schematic on nc200.

soydemadrid, 2018-12-22:
Hi here is the PCB diagram of the 34 to 26pin adapter I used:

[image]

The straight-through connectors of the ribbon plug alternate so 26pin pin no.1 is top left on the PCB and then pin 2 is top right etc in a zig zag.

Thanks again for any help :)

keirf, 2019-01-07:
Above adapter is for a standard PC "micro floppy" as used in laptops for example. I don't know whether NC200 is the same pin out. Bryce would know I'm sure. Did you make any progress?

soydemadrid, 2019-01-07:
Hi thanks for the reply. Bryce has kindly supplied me with the NC200 pinout which is as follows:

[image]

Is there anyway you can add this to the FlashFloppy firmware somehow so it works with NC200? Thanks again for any help :)

keirf, 2019-01-07:
Are you able to modify your PCB adapter (ie cut tracks and add jumper wires)?

Pin 2 Gotek should connect to pin 6 NC200. (Existing Pin 34 -> Pin 6 connection should be cut).
Pin 34 Gotek should connect to pin 8 NC200.

If you can do that, see how you get on.

soydemadrid, 2019-01-07:
Fantastic keirf. Thank you!! It is now working perfectly so it looks like the Amstrad NC200 can be added to the compatibility list! I'll make up some corrected PCB adapters and then if any of the other 10 users out there in the world want to fit a Gotek in their NC200 I can make them available to them!

Thanks once again, you've helped make life much easier as I still use this computer but now at least I can just stick things on USB and not worry about my floppies all dying on me...

closed 2019-01-07 by soydemadrid

weerd01, 2019-01-14:
Hi! I am probably one of those ten users! :) I would love to know which Gotek drive physically will fit into the NC200. Can you share with us which model/type you used? Thanks!

soydemadrid, 2019-02-10:
> Hi! I am probably one of those ten users! :) I would love to know which Gotek drive physically will fit into the NC200. Can you share with us which model/type you used? Thanks!

Hi we had to develop a PCB, some extra wiring and mounting bracket etc but we're selling them on our website as a complete kit to fit the NC200 here:

[Amstrad NC200 Gotek usb floppy emulator drive, cables & internal mounting kit](https://www.simulant.uk/shop/retro-computers-games-consoles/amstrad-cpc/Amstrad-NC200-Gotek-Floppy-Emulator-usb-drive)

weerd01, 2019-02-11:
Thanks! However, we managed to fix the internal drive by now (belt replacement). I will have to think about the amount of usage vs expenses for your (great) gotek/board solution.

explit7, 2025-03-28:
I think i missed the train couple of years ago. The Citizen Floppy in my Amstrad NC200 become a new drive belt in 2024, sometime its working fine, but the most time it has problems to read or format the floppys. If someone could help me to build/optain the Gotek Floppy drive for the NC200 - it would save my system...
Thanks

weerd01, 2025-03-31:
Hi @explit7 

I actually bought the following kit for my nc200, and it's fits really nicely:

https://www.simulant.uk/shop/Amstrad-NC200-Gotek-Floppy-Emulator-usb-drive?search=nc200

It contains the flashfloppy fw which is able to read nc200 disk images. I should be able to find a 'bare' image to which you can use in the nc200. Drop me an e-mail at weerd01@gmail.com if you need further information (not that i am expert!).

I mainly use it to run cp/mish (and thus, boots from the floppy image), but it's great for using the internal software too (which i find quite good).

My FF.CFG contains only one line: image-on-startup = static

All the best,

Dennis

explit7, 2025-04-01:
@keirf @weerd01 Sorry, one more question: How i need to configure my Gotek SFR1M44-U100 ?
As ibmpc ? Amstrad NC200 is not on supported list (yet). This part of the puzzle is missing for me yet.
Thanks