keirf/flashfloppy #191: FFManager error when gotek mounted externally.

ccapublic, 2018-12-22:
Hi,

In a nutshell, I have a gotek flashed with latest flashfloppy for use with an amiga. Whether I connect it internally or externally (in an old external df1 reader), it works super great (except for those games/demos that absolutely want df0 but this is not a FF issue).

However, when selecting autoboot image (FFManager), behaviour is strange.

- internal(DF0): Ok

- external (DF1) : starts then says "error : hxc drive not detected"

- external (with hw mod to make it df0) : starts then says "error : hxc drive not detected" 

Is it a known behaviour ?

Thanks A LOT to whoever could help !

Cyril

keirf, 2018-12-23:
Probably rdy signal (pin 34) is not getting through to the Amiga from your external enclosure. Eg try running Systest (linked from my Amiga--Stuff github repo) and run floppy read test Gotek as df1. Likely will warn no rdy signal.

keirf, 2018-12-23:
Or try running Arte by Sanity (famous demo) in external drive hacked to be df0. You will probably find it hangs at boot.

ccapublic, 2018-12-25:
Hi. Thanks a lot for your feedback. I checked my external enclosure and it actually doesn't propagate Gotek's RDY signal. Because all outputs are open collector, I created a wired OR by connecting Gotek's RDY signal directly to enclosure's generated RDY signal and
... Voila ! Works like a charm. Next step is to implement a DF0<=>DF1 swap mod (I'll probably use an arduino).

By the way 2 features that could be great for next releases :

- implement amiga motor latch and floppy identification directly in fw so that gotek can be directly connected to DB23.

- Implement a DF0<=>DF1 swap in fw. Provided the necessary signals are connected to the gotek (sel0, sel1, motx, INT_sel0, INT_mot...), it shall be pretty easy and would be GREAT too.

Thanks again

Cyril.

keirf, 2019-01-03:
Specify `interface = amiga` in FF.CFG will already provide you direct floppy identification. It is recommended when directly connecting Gotek to DB23.

keirf, 2019-01-03:
Drive swap in firmware is harder. The correct select and motor signals must be generated for the real floppy drive if it is switched to DF1, right?

ccapublic, 2019-01-03:
OK Great !

So :

1) If Gotek is directly connected to DB23 with interface = amiga, it will self identify on RDY line so that amiga sees a DF1 (if SEL1 is used). However, unlike a real amiga floppy, it does not need a MOT signal so it does not have to handle the amiga special MOTX signal to generate MOT1. Correct ?

2.1) DH0 hack : Here is what I do. I cut the SEL0 and MOT0 from the internal floppy ribbon and redirect the 4 cable ends (let's say SEL0in SEL0out MOT0in MOT0out) to a female DB9 at the back of the amiga. Under normal operation, Gotek is in an external enclosure (and behaves as DF1), and a DB9 male plugged in the female mentionned above short circuits SEL0in/Sel0out & MOT0in/MOT0out respectively(so internal drive acts as DF0). When I want the Gotek to be DF0, I simply leave SEL0in/SEL0out open MOT0in/MOT0out (so internal drive is off) and redirect SEL0out to SEL1in of DB23 connector of external enclosure. Works fine.

2.2) What I would like to do is that Gotek is connected to SEL0in, SEL1in, MOTx and generate a MOT1 signal (see hereafter).  In short, it will not have to generate a RDY identification for itself as DF0 because MB  of amiga knows there is one. It would just use SEL0 to operate or not. However, from SEL1in and MOTx it would 1/ generate a DF1 identification 2/ generate MOT1. MOT1 and SEL1 would be connected to internal floppy. We would end up with External (gotek) DF0 and internal DF1. Note : need to find IOs on the gotek.

2.3) If SEL0, SEL1, MOTX from amiga motherboard and SEL0 MOT from internal floppy were routed to gotek MB, we could even imagine a SW flippable DF0<=>DF1.

If I'm so interested by this it's because I need Gotek as DF0, I refuse to put it Inside my A1200, and sometimes I need to use gotek and real floppies.

I don't know if this is clear. I did my best ;)

Thanks again !

Cyril

keirf, 2019-01-04:
If you want to generate latched motor signals you will need some outboard hardware. I don't intend to implement that in firmware, and latching fast enough may not be possible anyway.

I think Zaxon on SMR may have some prototype boards for doing motor/rdy signals for all 4 Amiga drives.

closed 2019-01-04 by keirf

ccapublic, 2019-01-04:
Actually, it would be both generating the identification from SEL1 and SEL0 and latching motx on SEL1. I have no idea how long the motx signal remains after sel1 falling edge so I understand it might be difficult in fw. I'll resort to the good old 7474 and 7438 :)

Thanks a lot for having taken time to answer and for your work on this awesome fw.

Cyril

ccapublic, 2019-01-04:
Sorry for last question : can you let me know where in  the code you generate the amiga identification signals when amiga mode is activated ? Cyril.

keirf, 2019-01-04:
See `update_amiga_id()` in src/floppy.c. For DD images it constantly asserts pin 34 (RDY). For HD images RDY must be toggled and that is handled in `Amiga_HD_ID()` in src/gotek/floppy.c which is called each time the drive is selected or deselected.