keirf/flashfloppy #192: HxC_FF_File_Selector Build Error

sonicthewedge, 2018-12-25:
Hi,

Building with the following script from ~/HxC_FF_File_Selector with all the prerequisites installed: 

```
export PATH=$HOME/install/bin:$PATH
export PATH=$HOME/exe2adf:$PATH
export LD_LIBRARY_PATH=$HOME/install/lib:$LD_LIBRARY_PATH

make clean
git clone https://github.com/keirf/HxC_FF_File_Selector.git
debug=y make -j8
```
yields the following error:

```
make bootblock.bin
make[2]: Entering directory '/home/ff/HxC_FF_File_Selector/amiga'
m68k-amigaos-gcc -o bootblock.o -c adf/bootblock.S -Os -I./inc -I.. -I../fat32 -nostdlib -DFATFS_IS_LITTLE_ENDIAN=0 -m68000 -Wall -Werror -std=gnu99 -msoft-float -fno-builtin -D__ASSEMBLY__ -Wa,--register-prefix-optional -Wa,-l -Wa,--bitwise-or -Wa,--base-size-default-16 -Wa,--disp-size-default-16 -Wa,--pcrel
m68k-amigaos-objcopy -O binary bootblock.o bootblock.bin
**m68k-amigaos-objcopy:bootblock.bin: invalid bfd target**
Makefile:29: recipe for target 'bootblock.bin' failed
make[2]: *** [bootblock.bin] Error 1
make[2]: Leaving directory '/home/ff/HxC_FF_File_Selector/amiga'
Makefile:18: recipe for target 'AUTOBOOT.HFE' failed
make[1]: *** [AUTOBOOT.HFE] Error 2
make[1]: Leaving directory '/home/ff/HxC_FF_File_Selector/amiga'
Makefile:14: recipe for target 'all' failed
make: *** [all] Error 2
```

Possibly something to do with the latest version of the gcc toolchain?

Warm regards.

keirf, 2019-01-03:
Reproduced this error myself. "binary" no longer seems to be a valid target for m68k-amigaos-objcopy. I will need to write a script to strip the Amiga hunk headers.

keirf, 2019-01-03:
Now fixed.

sonicthewedge, 2019-01-04:
exe2adf v0.3e - coded by Bonefish/Reality 2015-2017.
----------------------------------------------------
disk-analyse AUTOBOOT.ADF AUTOBOOT.HFE
disk-analyse: could not open config file "formats"
Makefile:18: recipe for target 'AUTOBOOT.HFE' failed
make[1]: *** [AUTOBOOT.HFE] Error 1
make[1]: Leaving directory '/home/ff/HxC_FF_File_Selector/amiga'
Makefile:14: recipe for target 'all' failed
make: *** [all] Error 2

sonicthewedge, 2019-01-04:
Copied file "formats" into build directory and that seemed to resolve the problem.  Now I am getting it for the Atari ST build but the fix does not work this time:

```
file OUT.ST written, length: 737280
disk-analyse --format=atari_st OUT.ST AUTOBOOT.HFE
disk-analyse: could not open config file "formats"
Makefile:39: recipe for target 'OUT.ST' failed
make[2]: *** [OUT.ST] Error 1
make[2]: Leaving directory '/home/ff/HxC_FF_File_Selector/atari_st/loader'
Makefile:30: recipe for target 'AUTOBOOT.HFE' failed
make[1]: *** [AUTOBOOT.HFE] Error 2
make[1]: Leaving directory '/home/ff/HxC_FF_File_Selector/atari_st'
Makefile:14: recipe for target 'all' failed
make: *** [all] Error 2

```

keirf, 2019-01-04:
Did you `make install` disk-analyse to the default location (/usr/local)? It will looks for formats in `/usr/local/share/disk-analyse/formats`.
If you changed the install location then it should look in `<install_loc>/share/disk-analyse/formats`.

So, basically, check you installed disk-analyse properly.

sonicthewedge, 2019-01-04:
Copied "formats" file to ~/HxC_FF_File_Selector/atari_st/loader directory and the build works.

:+1:

sonicthewedge, 2019-01-04:
/usr/local/share/disk-analyse/formats exists.

keirf, 2019-01-04:
Run `strace disk-analyse` and see what location it tries to access for formats file.

keirf, 2019-01-04:
Or run `strings` on the disk-analyse binary to see what search path is embedded in it

closed 2019-01-04 by keirf