keirf/flashfloppy #193: Gotek unable to recognize 1.44MB images with 0.14

Springboks1970, 2018-12-26:
Hey all. I have a GoTek flashed to 0.14. It is installed in an IBM 5160. I have it attached to a LCS 6814 8-bit ISA Floppy controller that supports 360/720/1.2/1.44.

I have the controller jumpers set to be 1.44 for drive A: and 360KB for B:

I also have the A: drive configured in my config.sys as drivparam=/d:0 /f:7, i.e. Drive A is 1.44MB

Prior to flashing it with 0.14, it was able to read 1.44MB images created using the Batch Manage Tool V1.40.

After flashing, the GoTek is only able to read 720KB images. Attempting to read 1.44 or 1.2 (when drive A: is configured as 1.2) generates a "sector not found" error. You can see the green activity light flashing. It just can't read.

keirf, 2018-12-26:
You aren't using the batch tool with FlashFloppy right? As that tool is for the original firmware only.

What image format / file extension are the 1.44 images?

Tried another USB stick?

Springboks1970, 2018-12-26:
Hi. 

No. Was just point that the tool worked with stock GoTek and was able to read 1.44 stock. 

Using .img with 1.44 generate using BFI.exe

Springboks1970, 2018-12-26:
Oh forgot the last question. Tried 3 different usb sticks. Both usb 2.0 and 3.0

keirf, 2018-12-26:
Could you try an older firmware version? Older versions are on the download page, and you can downgrade by the same process as upgrade. Perhaps the latest available 0.9.x. Just to see if this is a regression.

Springboks1970, 2018-12-26:
OK... Will try 0.9. I did flash to 0.11 and still had the same issue.

Springboks1970, 2018-12-26:
So I installed 0.9.31a. Switching to A: gave a sector not found. The USB only had the cfg files and a blank 1.44 IMG. I then attempted a Format A: /S. The format command froze. After rebooting, my C: drive is now hosed as well as the A drive and my PC is bricked. I have tried 3 floppy controller cards, and I am receiving a 601 error of all 3. I suspect the  controller board on the drive is now damaged. The HD Controller refuses to even see the C: drive now. I am not saying that 0.9.31A did that. I am just reporting what happened. I will attempt to recover my PC. Will install a new floppy drive to verify that controller board on other is hose. Will them low level format the C: to see if I can recover it. It really sucks though. This was a MINT "Like new" 5160. The HD was a 30 MB with ZERO defects too. I will repeat, I am not blaming FlashFloppy. I am just reporting what happened.

keirf, 2018-12-26:
That's a very unhappy ending :( I'm not sure how the Gotek could cause something this catastrophic, if it is the cause it would be something like a catastrophic power short (a Gotek hardware fault) that's blown a power rail on the controller card (for example). I imagine the FORMAT is not to blame, but perhaps the first thing you tried after the catastrophe (best guess: C drive was already gone, and DOS hung when it tried to access C:FORMAT). The root cause could be hardware borkage on the controller or the Gotek though....

Sounds like a hardware fault tracking issue for now... :(

Springboks1970, 2018-12-26:
Any chance you could check IBM 1.44 functionality just to verify that it works?

keirf, 2018-12-26:
I can do but not until next week. But there are 1000s of FlashFloppy users a decent proportion using IBM HD images on PC or on retro synths (among other things). If I break a major format like that then I usually hear about it from multiple users pdq. However, I can break out my Supercard Pro and confirm it can read a full HD image off the Gotek when I'm back at desk.

Springboks1970, 2018-12-26:
May I ask what extension to use for 1.44 image and can you tell me what application you use to create your 1.44 images?  If possible,could you email me a valid image just to verify I am not crazy?

CarlosTex, 2018-12-26:
I seriously doubt your issue was related to Flashfloppy. I have 0.14 on all my Goteks, including a IBM 5160, Tandy 1000 SL, 286 AT clone, and Pentium MMX machine. All DOS based systems using Flashfloppy, with all formats working well.

Is your 30MB HD an MFM drive? Those things just die... In this time of year MFM drives are very sensitive to cold temps, specially if the stepper motors aren't lubricated properly. 

Remove the HD controller card from the system, get the system to boot without it. Leave only the video and floppy controller cards in. Check if the 601 error still appears. Also check if the floppy card is inserted properly and try it on other slots too.

I actually recapped my IBM 5160 64-256KB motherboard, removed all tantalum capacitors and soldered brand new solid polymer capacitors. Your IBM 5160 may look MINT but it is still probably 35 years old or so. These things need maintenance. I'm sure yours can be brought to normal operation again.

CarlosTex, 2018-12-26:
So you are trying to use a 1.44MB floppy image with your 5160? What kind of floppy controller do you have?

Remember one thing: the IBM 5160 BIOS does NOT recognize 1.44MB formats, thus its not a valid format. Unless you have a floppy controller with a custom ROM that adds High Density Floppy extensions you will never be able to read or write 1.44MB floppies.

Springboks1970, 2018-12-26:
I have an LCD 6814. Please read the first comment for details.

CarlosTex, 2018-12-26:
Try to create a 1.44MB image with WinImage. It works for me everytime. If that does not work try to use 2M driver, it hooks to INT 13H and gives you HD floppy support via config.sys. You can ditch that driveparm command, i never used it and never needed it. In fact, remove that driveparm command and try again.

Springboks1970, 2018-12-27:
Can you link me to the "2M driver". Also, if you intend on formatting a blank image, you will need the drivparm command, otherwise the 5160/50 will only format it to 360KB, no matter what format params you specify.

Also... do you actually have a 1.44 working on a 5150/60? As mentioned,  I have NO issue with 360/720 on them.

I have created an image using WinImage, and it does not work.

If anyone that reads this thread has successfully accessed a 1.44MB image using a 5150/5160, I would love to hear from you. What FD Controller card are you using. What program did you use to create your image. Right now, only people saying "it should work" have chimed in, but no one that has actually tried a 1.44 on a 5150/60 has.

Finally... CarlosTex can you attached a valid 1.44 img to your reply so I can test it for sanity.

CarlosTex, 2018-12-27:
Like Keir has stated: thousands of users use flashfloppy with XT machines that do not have native HD floppy support just fine. If he breaks anything people will be all over here, myself included.

It's not the case of "it should work" because it works, i know because it works with myself and other thousands of users. I've been selling lo tech XT-IDE cards on amibay, sometimes with glued HD floppy BIOS extensions to the XT-IDE BIOS. Combined with a normal floppy controler with support for the HD floppy data rates it works fine.

When you start your XT do you see any BIOS extensions appearing on the screen? I bet that there is something fishy with your controller.

I'm not at home right now, but surely you can do a Google search for 2M MS-DOS driver?

Springboks1970, 2018-12-27:
Just thought I would provide an update.

I managed to unbrick my 5160. I suspect that maybe I shorted something by accidentally touching a trace, since the PC was open.

I redid the low-level format on the HD, followed by an exhaustive SpinRite analysis (took almost 16 hours). Good news is that there are still zero defects. Very happy about this.

As for the 601 error. My bad. I had harvested the 1/2 height floppy from another machine. It was missing the drive terminator.

As for the 1.44MB issue. After installing 2M-XBIOS I am now able to read 1.44MB images.

Here is the link the the minuszerodegrees article that provides detailed instructions. Thought I would add it in case anyone ever has this issue, and it will help them

http://www.minuszerodegrees.net/2M-XBIOS/2M-XBIOS%20-%201.44M%20as%20B.htm

Thanks CarlosTex and Keirf. Apologies for the thread. There really wasn't a good article I found on how to do this. Not many people have 1.44MB controller cards installed in a 5150/60.

Hopefully if all this thread accomplishes is to help someone else in the future...  it was worth it.

keirf, 2018-12-27:
Excellent news that this is resolved. Perhaps I should mention the 2M driver thing in the wiki under ibmpc platform specifics.

CarlosTex, 2018-12-27:
I just want to add that you should not need 2M driver if you already have a floppy controller card with its own high density floppy BIOS. 

Glad that ita working

Springboks1970, 2018-12-27:
Turns out you do. I pointed out I the first post that I was using a LCS card that supported 1.44. Of you look at 2M’s XT help, it actually states that some 1.44 controller bios’s may still require 2MXT to be installed. Turns out the LCS is one of those cards. Without it, 1.44 img does not work. With it. It does.

CarlosTex, 2018-12-27:
That does not make any sense, i'm sorry to put it this way. A floppy controller with support for high density floppies whether it is 1.2MB or 1.44MB must support both the data rates and supersede the system BIOS with its own extension BIOS. No additional driver is needed. Otherwise there's absolutely no point in the card having its own BIOS. Imagine this: if it only works by using the driver you'll never be able to use high density floppies during boot.

From what i can see you have 3 selectable ROM areas for your card CA000h to CBFFFh, CC000h to CDFFFh or CE000h CFFFFh, its a 8kb ROM then. I assume your HD controller should be set to the standard C8000h to C9FFFh. 

So if you really need th 2M driver to make it work there is some problem with your LCS card or some kind of ROM conflict. The point of having a floppy card with its own ROM was to have all floppy formats available from boot time. If you need a driver it defeats the point. 

In fact, you state you have a 6814 LCS card. I see BIOS dumps for 6810 LCS cards, but not the 6814. 

If you don't mind use this utility caled DUMP.EXE on your DOS prompt. Check first which jumper your LCS card is set, then you run this tool below:

ROMDUMP.EXE CA00h BIOS1.bin
then
ROMDUMP.EXE CC00h BIOS2.bin
then
ROMDUMP.EXE CE00h BIOS3.bin

If both 3 files return blank then it means that there is something wrong with your card, or the BIOS EPROM.

[ROMDUMP.zip](https://github.com/keirf/FlashFloppy/files/2713407/ROMDUMP.zip)

Springboks1970, 2018-12-27:
Will check it out tomorrow. 

A couple of things. 

1). I have 2 of the exact same lcs cards and both do not with with 1.44 images. 

2) Both these card were able to read 1.44 images using the stock GoTek bios and the tool to create images on usb

They only stopped working when flashed with flashfloppy

keirf, 2018-12-28:
Do you have a real drive plus some floppies you could compare with? Ie does real hardware work with card original bios?

Springboks1970, 2018-12-28:
Keir. I will check my stash. Not sure if I have a 34 pin one, but I will check

closed 2019-01-07 by keirf