keirf/flashfloppy #197: [Wishlist] Add ability to emulate disk image from folder

Tim5000, 2019-01-08:
Appreciate this will not be useful for all file formats, but  instead of supporting .IMG files, some versions of the GOTEK firmware allow 1000 folders to be put on the USB stick.
To use this, source files can be dragged/dropped into the folder on the USB drive and the host PC is able to access them as if the folder was the disk.
This is useful because no special tools required to manage .IMG files
Is this something that could be offered, if it isn't already?
Thanks.

DaTwo, 2019-01-29:
I 2nd this. I really hope to see this added. I use a gotek drive in an older CNC machine. Adding this ability with the ability of the OLED to see whats in the folders would be a godsend. I sure hope you can make this happen. Here is the drive I currently am using. This should help to explain how this feature works. 

http://iw.suntekstore.com/images_cache/270/10019783/136383371310019783.pdf

keirf, 2019-01-29:
How would this be more useful than just using the original firmware? It would take quite a bit of Flash space to implement, or would require a separate file-based version of FlashFloppy (c.w. existing image-based FlashFloppy).

DaTwo, 2019-01-29:
What would make this gold is the OLED screen. As of this point there is no way at the machine to know what is on the USB stick. Computers are not on the shop floor. Having the ability to look through the folders and see file names would be a huge help. I would be more than happy to help you test this. Hope you can make it become a reality.

pkurucz, 2019-04-11:
Agreed.  Paired with the oled, this would be awesome for the shop floor.  I'll take a look at the code over the coming weeks and see if I can hack a proof of concept together.

keirf, 2019-04-11:
What would be the user interface: Would you be able to browse all files/folders on the USB stick, and then pick one to form the root of a FAT12 volume presented at the floppy interface? Would you want to continue being able to browse files/folders while the FAT12 is mounted (this is harder to do, especially with the code structured as it currently is)? Would you want changes to the FAT12 volume to be transferred back to the files/folders on the USB stick when the volume is unmounted/ejected? etc.

This is not a project of great interest to me personally, and I think it would be a fork/branch of the existing firmware -- I don't see it coexisting in a single FlashFloppy firmware image. I would need to be contracted to work on it, though of course others are free to work on this feature as they like.

Tim5000, 2019-04-11:
From my perspective, I think it's moved on to a slightly different scope than what I originally raised. Maybe this should be split out into two wish-list tasks (even if the answer is neither of them are likely to be done)?
My use case is I generate IBM PC disk images for use on my musical instrument, and I cycle through them via the 000-999 7 segment display. 
What I thought would be nice is if I didn't have to do the disk image generation and I could just put a folder on the root of the USB stick instead. Cycling 000-999 would pick up the next image or folder based on the FAT as it does today with images.
For my use-case I've no wish to see the files inside the images or support other displays etc.

I haven't looked through the code but my thought was maybe once a folder was selected, it could essentially create a blank FAT12 disk image and copy the files to it - either into RAM or a temporary file on the USB, and then the existing code would then deal with that as if it was a typical image.

Manually I tend to perform those steps via the following commands
`
  dd if=/dev/zero of=blank.img count=1440 bs=1k
`
`  /sbin/mkfs.msdos blank.img
`
`
 mcopy -i blank.img folder/*.* ::/
`

I appreciate this is something difficult to keep common for all formats (which then might require custom configs in each folder) - just wanted to clarify my OP, but I'll look forward to any work from you or third parties.

pkurucz, 2019-04-12:
> What would be the user interface: Would you be able to browse all files/folders on the USB stick, and then pick one to form the root of a FAT12 volume presented at the floppy interface? Would you [...]

I think what your suggesting sounds good. You can pick a top level folder to be the root of the fat12 volume.  We make a temp disk image that gets mounted.  All read and write operations happen on this image.  On unmount the image gets unpacked and the contents written into the target folder.  

I don't think it's necessary at all to be able to see the contents of the folder change while the image is mounted.

keirf, 2019-04-12:
@pkurucz Is there open source code to pack and unpack FAT12 volumes? With that, a branch of FlashFloppy to implement what you suggest is pretty darn easy.
Doing the FAT12 code from scratch isn't *too* bad I imagine, but there must be something already out there...

penfold42, 2019-04-12:
Another fatfs instance ?
The underlying block store is linked to a file on the USB stick or sd card or even add SPI ram to the Gotek

keirf, 2019-04-12:
> Another fatfs instance ?
> The underlying block store is linked to a file on the USB stick or sd card or even add SPI ram to the Gotek

Oh yes that's obvious actually. This could be quite small code then, and perhaps actually coexist in the single FlashFloppy build.

DaTwo, 2019-04-13:
Does anyone happen to have the SFR1M44-FU-DL version of the Gotek drive? I would be happy purchasing one for a developer to look at. Basically I am hoping that we can make the drive function the same way as it does now with the original Gotek firmware but using the flashfloppy firmware with the addition to use a OLED screen to select the folder and display its contents (file names).  The way it works right now is you select the folder number on the drive LED display such as "002" and then insert the usb stick. One button loads that folders contents into the drives flash memory. You can the read or write to the drives flash memory. If you want to save the modified files back to the USB stick you press the other button and it overwrites the same folder number with the flash memory contents. If we could bypass the drives internal flash memory and just read and write directly to the USB stick that would be the absolute best. Then if you really want to fancy maybe allow serial commands to control the drive. That would really open up doors on these old machines.  Dare I even dream about adding in a wifi module? :) I would be happy to help anyway I can.

keirf, 2019-05-08:
Note that HxC now implements this feature, so purchasing a license is a good option for those who need this feature now. I'm also interested in reports on the user experience of how HxC has implemented it: it will help inform how to implement this functionality in FlashFloppy, from a UX perspective.

staticmem, 2019-05-13:
Getting back to the original post LibDsk supports this and is is referred to as "rcpmfs":
rcpmfs - a backend that makes a Unix/Windows directory appear to be a CP/M disc image.
I make use of it in an emulator that uses LibDsk.

Not sure if there is anything specific about it making it only for CP/M disks.  It's here...
http://www.seasip.info/Unix/LibDsk/

jcnventura, 2019-05-14:
I think that having a special folder for this use would satisfy everyone.

keirf, 2019-10-09:
This is probably next, now that Quick Disk is hopefully out of the way bar bug fixes.

DaTwo, 2019-10-09:
Great news. Happy to help anyway I can.

Tim5000, 2019-10-09:
> This is probably next, now that Quick Disk is hopefully out of the way bar bug fixes.

That would be amazing - Let me know what I can do to assist. I guess to start I should update my FF as I'm currently way back on 0.15...

keirf, 2020-09-19:
I'm not going to bother. On Linux you can loopback mount IMG files. On Windows you can use WinImage. The danger with this feature is it's a can of worms: Folder getting out of sync with dynamic IMG; and lots of time taken to sync back and forth between IMG and folder (circa up to a minute I would guess). It is better to take it on the chin and prep the IMG files offline on a PC.

Edimahler, 2020-09-27:
Oh holy. Was just reading through this posts and was sooo happy, finally having found a solution which would work in a easy way to get my midi files to my Clavinova, after hours of experimenting and not being successfull. As many others state as well here, this would be a really helpfull feature, especially together with the OLED display. Is there anyone else out there beside keirf, who's able to implement this in FlashFloppy? Would highly appreciate this!

Otherwise I will try and change to HxC. Didn't want to change there as I like the many configurations one can make in FlashFloppy. But having the feature of writing into directories directly would definitely be a key feature for the daily use.

keirf, 2020-09-27:
I encourage you to get an HxC license and try the feature out. Use it in anger for a while, reads and writes, and report back. Maybe I can be convinced to change my mind again.

Edimahler, 2020-09-27:
> I encourage you to get an HxC license and try the feature out. Use it in anger for a while, reads and writes, and report back. Maybe I can be convinced to change my mind again.

Will think about this! At the moment I'm already happy, that I finally found a solution how to make the Gotek working on my Clavinova CVP-208 (Yamaha) finally (Jumper settings: just S0 / Pin 34 to chg in FF.CFG). Partially happy now. ;-)

A big thank you in any case from my side for the great SW!

keirf, 2020-09-28:
That's great, thanks! However I'm always happy to receive reports good and bad from the "other side" too. :)

Edimahler, 2020-09-29:
Thanks keirf!

A question, which is about 80% off-topic here - but not completely: ;-)
You mentioned above the risk of defectuous images when implementing this idea here. - I just had an errorneous image after powering off my Clavinova the last time, can't read it in the Gotek neither in the HxC Floppy Emulator Software on the PC. Do I have to unmount my USB stick before powering off the machine? (I'm pretty sure, there was no writing on the stick at this moment, nevertheless the stick is blinking all the time.)

And as additional question: Would the risk of a destroyed "folder-image" even be bigger than in the normal usage?

keirf, 2020-09-29:
@Edimahler FlashFloppy does not delay writes to stick. If you did not power off during a write you should have been fine. Some sticks theoretically may cache writes in internal ram, but I'm sure even they must flush quite aggressively. Does Clavinova use FAT IMG files?

The risk with folder images is that you get the folder and the folder-image out of sync if you're not careful.

Edimahler, 2020-09-29:
@keirf: I was using *.hfe-files until now, which actually worked fine. But as I'm an absolute greenhorn with those image stuff, I don't know, if this was the perfect choise then. ;-)

Updating the folder image always when writing into the folder would be enough, if I'm right. But yes, you're right, when doing the update while powering off the system, one could destroy the image there (as well)...