keirf/flashfloppy #198: PC98 HDM floppy image support

barbeque, 2019-01-26:
FF has FDI support for the PC98 right now, which works great, but there are a lot of BKDSK-style disk images in circulation (e.g. the Neo-Kobe set on archive.org). BKDSK seems to be an image dumper/writer that generates raw images, and the format is popular with emulators.

I managed to work out from reading [Japanese PC98 enthusiast sites](https://melog.info/archives/2004/09/30223144.html
) that BKDSK supports creating images in these "formats," identified by their extensions:

# MS-DOS disk formats:
 * HDM -> 2HD/360rpm -> 1024 bytes * 8 sectors * 2 surfaces * 77 cyls = 1,261,568
 * HD5 -> 2HD/360rpm -> 512 bytes * 15 sectors * 2 surfaces * 80 cyls = 1,228,800
 * HD4 -> 2HD/300rpm -> 512 bytes * 18 sectors * 2 surfaces * 80 cyls = 1,474,560
 * DD6 -> 2DD/300rpm -> 512 bytes * 8 sectors * 2 surfaces * 80 cyls = 655,360
 * DD9 -> 2DD/360rpm -> 512 bytes * 9 sectors * 2 surfaces * 80 cyls = 737,280
# N88-BASIC disk formats:
 * HDB -> 2HD/360rpm -> (128/256) bytes * 26 sectors * 2(?) surfaces * 77 cyls. First track single-density (128-bytes), latter tracks 256 byte double density
 * DDB -> 2DD/300rpm -> 256 bytes * 16 sectors * 2 surfaces * 80 cyls

I have only ever seen HDM format in circulation, however, so I wrote this conversion code: https://github.com/barbeque/pc98-disk-tools/blob/master/hdm_to_fdi.py

FDI is basically the HDM raw image, with a 4096-byte header glued onto the front of it.

It seems to work really well with the FlashFloppy firmware on my 9821AP2, but it would be nice if I didn't have to convert it myself.

I looked into adding it to the code myself a few months ago, but didn't get around to it and now I've forgotten exactly what to do. From what I can remember, the steps to adding this to FlashFloppy are:
 - Add new filetype entries for all of those extensions (probably gated by `pc98` mode),
 - Pass it off to the 'raw image' handler with geometry set based on the extension (maybe just HDM for now),
 - (Ideally) find some examples of the other, non-HDM, images to test with.

If I get some time soon, I will look into doing this myself again and stand up a WIP PR for comment. Thanks!

keirf, 2019-01-26:
Yes, add a handler to image/img.c similar to pc98fdi but hardcoded parameters rather than reading from a header struct. You can see what to pass in to mfm_open() based on what pc98fdi_open() does with your generated header.
Then add your new handler to the list in image/image.c and you're done.
No need to gate on host=pc98 as this file extension is unique.

barbeque, 2019-01-29:
I just did this test (renaming HDM -> IMG, setting host=pc98) and it was successful. I'll update the firmware on my Gotek and test HDM.

closed 2019-02-04 by keirf