keirf/flashfloppy #201: d-R or d-A error message

vaskag, 2019-02-12:
In gotek with latest firmware (1.0) I get this 3 digit error if have in the usb, separate folder for each letter of the alphabet. Is it because of the many folders?

keirf, 2019-02-12:
This is expected, it just shows the Gotek is in file selector mode. Display should return to normal when you exit the selector.

vaskag, 2019-02-12:
Ok I didnt notice that it was ok. thank you
Is there a limit to how many files can exist in the usb? Because in the above scenario with 24 folders and 4000+files the hxc manager only shows 3 out of 24 folders and cannot load anything.

keirf, 2019-02-12:
There should be no such limitation. Is this on Amiga?

vaskag, 2019-02-12:
Amstrad cpc 6128

keirf, 2019-02-12:
I don't know so much about the Cpc selector. I would have to test it myself. But right now I don't do my own builds of it.

vaskag, 2019-02-12:
If I put all the files (4000+) in single folder, the manager works and loads the files. I think the problem is when inside the folder, are 24 additional folders.

keirf, 2019-02-12:
It's probably a selector bug. The cpc selector is the only 8 bit selector and is a different code base than Amiga and ST

vaskag, 2019-02-12:
Ok. Thank you for the feedback

closed 2019-02-25 by keirf