keirf/flashfloppy #203: Support for Quick Disk

graham-sportsmgmt, 2019-02-18:
Some of the oldest samplers use a format called Quick Disk ([wikipedia](https://en.wikipedia.org/wiki/History_of_the_floppy_disk#Quick)), and the drives are quite unreliable.

It would be great if FlashFloppy could support QD emulation.  This would breathe new life into these samplers:

Akai S612
Akai S700
Akai X3700
Akai X7000
Roland S-10
Roland MKS-100
Roland S-220

There are also some old drum machines, sequencers, and computers that use the QD format.

keirf, 2019-02-25:
Are there existing old QD images to be supported, or is this about supporting creation of new QD images (from a blank image)?

graham-sportsmgmt, 2019-02-25:
Yes, there is a Roland sample library for QD that I own most of.  Roland has been good at letting the older libraries go into the public domain.  I've seen the Akai disks on very rare occasions.

However, I'm sure most people would be happy with the creation of blank images and creating their own disk images.

chrispoacher, 2019-02-25:
Most of these old QD samplers have the ability for midi sample dump which is how most of the libraries are  on the net in .mid format. It wouldn't be much trouble for someone to re-dump those back to blank QD disk images to make a new repository :)

keirf, 2019-02-25:
Can these old samplers format a completely blank/unformatted QD?

keirf, 2019-02-25:
Further references:
https://archive.org/details/roland_S-10_SERVICE_NOTES/page/n9
https://nesdev.com/FDS%20technical%20reference.txt

graham-sportsmgmt, 2019-02-25:
I'd be happy to convert those to disk images.

chrispoacher, 2019-02-27:
> Further references:
> https://archive.org/details/roland_S-10_SERVICE_NOTES/page/n9
> https://nesdev.com/FDS%20technical%20reference.txt

I'm guessing all those 10 pins make up the shugart interface :) Maybe this will help me figure out the D20 drive interface pins to the Gotek.

ewanuno, 2019-04-22:
some more info:
the QD drive is also used by the Sharp-mz800 so the mz800 comunity already did some work on this including an arduino based qd emulator.
http://arduino.validalo.net/wiki/index.php/Quick_Disk_emulator_for_Sharp_MZ-800
https://sourceforge.net/projects/mz800emu/
https://web.archive.org/web/20170831235751/http://www.sharpmz.org/qdinside.htm

i have an akai s612 sampler
the service manual covers the QD drive:
http://www.florian-anwander.de/akai_s612/Akai_S-612_Service_Manual_Part_2.pdf

keirf, 2019-06-18:
The basis for support here would be a single-track HFE image, where that track is particularly long (Rotation rate might be approx 8 RPM) and with data rate chosen for the specific QD host.

The extra complexity is then the special signals used by QD (and which I think vary across QD drives):
1. There is a simple media switch signal, a bit like disk-changed
2. There is a home/ready signal that asserts for a couple of seconds while the head moves back to the start of the spiral track
3. Various signals may be active high rather than active low

This needs someone with a QD system to do some testing with a logic analyser. Fundamentally QD is a very simple format, it should be possible to shoe-horn support into FlashFloppy's Shugart-like emulation. Even if not in the same firmware, then shipped as an alt firmware.

pacific202, 2019-06-18:
That's great news.  I now have a Roland S-220 with QD so I can support these efforts as soon as the drive comes back from being calibrated after I replaced the drive.

I can't tell you how much of a pain these are to work with, but [this article](http://www.famicomdisksystem.com/tutorials/fds-repair-mod/belt-replacement-adjustment/) will show you the steps required to change the belt and re-align the heads.  I gave up trying to align them and sent the drive to someone else to do it.

Anyway, please let me know when you want to proceed and I'll buy or borrow a logic analyzer.  Just give me the steps and I'll hook it up.  Alternatively, I could just send you the S-220 when the QD has been repaired.

There will be a LOT of happy sampler users when this works.

keirf, 2019-06-18:
What you need is a Saleae 8-channel logic analyser. You can find them on Ebay for a few pounds. They tend to come with Dupont jumper cables which unhelpfully do not have grabber probes for clipping to signals... so you also need "mini grabber probes" (the flat sort are best, not the round ones) again 10 on Ebay is a couple of quid.

You then chop one end off the Dupont cables, solder on the grabber probes. Plug the connector ends into the Saleae, and clip the grabber ends onto the signals between QD drive and host.

Then you fire up the Roland and start a Saleae trace on your PC. I can give more details when you are ready.

pacific202, 2019-06-18:
Super, I'll let you know when the QD arrives back and I have it installed.

keirf, 2019-06-18:
You might want to order the ebay bits now as the lead time from China is at least two weeks.

ewanuno, 2019-06-21:
i already have one of those logic analyers and an akai s612, but alas, no QD drive.  if there's somebody near me with a sampler that has a working QD drive i am available to do these measurements.   i'm in turin, italy.

pacific202, 2019-07-12:
I'm sorry about the delay, but we've chosen to move back from the UK to the US and I've been busy with everything that entails.

I now have a working QD, and will be able to get you all the data in August once I'm back.

AMSynths, 2019-08-06:
Hi, keen to help this project move forward. The QD protocol is described in the S10 manual and its a simple datastream encoded in MFM, much like a tape recorder but a spiral across the disk surface.  By plan was to use a PLL locked onto the datastream as it will vary in speed and the clock will need to resync as the drive is very crude. I wasnt thinking of reusing a gotek but going with a custom micro controller but if the gotek can take a 64kb track length then maybe... but the drive selection protocol is not like IDE. Any views on a detailed design? Rob at AMSynths

AMSynths, 2019-08-07:
I think these are the challenges to overcome:
1. The QD MFM data rate is 101,564hz which is slower than Shugart. I assume the ARM clock rate can be adjusted in the source code
2. The RAM storage of the Gotex ARM chip is too small to hold the QD data stream. Well it’s 64kb but some I guess its used for other parameters. Maybe use a larger ARM or do a rolling FIFO read or write to the SD card so that less RAM is needed. The SD card should be fast enough as the data is coming from the sampler at 100khz
3. The format of the SD would need to hold a flat 64kb rather than tracks and sectors. I assume the source code can be modified to do this.

Any thoughts or views? I will scope out the QD drive handling IO but it does map to Shugart with some logic inversion. I have S220s and S10 to play with.

keirf, 2019-08-07:
I'm on holidays at the moment but I envisage mapping to a single long track hfe file. What's needed is some description of behaviour of the signal ios. Eg via inspection on logic analyser.

pacific202, 2019-08-07:
I should be able to hook up a logic analyzer to my S-220 next week.

AMSynths, 2019-08-07:
I will also take some scope measurements of the IO tomorrow and feedback. IO lines map except READ DATA which is not inverted on QD.

Have a great hols!!

keirf, 2019-08-07:
I think some qd also have a home/head signal indicating when read write is possible? Qd signals seem non standardised?

AMSynths, 2019-08-08:
Here is the Roland Read and Write timing 

[image]

keirf, 2019-08-14:
I'm going to aim for a qd firmware by mid September (I have little time to work on anything through rest of August).

AMSynths, 2019-08-14:
Hiya
Thanks for the update! Busy times....and many thanks for taking the project forward.
I will practice flashing the gotek drive with the current software.

Best Regards
Rob Keeble
AMSynths
www.amsynths.co.uk

keirf, 2019-10-02:
Ok I have QD spec'ed out and it is next to implement now On Screen Display is done. Hopefully firmware to test within a week.

AMSynths, 2019-10-02:
Hi
Good news!

Best Regards
Rob Keeble
AMSynths
www.amsynths.co.uk

keirf, 2019-10-08:
Here is a firmware to try. You should update using the UPD file in **alt/quickdisk**. There is also a blank QD image file for you to use, if you don't have any existing QD images.

Wire up as follows:
* Pin 2 = Media Sense (/MS)
* Pin 16 = Motor (/MO)
* Pin 20 = Reset (/RS)
* Pin 22 = Write Data (/WD)
* Pin 24 = Write Gate (WG)
* Pin 28 = Write Protect (WP)
* Pin 30 = Read Data (RD)
* Pin 34 = Ready (/RY)
* Odd pins (1,3,...) = May connect to GND. Probably not necessary.

You should also place a jumper at **MO** only. Optionally you may test with a jumper at **JC**: This changes the behaviour of the Ready line slightly.

[ff_qd_1.zip](https://github.com/keirf/FlashFloppy/files/3701947/ff_qd_1.zip)

keirf, 2019-10-09:
Here's a build with some QD tweaks and cleanups. Please test this one, but it should perform just the same as the previous. However this is a candidate for checkin to main branch for next release :)

[ff_qd_2.zip](https://github.com/keirf/FlashFloppy/files/3708803/ff_qd_2.zip)

AMSynths, 2019-10-09:
Hi
Many thanks for the new firmware, I will test on friday and report back!

Best Regards
Rob Keeble
AMSynths
www.amsynths.co.uk

keirf, 2019-10-10:
Here's a build that includes a "log to USB" build of the QD firmware, in alt/quickdisk/logfile. See https://github.com/keirf/FlashFloppy/wiki/Logging-To-USB-Drive for instructions on use. Note that the logfile firmware says "LOG" on the display at startup, **not** "QD". 

Only use the logfile qd firmware if explicitly asked to gather logging info for debug purposes.

[ff_qd_3.zip](https://github.com/keirf/FlashFloppy/files/3714569/ff_qd_3.zip)

keirf, 2019-10-13:
This is now supported in release v3.6a. In the not unlikely event of problems (reads have been seen working; writes not yet confirmed) please open a new bug issue and I will assist in fault finding. This ticket now closed.

closed 2019-10-13 by keirf