keirf/flashfloppy #207: Does ff support interleave?

alank2, 2019-03-04:
I'm using FF with a Zeta 2 SBC (Z80/CPM2.2).  The SBC has its own format utility and when I try it with a real floppy drive, it seems to need an interleave of 2 for the best performance - test application load time:

floppy [formatted 1:1 interleave] = 15.64 seconds
floppy [formatted 2:1 interleave] = 3.50 seconds

gotek/ff [formatted 1:1 interleave] = 16.06 seconds
gotek/ff [formatted 2:1 interleave] = 15.57 seconds

FF is in IBM PC mode using 1.44MB floppy images [DSKA0101.IMG, 1474560 bytes].  I am presuming that the file is raw without any interleaving present and that it isn't translated/emulated into the image file itself.

Does the Gotek try to wait for the proper rotational delay before responding, or if sector X is asked for, does it delivery it immediately?  I don't think my FF.CFG has any customized settings.

keirf, 2019-03-05:
IMG files are straightforward data dumps -- it is up to FlashFloppy to infer track geometry: sector sizes, numbering, interleave, etc. Normal PC HD disks are not interleaved, hence even if you try formatting one 2:1 that information is lost and the track reads back 1:1.

I could add a special host type here, but it seems there are many different CPM track formats. The better fix is probably in ticket #121

Simone2018, 2019-03-05:
+1 for description of file format in filename like in ticket #121
i suggest "diskname.[options].img" : img extensione make the image direclty work with i.e. winimage...
[options] can override default disk format:
-HD: high density format (indipendent from image size)
-DD: double density
-RPM360
-SS1024: Sector size 1024
-BR250: Bitrate 250.000

if unspecified an option stay in default for that format.
etc etc

alank2, 2019-03-05:
What do you compile it with?  True Studio?  I found where the formats were and the 1.44mb format.  If I change the interleave in that table from 1 to 2 and recompile it, might it work?  It is just a one off project, but it would be cool to get it working fast.  Thanks!

keirf, 2019-03-05:
Yes that change would work. Build instructions (you want an Ubuntu Linux VM preferably): https://github.com/keirf/FlashFloppy/wiki/Building-From-Source

alank2, 2019-03-05:
Thanks keirf; I'll find a Ubuntu Linux VM and see if i can make a build!

alank2, 2019-03-05:
That worked perfectly!  So much nicer to do a USB stick firmware update too.

Might be nice to have a FF.CFG option to override the interleave if it isn't a statically compiled thing that can be done easily.

keirf, 2019-03-05:
That's really what #121 is all about, I'm not sure whether to do it within the image filename, or via config files, though. I think perhaps the latter.

@Simone2018 are you suggesting there is a standard of sorts already for specifying disk layout within the image file name?

alank2, 2019-03-05:
I can see both sides of the coin.

Putting an interleave override in FF.CFG pros:
  If interleave is desired on one disk for a system, it is most likely good for many.
  Compatible with DSKA0001.IMG naming.
  Would it override ALL disk formats, or could it override just specific ones?

Putting an interleave override in the filename pros:
  You can have multiple images each with their own interleave settings.

You could also make a file specific cfg like IMAGENAME.CFG such as DSKA0001.CFG that contains image specific options such as interleave, read only, disk format override, something else, etc.

Simone2018, 2019-03-05:
@keirf, no sadly no standard, but is quite intuitive IMHO. 
I'll prefer file name appending vs FF.CFG because most of the time I take new FF.CFG from the update to have all new options, and I'll lose floppy settings. And different USB stick may have different FF.CFG so is sometime confusing...

very interesting the "IMAGENAME.CFG" option, (but not standard also here...). - it open the way also to advanced options like weak byte, track shuffled etc etc. 
You can take the same option that are in Imagedisk header: http://www.classiccmp.org/dunfield/img/index.htm
or implement IMD/TD0 format maybe (is difficult to write to because they are compressed)

Any option you choose will be a great improvment!

alank2, 2019-03-05:
I love the raw image format personally; it is excellent.  I think configuration info embedded in the filename is better than a non-standard file, though I still like the FF.CFG option or IMAGENAME.CFG option.

keirf, 2019-03-06:
Ok I think some combo of track parameters in a config file, coupled with tagged filenames, might be a winner. Eg imagename.tagname.img where tagname gets resolved in a formats.cfg global file.

I will link this ticket in #121 and close.

keirf, 2019-03-12:
See the test firmware uploaded to #121.

alank2, 2019-03-12:
ALSO - having a new issue.  I have GoTek as C: and a real floppy as D: on the Zeta 2 SBC running CP/M 2.2.  I can access easy drive just fine individually, but if I try to copy files from one to another, using them quickly with each other presents an issue:  If I access C:, it seems to take a long time for its LED to go out.  If I try to access D: before waiting for C: GoTek light to go out, the GoTek also responds even though it is not the correct drive.  I've tried it with and without the JC jumper.  I repeated the test the other way around to see if I could get the real floppy light to stay on and then access the GoTek, but that works as expected.  Any ideas?

keirf, 2019-03-12:
How is the gotek jumpered? Perhaps you jumpered the motor signal instead of correct drive select pin?

alank2, 2019-03-12:
S0 and JC are jumpered.  Using a cable that is straight through on both connectors.  Would it work if I took the cable apart and did the flip and went back to S1 on the drive?

keirf, 2019-03-13:
No don't bother doing that, I don't see this being a cable issue.

The drive light in this case is the Gotek green LED right? This indicates that the host has not turned off the Gotek's drive-select line. The LED is directly connected to that line. And it would therefore make sense that the Gotek responds to the D: requests, as the Gotek is still selected!

So I wonder why the machine is so slow to deselect the Gotek -- does it also happen when the Gotek is D:?

I'll reopen this ticket a little while just to work through this issue if we can.

alank2, 2019-03-13:
Thanks keirf - problem is now solved.  I think the jumpering of the GoTek design is a bit odd.  It turns out that with a straight cable and "s0", the motor signal is being used and not the select signal.  The signal that I wanted is actually on pin 14, but the GoTek PCB doesn't do anything with pin 14 and has no way to connect it for a straight through cable.  I took the cable apart, split and twisted it, went back to S1 and it works perfect.  Thanks for the help; I appreciate it!

closed 2019-03-14 by keirf