keirf/flashfloppy #209: Support for TRS-80 I/III/IV formats

alnwlsn, 2019-03-10:
Greetings,

Would it be possible to support the formats of the TRS-80 grey computers? There are three formats used by most emulators:

.JV1 - Model 1 FM encoded disks
.JV3 - Model 1/3/4 FM or MFM encoded disks (or a mixture, which is needed by the Model 1)
.DMK - More complicated format that can emulate everything a real floppy controller can do.

Tim Mann's site has documentation - http://www.tim-mann.org/trs80/dskspec.html

Annoyingly, most of the emulators use the .DSK file extension to refer to all 3 file types, but most can also accept the specific file extension also. Of the three formats, it seems like JV3 is the most common for normal disks, with DMK used for copy protected ones. I believe that the two JV formats are also similar to the JVC CoCo format (all 3 were written by the same author).

I will note that converting the images to HFE and using them that way works perfectly with my Model 1, but it would be great to see native support.

Thanks again for this great emulator!

keirf, 2019-03-12:
I support JVC already. JV1 looks very easy but probably isn't very common? JV3 is quite a bit more complex than JV1 and JVC. I'm not realistically sure I will get around to supporting it, really. I have other demands (eg. Atari ST STX image support) which would have more users.

lrowen, 2020-08-05:
Seeing this older issue here, and knowing that the main archives for the TRS-80 Models I/III/4 series use DMK as the 'gold standard' for the TRS-80's, I'll second this request.  While I understand that other formats may be more popular or for more interesting computers, there is a large archive out there already in DMK, and it would be very helpful to the TRS-80 community.

I'll also echo what was said in the OP that I appreciate this emulator; and if DMK support isn't too difficult, it would be very appreciated.

dittman1, 2020-08-05:
I would like to see DMK support as well as so much of the archived software for the TRS-80 family is in DMK and not having to keep two copies (DMK and HFE) would be useful.

DMK is also used in the MSX community.