keirf/flashfloppy #210: ICE FELIX HC2000 support (ZX Spectrum clone)

SilverGreen93, 2019-03-10:
Hi,
I just flashed the FlashFloppy firmware on my new Gotek and I want to thank for this great upgrade for my Gotek. I tested it with my Amstrad and IBM PCs and it works a treat!

Here in Romania, we have a ZX Spectrum computer clone (HC2000) which uses a CP/M disk format with some custom modifications to store ZX Spectrum programs on disk rather than tape.
The computer can run in 2 modes: CP/M and Spectrum BASIC. CP/M disks are read fine and work great from FlashFloppy, but BASIC disks (based on CP/M format) don't work.

The disk format is described here: https://sites.google.com/site/georgechirtoaca/documentation/hcfsv1.txt?attredirects=0
You can see section 4 (especially 4.1) for the complete specification of the BASIC disk format.
The main difference would be the fact that it uses 80 tracks x 2 heads x **16 sect/track** x **256 B/sect** = 655360 B

Also, a sample disk image can be downloaded here: https://sites.google.com/site/georgechirtoaca/software/BASICDEMO.ZIP?attredirects=0

Let me know if I can help in providing more info than this.
If you point me in the right dirrection in the FlashFloppy code, I might try myself to add support and test.

Thanks a lot,
Mihai

keirf, 2019-03-12:
These varied CP/M formats will be solved by having "self-describing" images. Essentially you will tag your image files like eg. DEMO.CPM.IMG and the tag "CPM" will be looked up in a config file to find the track geometry: C-H-S and so on. I am currently working on this. It is tracked in #121

keirf, 2019-03-12:
Since the solution is tracked in another ticket (#121) I will close this ticket.

closed 2019-03-12 by keirf

keirf, 2019-03-12:
See the test firmware uploaded to #121