keirf/flashfloppy #211: 2.0 improvement

Simone2018, 2019-03-15:
It's Amazing... I tested succesfully 16Mb Floppy Disk with it!
Can you unlock maximum number of sectors from 64 to 128 ?
I have reached 32768 sectors... the floppy FAT12 BPB has a maximum of FFFFh sectors so I want to reach the theorical maximum (32Mb floppy!)
Thank you

keirf, 2019-03-15:
Is there scope for increasing physical sector size to say 4kb? If your bios can handle more sectors per track could it also handle larger sectors and grow space that way?

Simone2018, 2019-03-15:
I'll try larger Sector size but i.e. Winimage support 512Kb sectors only. And I suppose MSDOS, IO.SYS etc etc all support 512Kb Sector only... so increasing sectors (and maybe sides more than 2) could be win-win. 32Mb is the FAT12 structural limit (but MBR could support more than that). If it's too difficult don't lose too much time.

keirf, 2019-03-15:
Increasing sides would be a bit tricky as there is only one head-select pin ;)

keirf, 2019-03-15:
Increasing max sectors yes not really that hard. Surprised DOS is flexible on that but not physical sector size? I suppose old hard drives had 512b sectors too?

Simone2018, 2019-03-15:
Yes, all (floppy, Hard Disk) was based on 512b sectors... buffers, minial data read, boot sequence, MBR, etc etc. Maybe old dos (<2.1) was less fixed on this.
Sector number is flexible because Hard Disk geometry was using it. To allow as much space I had to install 2M-XBIOS for floppy (it support also HD floppy but not so my controller)...

keirf, 2019-03-15:
I could even increase the limit to 256 sectors, I suppose.

Simone2018, 2019-03-18:
Talking for IBM-PC view:
"for disk access with the BIOS of IBM-PC compatible machines, the sector number was encoded in six bits, resulting in a maximal number of 111111 (63) sectors per track"
but 256 is welcome! :) thank you

keirf, 2019-03-18:
If there's a maximum support of 64 sectors in the IBM-PC BIOS, why is there benefit to increasing at all beyond the current 64?

Simone2018, 2019-03-18:
only for NON-IBM i think; 
but I can also write a new INT13h function to take it in account...

tlo2357, 2019-03-19:
The WD 179x controllers have 8-bit registers for the track and sector numbers.  I haven't seen anything in the datasheet stating that the full range isn't available.

I expect that reading unusual track or sector numbers in the ID field using a 179x should work but writing some of them (during format) might produce incorrect results because they might be interpreted as format control bytes, but a drive emulator shouldn't care about that.  If I can find the time and it hasn't already been done somewhere it *would* be interesting to test and document this with my Z-100 which uses a 1797.

keirf, 2019-03-26:
This is in the master branch now (pending v2.1a)

closed 2019-03-26 by keirf

tlo2357, 2019-03-28:
Some may be curious about the test results.  PC controllers *do* support track and sector numbers ranging from 0 through 255, although I suppose some wonky Super I/O chips might not – similar to how they don't all support 128-byte sectors.

I ran tests on my old test PC which has an ITE IT8712F-S Super I/O chip, which Linux identifies as a "post-1991 82077" controller.  I used fdrawcmd to format track zero of a real disk with different track and sector ID number combinations to test what would be accepted by the controller's read sector command.  These tests showed that all track and sector values worked.

To spot-check that the controller was actually writing to disk what it was supposed to, not truncating anything, I crafted a test and read the MFM data from the track using a Raspberry Pi.  All of the track and sector ID values were as expected, including 0 and 255.