keirf/flashfloppy #213: Problem with v2.0a FF reading sectors

staticmem, 2019-03-17:
Doing some testing with the new IMG.CFG and FF Version 2.0a I'm having issues not only with the new RAW image method but also DSK images that had worked on earlier FF release.  This is my disk definition:

`
[ds40]
cyls  = 40
heads = 2
secs  = 10
bps   = 512
id    = 1
mode  = mfm
`

Selecting either the RAW or DSK image results in the same problem.

I have 2 programs I use to access the disk images on the Gotek.  They both produce the same error.  I use a Floppyio hardware interface that the Gotek is connected to.  I can normally boot Microbee DS40 disk images via emulation but now can not.

Here is my testing example...

This will copy the disk image to a local disk image as a DSK image.

`
$ ubeedisk fiorb --disk=copy --format=ds40 --of=test1.dsk

(The "fiorb" is a macro that simply tells ubeedisk to use the Gotek drive as the input disk to save me some typing)

Internal format selected:
ds40: Microbee DS DD 40T 5.25" (SBC)
ubeedisk: Destination file already exists. Overwrite "test1.dsk" ? [N]o [Y]es: y

Enter a disk description.  A '.' on a line by itself ends input.
A '*' by itself quits (no disk copy made). Maximum 100 lines
allowed. (Cut and paste from a text editor to make this easier)

.

Copying disk/image 'serial:/dev/ttyACM0,19200-crtscts,b:' to 'test1.dsk'

Copying data (with format):
Cylinder: 01/39 Head: 0/1 Sector: (log) 000/009, (phy) 002/010
ubeedisk: read_sector() - Missing address mark.

INTERACTIVE ERROR MENU
----------------------
re[T]ry (no home) [R]etry (home) [I]gnore [S]kip [A]bort [C]lean [M]ode ?:
`

One thing I have noticed is that the Gotek display shows "T:02:0" when accessing Cylinder 1 as can be seen from above is not accessing.  Why is it showing cylinder 2?

If I select the DSK instead of the raw image it comes back with the same error.

It could be because Microbee disks are pushing the boundary limits with 10 sector/tracks could be causing this even though it had worked before?

keirf, 2019-03-17:
DSK is interpreted just the same as IMG/IMA unless it is a CPC-style DSK or EDSK image.

Can you try both tagged and untagged IMG names? For example "my.ds40.img" and "my.img". Be sure you do not have a [default] tag in your IMG.CFG! This is an interesting test because the untagged code path should behave exactly as it used to.

Also can you try with FF v1.1?

staticmem, 2019-03-17:
The DSK images are CPC-style disk images that I'm using.

I have again tried tagged and untagged IMG names with FF v1.1.  I can confirm there is no [default] tag in the IMG.CFG file.  This also fails in both cases.

So for DSK images at least there seems to be something broken somewhere along the line.

flashfloppy_v0.9.24a.zip is the last FW I had installed I think which works for the DSK images.  I just re-flashed that version and it now works for CPC DSK images.

Hopefully you can see where it became broken.  If you want I can try other versions.

keirf, 2019-03-17:
0.9.24 is a long time ago. Could you try intermediate versions and work out which one caused your problem? I did a lot of work on dsk after 0.9.24

staticmem, 2019-03-17:
ok, will do.  quite a few there so I will use the divide by 2 method to narrow it down.  Expect result tomorrow though as we are located in different places of the world.

keirf, 2019-03-17:
And to be clear raw is not a regression? You were (reasonably) hoping that you could get img/ima working via new img.cfg?

staticmem, 2019-03-17:
Correct, I know the older releases did not support the new raw IMG format without the new IMG.CFG code.  I will only be testing to see where the CPC DSK images became broken in the release chain.

staticmem, 2019-03-17:
I'm seeing inconsistent results even in that older 0.9.24 version I claimed was working,  now I can't repeat it.  I'm starting to suspect there is some other HW issue happening here.

keirf, 2019-03-17:
Can you attach one of your working (on older flashfloppy) dsk images for me to take a look too?

staticmem, 2019-03-26:
I did send you 2 disk images via your Github email account.  Apart from that I have not had a chance to look any further at this stage because of other issues.

keirf, 2019-03-26:
I don't think I ever received them. You can zip them and attach them to this ticket.

staticmem, 2019-03-26:
I've sent them to your email address (I got email address wrong before)

keirf, 2019-03-26:
Got them now

keirf, 2019-03-26:
The DSK and IMG (using your IMG.CFG) seem to match in layout and I would expect them to work fine. Both mount just fine in latest v2.x FlashFloppy and firmware seems to interpret them correctly.

staticmem, 2019-03-26:
ok, thanks.  I will look into it further this side and possibly it's an issue with my hardware accessing them

keirf, 2019-03-26:
I will close for now. Please reopen if necessary.

closed 2019-03-26 by keirf

staticmem, 2019-10-05:
I finally managed to get back to this problem.

> Copying data (with format):
Cylinder: 01/39 Head: 0/1 Sector: (log) 000/009, (phy) 002/010
ubeedisk: read_sector() - Missing address mark.
> 

This was my own fault.  I had not configured the Floppyio unit correctly and had specified an 800k drive when it's only a 400k disk.  The result was that the Floppyio FW switched to "double stepping" mode and why it failed as shown above.

I have the RAW image working now and using FF v3.2a.

Now I just need to test a DS80 format disk as a RAW image to see if that's working.