keirf/flashfloppy #215: Amiga 500 - Gotek w/ FF 0.11 not working

shan2752, 2019-03-23:
Hi,

I narrowed down the issue to the DF0: floppy drive cable.  When the floppy drive cable is unplugged, the Gotek will boot as DF0: with my ACA500 Plus.  Once I connect the real DF0: floppy drive cable, the Gotek will not boot.  Floppy disk games work fine so it doesn’t appear to be a cable issue.  Any ideas?  I’m truly stumped here.  Thanks!

_Originally posted by @shan2752 in https://github.com/keirf/FlashFloppy/issues/184#issuecomment-475892314_

keirf, 2019-03-23:
What's the exact setup? You say Gotek boots as DF0, but DF0 cable is unplugged??

shan2752, 2019-03-23:
I have the Gotek as my exterior drive.  I have a regular internal floppy drive.  I have an ACA500Plus that will switch the exterior drive to act as df0 without an internal boot selector.  When the interior floppy is disconnected, Flashfloppy works fine on the exterior Gotek as df0:.  When I plug in the internal floppy drive the Gotek won’t boot either the selector menu or any adf’s.  The internal floppy drive boots fine when connected as df0.

I had the same problem before with an internal df0 boot selector, same result.  As soon as I connect the internal floppy drive, it stops working.  Any thoughts?  Thanks!

keirf, 2019-03-26:
The problem is probably that internal drive is a hacked PC drive, with pins 30 and 34 connected by a wire (check the underside of that floppy drive for modifications around the 34-pin header!).

keirf, 2019-03-28:
Closing for now, reopen if diagnosis is incorrect.

closed 2019-03-28 by keirf