keirf/flashfloppy #216: Current track on 7-segment display

matburton, 2019-03-23:
Just a suggested feature ;-)

It might be nice to have an option to display the current track being read on the 7-segment display. i.e. somewhat like WinUAE does. It'd be a nice visual display of activity, especially for those without the sound mod.

keirf, 2019-03-25:
I'll have a look

Simone2018, 2019-03-25:
Consider also showing tracks >99 in LCD display if you can :)

keirf, 2019-03-25:
99+ tracks doesn't really fit on a 16 char display but could do that for oled display with default small font?

Simone2018, 2019-03-25:
yes, OLED, pardon me

matburton, 2019-03-25:
So as a test I added a call to `led_7seg_write_decimal(drv->cyl);` [here](https://github.com/keirf/FlashFloppy/blob/df02509547441aeed9453bd4febc7c372cb01afa/src/floppy.c#L942) in the `STEP_latched` case of `drive_step_timer`, built and took the `.upd` to my Gotek.

That had the desired visual effect. I tried a `.adf` game (Super Skidmarks) which worked fine and then a `.hfe` (v3) game from a `.ipf` (Zool 2) and that also seemed to work correctly.

I did however also try a couple of auto-swap games (Beneath a Steel Sky and Monkey Island 1) but neither would load :disappointed:

keirf, 2019-03-25:
I have a patch just about ready to go. It may be released tomorrow.

matburton, 2019-03-25:
Woops! I probably cloned master with [your commit](https://github.com/keirf/FlashFloppy/commit/df02509547441aeed9453bd4febc7c372cb01afa) then which is probably why my 'bodge' worked! :nerd_face: 

Will revert and test the auto-swap games again :+1:

keirf, 2019-03-25:
I expect your patch worked okay, as mine needs to be explicitly enabled in FF.CFG. Not sure why autoswap games aren't working. Have they ever worked for you?

matburton, 2019-03-25:
Yes they previously worked, I re-tried Beneath a Steel Sky and Monkey Island 1 as I've previously had those working with the [last release](https://github.com/keirf/FlashFloppy/releases/tag/v2.0a).

I assume I just need a `FF.CFG` with nothing but the line `display-type = auto-trk` within. If so I'll give them another try now.

matburton, 2019-03-25:
OK I re-built from master after reverting and applied the `.upd` to my Gotek.

I didn't create a `FF.CFG` because I'm using `HxC-compat-mode`. However the 7-segment display seems to be showing track changes anyway. I assume therefore it's intentional it's the new default behaviour? (fine by me!)

I then tried the auto-swap versions of Beneath a Steel Sky and Monkey Island 1 and both loaded correctly.

I notice the tracks numbers go high, above the 80-84 I was expecting to see. Is that because it's also showing which side is being accessed? So 156 means cylinder 78 side 0 and 157 would mean side 1?

Also noted I can no longer press both buttons simultaneously to re-mount  `AUTOBOOT.HFE`. I can however press the left button to move to index 0 which has the same effect.

keirf, 2019-03-26:
Showing track changes should not be the default, I will double check that.

Yes shows track number not cylinder number. Do you have a preference?

Currently you press a button to reveal the current slot number, but that button press is then consumed. You should then be able to press both buttons to jump to slot zero, while the current slot number is displayed. This is the same logic as switching on the lcd back light or oled display. The alternative is of course to respect all button presses and treat even that first one as a navigation event. What do you think?

keirf, 2019-03-26:
That's great then. It's the usual interleaved track numbering: track = cylinder*2 + side

closed 2019-03-26 by keirf