keirf/flashfloppy #217: ATR -image support request (Atari floppy disk image format)

tenuke, 2019-03-26:
[http://fileformats.archiveteam.org/wiki/ATR](url)

keirf, 2019-03-26:
Any source of ATR images for reference?

v-retro, 2019-03-26:
There is a collection here, Keir:
http://www.mushca.com/f/atari/

On that page, if you click on the blue disk icons to the left, it will
download a Zip file containing an ATR file.

keirf, 2019-03-28:
```
Density              SD   ED   DD   QD
---------------------------------------
FDC mode             FM  MFM  MFM  MFM
Disk sides            1    1    1    2
Number of tracks     40   40   40   40
Sectors per track    18   26   18   18
Bytes per sector    128  128  256  256
---------------------------------------
Disk space (kB)      90  130  180  360
```

keirf, 2019-03-28:
Because track 0 first 3 sectors may be different size than the rest, requires me to support mixed-sec-size tracks at last. So I need to put that in then ATR parsing (which should be trivial).

keirf, 2019-04-05:
Supported in v2.2a. Please test and raise a new ticket for any issues.

keirf, 2019-04-05:
Reopening pending more tests. Can anyone test some ATR images off the net, via standard Atari floppy to SIO interface, and see if they load?

keirf, 2019-04-05:
Here is a possible fix firmware, please try it if you are unable to load ATR images from your collection or downloaded off the internet:
[ff_217_1.zip](https://github.com/keirf/FlashFloppy/files/3049085/ff_217_1.zip)

keirf, 2019-04-08:
No responses, so I have committed the above fix into v2.3a. Reopen if this raises issues for anyone.

keirf, 2019-04-19:
Temporarily reopened in light of #223 -- please can you test the new firmware posted on that ticket, as it will appear in the next 2.x FlashFloppy release unless there are negative bug reports against it.

It is expected to speed up disk transfers by approx 2x.

keirf, 2019-04-23:
Improvements in v2.7a

closed 2019-04-23 by keirf