keirf/flashfloppy #218: Bubble Bobble does not work with external Gotek set to DF0

Scott-Pearce, 2019-03-29:
Hi,

I have a Gotek flashed with latest FlashFloppy for use with an Amiga 500 (rev6a). When the Gotek is connected internally, Bubble Bobble works.

If I connect the Gotek externally (using an amiga external drive case), and use a boot selector (https://sordan.ie/product/530/boot-selector-df0df1-df1df0-signal-sel0-from-sel1-amiga-500/) to set the external Gotek to be df0, Bubble Bobble does not work. It loads to the splash screen, then stops.

If I connect the Gotek externally, set it to be df1, turn on the Amiga, then flick the selector switch so that the Gotek is now df0, boot Bubble Bobble, it works. It will continue to work after warm boots until the next cold boot.

I've tested using FlashFloppy 1.0 and 2.1a (downloaded today).

I have tried many different Bubble Bobble adf images with the same results.

I've had someone try this using the HxC firmware and Bubble Bobble worked on their setup. They got their boot selector switch from AmigaStore.eu.

Any info regarding this matter will be appreciated.

Scott

Scott-Pearce, 2019-03-29:
Further testing. 

This only occurs if the system is started with a disk image "ejected" on startup. I had ejected-on-startup = yes. Setting this back to no in FF.cfg, resetting the onboard config, rebooting, "inserting" the Bubble Bobble disk image, rebooting again - now works every time.

I guess this may be by-design or because of some esoteric hardware limitation of the Amiga 500. In any case, hopefully this information will help others who have this problem to work around it (i.e. don't set ejected-on-startup = yes!)

keirf, 2019-03-29:
I wonder what happens if the gotek is connected to the amiga df0 header instead of the real floppy drive?

Or, what happens if you leave the gotek external, selected as df0, and disconnect the real floppy drive?

Another thing if you like ejected on startup is to try `interface = amiga` in FF.CFG.

Scott-Pearce, 2019-03-29:
>I wonder what happens if the gotek is connected to the amiga df0 header instead of the real floppy drive?
I will test this.

>what happens if you leave the gotek external, selected as df0, and disconnect the real floppy drive?
The selector switch essentially does this. When the gotek external is set to be df0 (via the selector switch), the internal floppy drive is actually disabled and not usable. Would it be worth *physically disconnecting* the real floppy drive as a test?

>if you like ejected on startup is to try interface = amiga in FF.CFG.
Tried that. No change, issue still occurred.

keirf, 2019-03-29:
Yes I meant physical disconnection.

I don't have a boot selector so can't reproduce your exact setup. I think aca500plus faithfully emulates one though so I could try that.

keirf, 2019-03-29:
Tried with A500 Rev 6A, Kickstart 1.3, Gotek as internal drive, DF0. Couldn't reproduce.
I noted that the game hangs at the title screen using Kickstart 3.1.4 (not very surprising).

Scott-Pearce, 2019-03-30:
> I wonder what happens if the gotek is connected to the amiga df0 header instead of the real floppy drive?

No change, issue still occurred. You say that this config worked for you! Please point me to the adf you used, as well as your ff.cfg and I’ll test further.

> what happens if you leave the gotek external, selected as df0, and disconnect the real floppy drive?

No change, issue still occurred.

keirf, 2019-03-30:
I'll attach it here in a short while. To be fair it didn't work every time even with kickstart 1.3 but that seemed unrelated to the gotek. At the end of the title screen loading stage the game would sometimes hang, with minor onscreen gfx corruption. But it did load a couple of times okay from ejected on startup.

keirf, 2019-03-30:
I wonder if the game loader is buggy and depending on startup timing or something you happen to systematically tickle that bug. It would be interesting if you can pin blame on ejected-on-startup for failure of other titles.

keirf, 2019-04-02:
Here is the ADF I used. Only thing in my FF.CFG was `ejected-on-startup = yes`.

[bb.zip](https://github.com/keirf/FlashFloppy/files/3032829/bb.zip)

keirf, 2019-04-04:
Could you write ADF to real floppy and try booting your machine with just real DF0 connected. Then insert the real floppy and see if that works?

keirf, 2019-04-05:
I have nothing more to add here. Closing, but reopen if your testing raises any more interesting info, or answers to my questions.

closed 2019-04-05 by keirf