keirf/flashfloppy #219: Auto detection of SH1106?

apolkosnik, 2019-04-05:
Hi, first and foremost, thank you for working on this project!

I was wondering if there's a simple way to detect SH1106? Or at least perhaps add a note that if you see 1 line and garbage below, to switch to SH1106 in FF.CFG.
Recently I got couple of OLEDs that were supposed to be SSD1306, but the display turned out garbled just as seen on https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/228
After copying FF.CFG and setting 
```display-type = oled-128x32-sh1106```
everything started looking great.

Thanks,
-A

keirf, 2019-04-06:
Have you tried v2.2a firmware? I adjusted horizontal offset for 128x32 sh1106 displays as a user had a 2 pixel band at the left and display was shifted.

I need to know if this offset needs to be configurable...

keirf, 2019-04-06:
To your original question, the displays do not have a command set that allows them to easily be probed, but they do have ram/status read commands and the different addressing methods and ram sizes I'm sure leads to some tricksy detection method.

If the SH1106 horizontal offset does vary, that will require manual configuration, since it is a property of how the display is wired to the controller: it is fundamentally not probeable. But at least you would automatically get a display that basically works except perhaps for a few pixels offset that can be corrected later via FF.CFG.

keirf, 2019-04-08:
Hi, any comment on whether you have tried fw v2.2a and whether display looks correct on that versus v2.1a? Cheers!

keirf, 2019-04-08:
Auto-detection is now in v2.3a. If the display has a vertical bar at left or right hand side then please see the FAQ: https://github.com/keirf/FlashFloppy/wiki/Troubleshooting-FAQ#oled-displays

closed 2019-04-08 by keirf

apolkosnik, 2019-04-08:
Thanks, 2.2a has clean display, the stable had a vertical bar when I tested last week. I'll grab 2.3a and let you know if anything is wrong. Thanks!

keirf, 2019-04-08:
Thanks, that's great and as I hoped. I will backport the fix for 128x32 SH1106 displays (to avoid the vertical bar) to stable 1.x series.

apolkosnik, 2019-04-08:
I've updated two to 2.3a, and the display looks good after config reset. Thanks!