keirf/flashfloppy #220: Acorn DFS 1770 controller 2.2a and above

MadeGray, 2019-04-08:
After upgrading from 2.1a to either 2.2a or 2.3a access on BBC Model B (1770 DFS) now fails using a known working SSD file. FF.CFG remains the same across all tests. After upgrading to 2.2a or 2.3a a Shift-Boot shows disk cycling between track 0 and 1 and after a few seconds disc error 18 appears 00-01. A clean boot and doing a *. causes the machine to hang for a few seconds (can hear a tick on the gotek speaker) and the disc error 18 appears 00-00. No IMG.CFG file has been created and the FF.CFG remains the same. All current release versions 1.x up to 2.1a work fine. 
Attached ff.cfg file and simple ssd image that works up to and including 
[demo.zip](https://github.com/keirf/FlashFloppy/files/3054959/demo.zip)
2.1a  happy to test out any requests etc.  thanks

keirf, 2019-04-08:
Thanks I will investigate and come back with further questions if necessary. If you have time to test a dsd that would be interesting. No rush.

MadeGray, 2019-04-08:
Will do, I’ll check that this evening and hopefully adfs as well. Just need to dig out the master and set up

keirf, 2019-04-08:
Yes actually I meant to say adfs. Dsd not so interesting as it's also single density. Double density more interesting.

MadeGray, 2019-04-08:
Adfs on a master - a few broken directory errors but I then unplugged beebscsi. Loading various adfs images such as welcome disk etc no issue at all on 2.3a all load fine and run. Dfs on same master fails loading working ssd or dsd images.

MadeGray, 2019-04-08:
I’ve tested approx 10 adfs images on my master and all load and run no problem. Think the broken directory errors was me forgetting how to move from beebscsi to the gotek. Disabled beeb scsi. Shift boot and cat etc all work fine on adfs on all the samples I’ve tried. Same master moved to dfs gives same error as the model b does on ssd and dsd. Hope this helps!

MadeGray, 2019-04-08:
Just to confirm, I’ve done a full verify on adfs gotek and copied the full master welcome disk adfs gotek to beebscsi with no issues so I’m pretty sure adfs is fine.

drdpj, 2019-04-08:
Sounds like the fm support is borked. Someone said yesterday it wasn't
working with 8271 either... (John, with beebscsi connected, floppy is drive
4 in adfs). I haven't had a chance to check myself...

MadeGray, 2019-04-08:
Thanks Daniel, I finally remembered and did a full copy of the adfs master disk of the gotek drive 4 to beebscsi drive 0 to a subdir welcome and all worked fine so I’m convinced adfs is all good. Dfs Not as happy. I had to count up from 0 on fingers to bloody remember lol

keirf, 2019-04-08:
Thanks, I think the FM flag isn't plumbed through. I will post a test firmware later.

No need for any more tests for now please.

keirf, 2019-04-08:
Here is a firmware to test. A couple of flags were not plumbed from the handler frontends (SSD/DSD/etc) to the track-definition structures. This included the is_FM flag. As well as confirming that DFS (SSD/DSD) works, it would be handy if you could have a go at disk writes, both in DFS and in ADFS (ie. both single- and double-density). The write code got modified quite a bit and hasn't really been tested at all so far.

[ff_220_1.zip](https://github.com/keirf/FlashFloppy/files/3056221/ff_220_1.zip)

MadeGray, 2019-04-08:
It’s much much closer catalogue now works but data is getting scrambled when loading apps etc. Loading standard apps such as the code demo it scrambles the loaded data and can’t list or run the programs. Get the infamous Bad Program and *type or list a readme file displays some text and then lots of random teletext characters. It’s very very close almost as if some bits being swapped somewhere and corruption occurred. Created a 10 print “hello world’ saved and when loading gives bad program. Loaded the ssd I just created into beebem and the program I saved hello world is absolutely fine so saving looks good - loading programs is failing. Tested on master and model b loading both fail same way. Tested the files I saved on gotek with this version fail to load on gotek but DO load on beebem. So saving and cat work fine on gotek. Loading is not quite there.

MadeGray, 2019-04-08:
Just read that again and didn’t make sense to me lol.so - catalogue works fine on beeb and master on ssd etc. Loading programs doesn’t work. They get corrupted when being loaded so either give bad program etc or another error. Created a simple program hello world and saved to gotek. Loading the saved program on gotek gives bad program error. Loaded the ssd into beeb em and the saved program loads fine. Basic testing suggests very close but loads are being corrupted. Saving files works fine as can load sample code back into beebem after mounting the ssd but not back into beeb or master.

keirf, 2019-04-08:
Okay thanks. Could you give ADFS writes a go? I will have to try dumping an SSD from Gotek and look at the raw tracks.

keirf, 2019-04-08:
Here's another firmware to test. Any track with interleave or skew was getting its sectors molyneuxed.

[ff_220_2.zip](https://github.com/keirf/FlashFloppy/files/3056446/ff_220_2.zip)

MadeGray, 2019-04-09:
Good morning Keir, That seems to have done the trick! Tested reading and writing DFS and ADFS on both beeb and Master. loaded several games via Gotek etc to test reading and created simple programs and saved and reloaded. I'll do some further testing this evening but early indications look very good. Many thanks

keirf, 2019-04-09:
Excellent, thanks for the testing!

keirf, 2019-04-09:
Released v2.4a with following fix:
https://github.com/keirf/FlashFloppy/commit/d032e02357fa0d48e78ce6272dd699358748acfb

closed 2019-04-09 by keirf