keirf/flashfloppy #221: v2.5a buttons not working

SilverGreen93, 2019-04-14:
Hi, just updated to v2.5a.
The up/down buttons do not work anymore, the Gotek resets when the buttons are pressed.
Also, when pressing the 3rd button (the EJECT/write-protect), a weird pattern appears on the OLED:
```
********************
*FATFS* 09
```
What happened?
I tried various settings for twobutton-action = zero and rotary = none, but it doesn't seem to change the behaviour.
I also re-formatted my stick, but no luck.

keirf, 2019-04-14:
Can you downgrade and confirm it works again? If so perhaps work out latest version that works?

SilverGreen93, 2019-04-14:
Yes, if I downgrade, it works again.
I might have identified the problem though... I have "ejected-on-startup = yes" in my FF.CFG file
Maybe this in conjunction with this update which fixes "USB stick will usually stop flashing when drive is not being accessed" is a bad combination.
I reverted the ejected-on-startup = no and it works. If I change again to yes, I get FAT error 09 again, so this is the culprit. :)

keirf, 2019-04-14:
Thanks I will try reproducing this bug later

keirf, 2019-04-14:
It was a stupid bug. I'll do a 1.26a release this evening. Thanks.

keirf, 2019-04-14:
v2.6a is now released.

closed 2019-04-14 by keirf

SilverGreen93, 2019-04-15:
Yup, now it seems to work fine, thanks!