keirf/flashfloppy #223: ATR - slow loading with 2.04/XF551 board

excommie, 2019-04-19:
I just tried the Gotek with 2.04 and FX551.  However, the load process is slow, it seems like it is missing the next sector?  

Please review the video I uploaded (turn on volume).

https://www.youtube.com/watch?v=_OGTVQXotDE

keirf, 2019-04-19:
Do you know how fast it should be?

keirf, 2019-04-19:
And how big is the ATR file?

keirf, 2019-04-19:
Here is a firmware for you to test. It has a couple of changes:
1. Significant sector interleave to account for slow SIO transfer speeds.
2. Speeds up the bitrate (better match for XF551).

[ff_223_1.zip](https://github.com/keirf/FlashFloppy/files/3097288/ff_223_1.zip)

excommie, 2019-04-19:
This new firmware fixes the slow loading issue.
Thank You - this is an awesome development for 8bit Ataris.

To answer you other question - 2.4a was about 50% slower than loading from a physical media with all ATR file sizes.

keirf, 2019-04-19:
Thank you. Could you also check writes work okay too? A simple file save would do.

keirf, 2019-04-19:
Also there may be scope for a little more speed up. I don't know if atari floppies were skewed track to track. That might be the case if flashfloppy is still 5-10% behind real floppy.

excommie, 2019-04-19:
OK, will try writing tonight.  This is great, thanks again.

excommie, 2019-04-20:
I loaded Dos 2.6f via the Gotek.  Then I selected another disk image on the Gotek, and formatted it.  Then I wrote the Dos files to this new virtual drive.  I then booted from this drive.  I did not encounter a single problem.  All is working without a hitch.

keirf, 2019-04-23:
Okay thanks. Closing now. Fixed in v2.7a

closed 2019-04-23 by keirf