keirf/flashfloppy #224: Can't upgrade from usb stick

leothebest2014, 2019-04-20:
Sorry ,I'm a new to this, now i have a problem: I download the v1.2 hex file ,then i use Flash Loader Demo in win10 to flash the firmware to the Gotek(sfrc922d) . So far no error occurs. 

But after flashing, I disconnect the boot0 and  vcc3v3 ,then power up the Gotek, the 7-segment lcd showed 'F-F', I plug the usb with DSKA0000.FDI to the drive. nothing changed. I pressed the Next button, only show the version.

I try to upgrade the drive, I copy the 'FF_Gotek-v1.2.upd ' to the root of the usb,plug it, pressed two buttons, power up,LCD showed 'upd', then I released them , Lcd showed 'usb' and paused.

I try to flash v2.6a,but still the same. It seems the drive doesn't work. My usb is Kingston 4GB,and MBR,FAT32 formated. The working voltage is 4.98V. I hope there is a way to solve my problem,Tks!

keirf, 2019-04-20:
The problem is that the filesystem on the USB stick is not being recognised. Either the stick is not compatible (that would be very rare!) or the filesystem is not in fact FAT32, or the partitioning is not MBR. Those are the only options. If the filesystem were being detected then in normal operation (ie. not instigating update) the display would always change from 'F-F' when the USB drive is inserted.

So... you need to try another USB drive, and double check how you are formatting your drives.

Oh, one other possibility is a hardware flaw with your Gotek, and the USB port is not working properly. That would also be very rare, much rarer than mucking up the formatting. Only way for you to be sure is if you have another Gotek to test against.

leothebest2014, 2019-04-20:
Hi, thank you for your replay!
Is there any possibilities that it's the reason i didn't connected the host pc to the drive? I just connect the power supply! But I think the firmware upgrade procedure don't need the host pc,right?

| |
truename
|
|
chenyan9@yeah.net
|
签名由网易邮箱大师定制

keirf, 2019-04-20:
The Gotek only needs power for the above test. You should move on from F-F display even if ribbon cable is not connected.

leothebest2014, 2019-04-21:
HI,finally it works after I changed another usb stick(SANDISK CRUZER 4GB). 
But the Kingston can be used on the HXC firmware Gotek Drive( before I know your firmware, I've already flashed a Gotek with HXC firmware) .  On the FF firmware Gotek, not work properly. I'll try low-level format this stick and see.  Anyway , thanks again!

keirf, 2019-04-21:
Can you post a link to the exact Kingston stick you have? Or send me it?

leothebest2014, 2019-04-21:
I live in suzhou, China. It's a longway to send the USB. I bought this USB for a long time , I can't find the link either, sorry! Is there any other way i can do to help you analyze the USB?

keirf, 2019-04-21:
I think it's not worth the bother. It's not a currently available stick, it's old, maybe it's on its way out. I haven't had a bad stick report in over a year so this is definitely a rare thing.

I do like to take a look myself if I can but I agree in this case, for an old 4GB stick, it's not worth sending it half way around the world.

closed 2019-04-21 by keirf