keirf/flashfloppy #225: Feature: WiFi support via ESP8266 ?

penfold42, 2019-04-23:
What’s the interest level for adding WiFi control and possible USB access ?

We could do it over SPI bus for faster image up/download suppprt

keirf, 2019-05-24:
An encroaching problem is Flash space on the Gotek. Do you have experience with ESP8266 and what would be best firmware to use there?

penfold42, 2019-05-24:
I’ve done a fair bit with esp8266 - Arduino libraries are pretty stable.

I was suggesting it as a second processor for flashfloppy.

A cleaner solution if you’re needing to replatform would probably be the ESP32 as the 8266 is really tight on GPIOs

They’ve just announced a new ESP32 variant due later this year that has USB OTG support which is needed for USB storage support.

One big thing to check first is interrupt latency - something kinda important for this project I imagine

They just announced a new variant dure out

keirf, 2019-10-09:
From FlashFloppy's POV this may be a dupe of #180

penfold42, 2019-10-09:
Yep - another option is port FF-OSD to the 8266 and provide websocket access to a frame buffer

closed 2019-10-09 by penfold42