keirf/flashfloppy #227: WRITING PROBLEM ON EPSON-286U-ST

mrdavoudi1, 2019-05-19:
Hi
I use v1.2 for a 720k (9sec/track x 512 byte/sector x 80 track) for a Epson 286u-std  that use in CNC machines as a replacement of NEC PC9801.
all things is OK and images read successfully and machine is boot from tow floppy drives that connected as SYSTEM and USER  as DS0 and DS1 with SHUGART interface.
The problem is when I save the program from machine to diskette , machine says that program is saved on floppy , and machine show that free space on the 720k disk on machine is reduced as file size.
But when I eject the stick and reinsert again in emulator or change the image on stick by forward and backward key and the load the diskette nothing is saved on diskette.
and when I open the Image on PC nothing on image is saved.
I update the firmware from 1.2 to 2.8 and when I write on diskette 7-segment show me the "0_."
but also nothing is saved on stick on image.
I also change the write-protect tag on FF.cfg and machine understand and say me to remove write protect key.
I am confused what is the problem?

keirf, 2019-05-19:
Update to latest firmware, using the update file in subfolder alt/logfile. Attempt a write. Now eject the disk image (eg press any gotek button). Then pull the USB stick and attach here the FFLOG.TXT file that's saved on it.

mrdavoudi1, 2019-05-19:
Thanks.
I'll do and inform.

mrdavoudi1, 2019-05-22:
Hi Jeff
I do the procedure
This is the result in text.
I can't find where to attach the file.
So copy here the content of it :

** FlashFloppy v2.10a for Gotek
** Keir Fraser <keir.xen@gmail.com>
** https://github.com/keirf/FlashFloppy

Board: Standard
Config: Flash Slot 5 (ver 2, size 63)
Interface: Shugart (pin2=chg, pin34=rdy)
Motor: Ignored
I2C: Invalid bus
Display: 3-Digit 7-Seg LED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 1005
 PID : B113
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : USB
 Product : FLASH DRIVE
 Serial Number : DCAF2791
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 105 items
Mode: Native
Cache 97 items
Current slot: 0/0
Name: 'USER' Type: ima
Attr: 20 Clus: 00000027 Size: 737280
Fast Seek: 1 frags
Cache 34 items
New max: read_us=1412
New max: read_us=1567
New max: read_us=1781
New max: read_us=1802
New max: read_us=1808
New max: read_us=1818
New max: read_us=1854
New max: read_us=1898
Cache 97 items
Before: After: Current slot: 0/0
Name: 'USER' Type: ima
Attr: 20 Clus: 00000027 Size: 737280

keirf, 2019-05-22:
No writes are being recognised by FlashFloppy. Either the host is not correctly initiating a write, or FlashFloppy is throwing them on the floor for some reason. I will attach an updated firmware here which will provide more logging about writes.

keirf, 2019-05-22:
Please update to this firmware and collect logs exactly as you did before.

[ff_227_2.zip](https://github.com/keirf/FlashFloppy/files/3207794/ff_227_2.zip)

keirf, 2019-05-27:
Please reopen thus ticket if this issue persists and you have new information.

closed 2019-05-27 by keirf