keirf/flashfloppy #229: Unable to update from SD card on enhanced Gotek

salacpavel, 2019-05-22:
as a title says, I have tried to follow firmware update procedure off the SD card, but seems like FF update does not scan the SD card for .upd file.
Another issue I have observed is that SSD1366 OLED screen is not properly initialized, when Gotek is started to update mode. The good old pre-0.15 FW garbage is displayed instead. The same display I works in normal operation mode tho.

keirf, 2019-05-22:
Update from SD card is not yet supported. I may add it.

Update mode runs in the bootloader, which is not usually itself updated. So you probably need to update your bootloader. See instructions here: https://github.com/keirf/FlashFloppy/wiki/Firmware-Update#updating-the-bootloader

keirf, 2019-05-23:
I've fixed the SD card thing and will make bootloader update simpler in next release v2.11a. Look out for new Bootloader Update instructions in the wiki.

commit 20116bc0 (2019-05-23): Replace the Reloader mess with an all-in-one Bootloader update file.

Also allow the update file to be supplied on SD Card (Enhanced Gotek only).

Fixes #229
commit 756333a0 (2019-05-23): Replace the Reloader mess with an all-in-one Bootloader update file.

Also allow the update file to be supplied on SD Card (Enhanced Gotek only).

Fixes #229
closed 2019-05-23 by keirf