keirf/flashfloppy #232: FDD-UDD U144K

nretro, 2019-05-28:
By accident (resp. stupidity) I bought a FDD-UDD U144K. 

Looks quit similar, but it's not obvious how the jumpers have to be connected to flash a new firmware. It has got a 7x2 pin header labeled J0, J1, ... J6 and a 5x1 pin header labeled 07,46,47,48,49. Did anybody already figure out how these relate to the headers of the standard gotek?

keirf, 2019-05-28:
Can you take if apart and take a picture of the pcb please.

nretro, 2019-05-28:
Here it is. Looks similar enough... tough it is probably better to check first.... Thanks for looking into it.

[image]

keirf, 2019-05-28:
Yes it is probably using the same firmware and broadly compatible. But someone needs to reverse engineer the schematic of the jumper blocks. For now I'll have to add this to the goteks-to-avoid wiki page.

nretro, 2019-05-28:
Is there a schematic for the usual gotek? I also have two of those, but it would be easier to check if I would not have to reverse engineer those first. I mean the zero hypothesis would be that they did not change the headers.

nretro, 2019-05-28:
Just for the fun of it, I checked and found out, that the lower pin of the 1x5 connects to the same pin of the processor. So there is hope...

nretro, 2019-05-28:
46 (2nd from below) connects to the third pin on the upper side of the CPU (from left). Usual gotek connects this to the 6th pin on the upper side. 
47 connects to the 2nd pin on the upper side of the CPU.
48 connects to the 3 digit display where it says 3v3.
49 connects to the upper most pin on the left side of the cpu.

keirf, 2019-05-29:
A high res gotek schematic is available on the cortex amiga website. You can find it via Google images.

You can work out pin numbers and io assignments on the stm32 by downloading stm32f105 data sheet. This includes pin assignments on an illustration of the chip package (package is lqfp64).

nretro, 2019-05-29:
On the 1x5 header there is NRST, and then CPU pins 46-49: PA13, VSS2. VDD2, PA14. PA9 and PA10 (USART1) are not connected at all. Both methods to flash the firmware use USART1. Is there an alternative or do i have to solder a connection to the CPU?

nretro, 2019-05-29:
It does seem to support some kind of flashing via USB. When both buttons are pressed at startup, the display shows U00 -> 126 -> F01. But I could not  find any suitable files on the internet or on the supplied CD.

drdpj, 2019-05-29:
PA13 & PA14, vdd and gnd will let you program it with an stlink v2 adapter.

keirf, 2019-05-30:
Yes the 1x5 is a programming header. I've never used the jtag interface but I think I have the adapter in my box of bits and it's supposed to be fairly straightforward.

Can you work out the other header?

nretro, 2019-06-02:
The 2x6 is almost the same as in the usual gotek, only left and right got switched:
J0-J2 right PA0
J3-J6 right VSS
J6 left PIN4 (of floppy connector)
J5 left 37 PC6 (Jumper JA)
J4 left 27 PB1 (Jumper JC)
J3 Left 16 PA2 (Jumper JB)
J2 left PIN10
J1 left PIN12
J0 left PIN16

I ordered a PL2303HX a while back. Is that likely to work or do I need another adapter?

keirf, 2019-06-03:
The PL2303HX is a serial adapter and would program via USART1, which is not connected to a header on this Gotek. Similarly, you need access to PA9 to strap it high to program via the USB-DFU method.

So, the only method available here is via the SWD-JTAG interface on the 1x5 header. For this you need an ST-LINK v2 adapter. You can get clones on Ebay for a pound or so from China, or for a little more if you want quicker delivery.

I have one or two stashed somewhere although I've never used them, but I am happy to work out a method on Linux at least, following one of the online guides such as: https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Programming-an-STM32F103XXX-with-a-generic-%22ST-Link-V2%22-programmer-from-Linux

nretro, 2019-06-03:
Thanks. Just ordered one of those. The guide looks detailed enough. If I just have to flash the usual FF_Gotek-v1.2.upd  to 0x08000000, this should not be hard.

keirf, 2019-06-03:
v2.12 is now the latest stable release. 😊

But yes should be fairly easy. Please report on correct programming steps and/or which online guide you successfully used, and I will document this method and this gotek variant on the wiki.

nretro, 2019-06-06:
Success. 
1. Connect GND and +5V to power connector of gotek
2. Connect  SWDIO to header pin 46 (PA13)
3. Connect SWCLK to 49 (PA14)
4. Connect header pin 07 (NRST) to GND
5. Execute (I used OpenOCD 0.10.0 on Debian unstable)
openocd -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f  /usr/share/openocd/scripts/target/stm32f1x_stlink.cfg
6. In a different shell execute "telnet localhost 4444". From now on we use the telnet session
7. type "reset halt" and immediately disconnect header pin 07. The reply should be something like

target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x080061d4 msp: 0x200006c0

8. "stm32f1x unlock 0" was needed to flash on this board to enable writing to the flash bank. ("flash banks" displays all available banks)
9. type
" flash write_image erase /tmp/FF_Gotek-v2.12.hex"
10. type "reset run"
11. disconnect st-link.

nretro, 2019-06-06:
OK... and does work as expected loading an adf on an Amiga. I'll be busy with other things over the next couple of weeks, but at some point I'll also try to add display and rotary mods. I'll post the results.

keirf, 2019-06-06:
Will be interesting. Especially the rotary pins may not have a header. Will have to see if there's anything to solder to other than mcu pins directly.

penfold42, 2019-06-06:
Worst case, use the JTAG pins or the SPI flash pins

keirf, 2019-06-07:
Yes can remap to JTAG pins if needed

keirf, 2019-06-12:
I have linked to this ticket from the wiki: https://github.com/keirf/FlashFloppy/wiki/Gotek-Compatibility#models-to-avoid

I will close this ticket now, but we can add to it with further info in future (eg. if we work on support for rotary encoder).

closed 2019-06-12 by keirf

LukaszDziwosz, 2019-12-18:
Thanks for the pinout and instructions. I flashed on Windows. Here is some photos. To fit Oled I had to trim inside case as hole is too close to edge.
[image]
[image]
[image]
[image]

sirkaa, 2020-03-06:
hello

reviving this thread, made the error too and got this board by mistake. Now trying to program it but it seems that RDP is on and i have found no way to disable it with option bytes thus no letting me program it. Any help would be appreciated. cheers

ramparts, 2020-03-07:
I have this gotek too and from st link v2 chinese programmer can´t be programmable with st-link utility.
On automatic programming gives the read programming protection error and won´t go further more...

LukaszDziwosz, 2020-03-07:
While in automatic programming mode you must do a reset on pin 7. I thought you guys will read nretro comment fully before reading mine.

ramparts, 2020-03-07:
Gotek drive reset line 07 constantly grounded before gotek 5v power up..
Programmer say can,t remove read protection and loop to device 2...
Maybe some issue on the programmer or the gotek U144k it self.
Never had issues on other gotek through serial programming..

LukaszDziwosz, 2020-03-07:
you need to un-ground pin 7 when in automatic mode to unlock this Gotek

ramparts, 2020-03-07:
Thanks I will try it.
Regards

JOSÉ

ramparts, 2020-03-08:
I have try and had success flashing the drive but it´s sillyest thing i ever encounter.
It did really had to make a temporary reset while in programming mode,everytime the programmer started to read i had to make a reset.
sometimes game error while recording the bytes option with the disabeled read out protection.
But after o erased all the banks and flash memory,i could save the table without any issues..

Still i prefer ther serial programming... :)

Thank you for all the help.

sirkaa, 2020-03-09:
> While in automatic programming mode you must do a reset on pin 7. I thought you guys will read nretro comment fully before reading mine.

Thank you very much for your reply. I have no experience in programming, i am a musician and i really appreciate you guys help. I did read all posts before commenting but as nretro used Debian i did think it was different on windows.Your pictures did not show this action nor it was mentioned so i got confused. Thank you anyway. have a good day.

hexbus, 2020-04-07:
Did what @nretro mentioned above - 

- Connect GND and +5V to power connector of gotek
- Connect SWDIO to header pin 46 (PA13)
- Connect SWCLK to header pin 49 (PA14)
- Connect header pin 07 (NRST) to GND

Installed the ST-Link V2 and had the three pins above connected to the U144K.  Powered up the Eilasung FDD-UDD U144K on the rear floppy power connector.  Ran this command and keep getting this:

 /d/dev/openocd/bin/openocd.exe -f /d/dev/openocd/share/openocd/scripts/interface/stlink.cfg -f /d/dev/openocd/share/openocd/scripts/target/stm32f1x.cfg

```
Open On-Chip Debugger 0.10.0 (2020-03-10) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK v2 JTAG v31 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.116464
Error: init mode failed (unable to connect to the target
```)

Tried putting it in a loop, turning on and off the unit, etc.  Thoughts on what I should try?

keirf, 2020-04-07:
Try connecting pin 07 (NRST) to NRST/RESET on the ST-Link.

hexbus, 2020-04-07:
Tried that as well.  I even dropped the external +5V connector and hooked up the following (on the ST Link V2):

- SWCLK to 49 (PA14)
- +3.3V to 48 (apparently, 3.3V regulated when I measured with the  5V connected instead)
- GND to 47 
- SWDIO to 46 (PA13)
- RST to 07 (NRST)

The unit powers up just fine hooking up that entire 5-pin header directly to the ST Link, but alas, still getting the same errors.  I bought three of these, so wil try the other two.

I brought up the ST-LInk software and did a Target->Connect to target.  Same error.  Went into Settings-> Mode and changed from Normal to "Connect Under Reset".  No change, still can't connect to the target.

Very strange.  Either I got a bum drive or a bum ST-Link.  (or they've changed the firmware on the chip maybe?)

hexbus, 2020-04-07:
Just an update.  Here's what worked for me:

Hooked up the four pins to the ST-Link.  Did **not** hook up floppy power.

- SWCLK to 49 (PA14)
- +3.3V to 48 (apparently, 3.3V regulated when I measured with the 5V connected instead)
- GND to 47
- SWDIO to 46 (PA13)

Shorted Ground on the floppy power connector to 07 (NRST).  Ran the STM steps from 5 onward that @nretro mentioned above.  Success!

keirf, 2020-04-07:
Ah ST Link clones provide 3.3v power. You don't separately power the gotek in that case (potentially bad times if you do).

zezba9000, 2020-10-19:
I feel the need to share some better images of this process etc and what works for me every time.

**<< To to flash Gotek (FDD-UDD U144K) >>**
* Remove all jumpers from the Gotek drive
* Wire an ST-Link-V2 programmer to the Gotek (Image 1)
* Install and open "STM32 ST-Link Utility" in Windows (Image 2)
* Hit connect button in ST-Link Utility then quickly unplug NRST (aka Pin 07)
* The device should now be connected (Image 3)
* NOTE: look at Image 3.5 to validate your connection settings if you have an issue
* Now connect NRST (aka Pin 07) to RST pin on your ST-Link-V2 programmer
* In ST-Link Utility go to Target->Program (Image 4)
* Download FlashFloppy and extract
* Select a FlashFloppy .hex firmware file and flash your Gotek (Image 5)
* Now put a jumper on pins J2 (aka SO)
* You're done!

**<< How to configure Gotek USB drive >>**
* Download FlashFloppy and extract
* Copy "flashfloppy-vX.XX\HxC_Compat_Mode\<Atari_ST OR Amiga>\AUTOBOOT.HFE" to "flashfloppy-vX.XX" folder
* Copy "flashfloppy-vX.XX\HxC_Compat_Mode\HXCSDFE.CFG" to "flashfloppy-vX.XX" folder
* Copy all files in "flashfloppy-vX.XX" to your USB root
* Create a games/app folder and add ".st" files.
* You're done!

**<< How to mount .st files >>**
* Turn on your Atari/Amiga and use arrow keys to navigate menus (Hit help key for help)
* Select a .st file and hit enter
* Now select a mount slot (which is selectable via Gotek buttons)
* Hit F7 to restart the computer with the last mounted .st image file
* Use Gotek buttons to change what .st image is mounted from your configured slots or load bad the Gotek manager app

**<< How to update Gotek firmware >>**
* Download new FlashFloppy firmware
* Delete old FlashFloppy files on you USB drive
* Follow steps in "How to configure Gotek USB drive"
* With the power off hold both Gotek buttons down
* Now with buttons held turn on power (the Gotek will update itself and restart)
* Restart computer
* You're done!

[image]

**For more images and files...**
Here is a download link of all the files needed for this: https://1drv.ms/u/s!Alt5wRhQYKD7gfdE-n93FjfBJv0ECQ?e=hg2LdO

JacekHelka, 2021-01-22:
> 
> 
> While in automatic programming mode you must do a reset on pin 7. I thought you guys will read nretro comment fully before reading mine.

> 
> 
> Thanks for the pinout and instructions. I flashed on Windows. Here is some photos. To fit Oled I had to trim inside case as hole is too close to edge.
> [image]
> [image]
> [image]
> [image]

> 
> 
> Thanks for the pinout and instructions. I flashed on Windows. Here is some photos. To fit Oled I had to trim inside case as hole is too close to edge.
> [image]
> [image]
> [image]
> [image]

Can you please tell me the type of display that is mounted in this kind of GOTEK?
Normally it is a LCD1602 HD44780 2x16 with I2C Converter for LCD 1602 Display.

Maybe some wiring diagram?

Thank you in advance for your reply.

zezba9000, 2021-01-22:
https://www.ebay.ca/itm/New-Amiga-Gotek-OLED-IIC-I2C-0-91-128x32-LCD-Display-Module-3-3V-5V-White-679/114641152709

That item also shows wiring.

LukaszDziwosz, 2021-01-23:
It's the same cheap 0.91" I2C OLED that is used for normal Gotek o Arduino, any 4 pin display will work. I used to buy on Ali for super cheap like £2, wiring is the same as normal Gotek but you have to desolder numeric display first. I have this one installed in my Akai sampler so there is no way I'm taking it apart. Good luck and have fun.

JacekHelka, 2021-01-23:
Thank you all for your replies.
I hope I can count on help when Gotek arrives.

claaaaaay, 2021-02-10:
You can flash it the classic TTL way in Windows, you just need to solder 2 wires to the STM32 IC.

Provide 5v and GND to the power connector.
Solder wires between STM32 pins and TX/RX on TTL adapter. I didn't take note of which pin was which. Trial and error will get you there.
Connect 3.3v (48 on JTAG) to the side of R6 closest to the front of the board (away from floppy/power connector) before powering up board. If 000 is displayed try again. When successful nothing will be shown on display and it's ready to connect.
Connect and flash as normal with STM32 Flash Loader Demonstrator. "remove protection" wasn't required.

[image]
[image]
[image]
[image]
[image]
[image]

JacekHelka, 2021-02-15:
Hello everyone,
Thank you for the description and photos. Gotek is now working with Amiga!
Now I'm waiting for delivery of the display.

disc0very, 2021-04-21:
Hi, 
thanks for the infos shared about this gotek clone.
Did anyone modded this gotek clone to add the encoder? Can anyone provide any instruction on how to accomplish this mod on this gotek clone?
Thanks

keirf, 2021-04-21:
> Hi,
> thanks for the infos shared about this gotek clone.
> Did anyone modded this gotek clone to add the encoder? Can anyone provide any instruction on how to accomplish this mod on this gotek clone?
> Thanks

See #456 but it's not easy (no header).

JacekHelka, 2021-04-21:
Is it possible to add sound module to this version of Gotek?
It is about emulating the sounds of a real floppy drive.

disc0very, 2021-04-29:
> [image]

can you please provide more clear picture of the oled connections to the circuit board? or can anyone give me instruction on how to make this connections? thanks!

tenuke, 2021-04-29:
Did you check that thread #456 linked above?
One post there has 2 pictures and written explanation

quantum8, 2021-05-12:
thanks @zezba9000 your guide worked perfectly to flash my accidentally bought U144Ks!

djkoelkast, 2022-05-07:
I have a board that looks a bit like this, but is slightly different, any tips on this? I've had it for years but never used it.

[image]

keirf, 2022-05-07:
Totally different main chip. Not supported.

drewdty, 2022-06-27:
Sorry to revive the topic but I have two questions
1º when following the steps to flash this type of gotek of the
mate LukaszDziwosz and I'm F-F blinking
It is right?
2nd to prepare the usb for amstrad? What file do I have to put?
for atari or amiga is the HXCSDFE.CFG
is it the same for amstrad?
thanks and sorry for my ignorance

markgm30, 2022-08-07:
> I modify my FDD-UDD U144K with a OLED Display and a Rotary encoder. I flash it with FF 3.25 I can select all images. It works as the GOTEK's. But I have'n a Access from my PC. The actvity LED is on. The other GOTEK's work on this PC correct. Have you an answer.

I have the same issue.  Once I removed the 7 segment display and added an OLED screen the red LED is always illuminated.

keirf, 2022-08-07:
The red LED is merely a display activity LED. Since FlashFloppy is continuously updating LCD and OLED displays, the red LED is permanently illuminated. It's a pretty pointless LED.

wmoindrot, 2022-11-30:
Also ended up with one of these by accident/not enough research. Followed instructions, mainly by @zezba9000. To get it to work for me (as a casual newbie):

- Found when had to kind of take power out to reinitialise the gotek in some way
- As following instructions, hit connect then pull pin 7
- But then found that in order to successfully flash programme had to leave pin 7 removed (and I think be hasty about it).
- Found some firmwares didnt work, tried 105 variant of 3.36 and no joy, one I found work was the original here of 3.7a which is from I guess before the 345 and 105 split
- For jumpers I followed the typical configuration. For my ibm compatible that was selecting S1 via jumper across J1, and then to choose the ibmpc interface type bridged J6 and JB on the left (a mirror image of the usual Gotek), I think this was finally got things working for me

chm-dev, 2023-02-02:
> Success.
> 
> 1. Connect GND and +5V to power connector of gotek
> 2. Connect  SWDIO to header pin 46 (PA13)
> 3. Connect SWCLK to header pin 49 (PA14)
> 4. Connect header pin 07 (NRST) to GND
> 5. Execute (I used OpenOCD 0.10.0 on Debian unstable)
>    openocd -f /usr/share/openocd/scripts/interface/stlink-v2.cfg -f  /usr/share/openocd/scripts/target/stm32f1x_stlink.cfg
> 6. In a different shell execute "telnet localhost 4444". From now on we use the telnet session
> 7. type "reset halt" and immediately disconnect header pin 07. The reply should be something like
> 
> target halted due to debug-request, current mode: Thread xPSR: 0x01000000 pc: 0x080061d4 msp: 0x200006c0
> 
> 8. "stm32f1x unlock 0" was needed on this board to enable writing to the flash bank. ("flash banks" displays all available banks)
> 9. type
>    " flash write_image erase <your_path>/FF_Gotek-v2.12.hex"
> 10. type "reset run"
> 11. disconnect st-link.
> 
> I still have to test if it actually works. Up to now I only checked that it says "F-F" on the display.

Didn't have st-link on hand and used raspberry pi 400 with openocd.
Can confirm - worked as a charm.

More details in case someone would like to save few hours ;)

GPIO pins connect to header pins according to @nretro post
[image]

Create Interface config (ie ocd_rpi.cfg)  for native raspberry pi 4 / 400:
` 
#Use RPi GPIO pins
interface bcm2835gpio

#Base address of I/O port - this is changes with different rpi models - use xxd -c 4 -g 4 /proc/device-tree/soc/ranges to check
bcm2835gpio_peripheral_base 0x00000000

#Clock scaling
bcm2835gpio_speed_coeffs 146203 36

#SWD                swclk swdio
# Header pin numbers 22    18
bcm2835gpio_swd_nums 25    24

# JTAG                tck tms tdi tdo
# Header pin numbers  22  18  16  15
bcm2835gpio_jtag_nums 25  24  23  22
`

And run 
openocd -f ./ocd_rpi.cfg -c "transport select swd" -f /usr/share/openocd/scripts/target/stm32f1x.cfg 

Telnet connection is available and you can follow @nretro instructions.

mjnman, 2023-03-17:
Can I connect the buzzer in this version?
The older version I connected with the pin JB

keirf, 2023-03-17:
Yes buzzer will work.

mjnman, 2023-03-17:
What are the connection pins? I didn't see the JB

keirf, 2023-03-17:
Position J3

duca750, 2023-06-24:
> I modify my FDD-UDD U144K with a OLED Display and a Rotary encoder. I flash it with FF 3.25 [image] see here. I can select all images. It works as the GOTEK's. But I have'n a Access from my PC. The actvity LED is on. The other GOTEK's work on this PC correct. Have you an answer.

Pease, have diagram for this connections please? Thanks!

[23 more replies]