keirf/flashfloppy #234: better support of 128x64 OLED

fook42, 2019-05-30:
having a display with 64pixel height, it would be nice to be able to configure which rows are doubled in size and not having it doubled all the time.
addition: yellow/blue displays (where only the first 16 pixels are yellow) should be supported by using a small first text-row and a large second text-row.
[image]
[image]

-> should be:
[image]
[image]

fook42, 2019-05-30:
i made some changes to the current code (based on 2.11a) - can i add a branch for this and create a merge-request afterwards? or how do you like to have the code?

keirf, 2019-06-12:
Please see https://github.com/keirf/FlashFloppy/commit/83186755a6da743b0f4dbe4bf7e230057e065354

And raise new issues as necessary.

closed 2019-06-12 by keirf