keirf/flashfloppy #236: selecting text row-content

fook42, 2019-05-30:
Would be nice to have an FF.CFG option to select which text-row contains what information to be displayed (on LCD, oled).. like "DiskName" or "TrackInfo" ...

keirf, 2019-05-30:
Yes this is a good idea and one I was contemplating. Thank you for your earlier patch for text size by the way. This will miss 2.x stabilisation but I will take it for 3.0a and work on top of it.

I would also like to add some extra optional lines eg image info, statistics,... Probably give them a default ordering with numeric assignment and make an all in one config option that allows to specify size and line, row by row. Eg 0s,1b,3s for a small-big-small non-default config.

fook42, 2019-05-30:
i'm currently working on a second option for this.. like "text-rowcontent=name,track,..." to specify the contents per row.. but your suggestion sounds good to me as well.
how many different contents do you think will be available ? (0=name, 1=trackinfo, 2=statistics, 3=imageinfo, ... )?
or shall we try to specify a "printf - replacement-string" format for each row?
like :  row1=%n , row2=%track %imagetype , row3=%access ... ?
this may be the most flexible way, but leads to a lot of rework i guess ^^

keirf, 2019-05-30:
Yeah it's a tricky one. I'm leery of making things too generic and configurable. Or maybe it's just that a printf style configuration probably needs a restructuring of the crappier code in main.c 😉 But no, I do hate projects with pointlessly millions of cryptic config options. This could be nice though. I'm torn. I would probably sketch out a few patches and see how I like them.

keirf, 2019-05-30:
I am pleased you are making patches and I think perhaps we will iterate on them in a branch after 2.x stable is branched off in a few days time. It is great to have someone else capable of knocking out decent feature patches and design suggestions and feedback.

keirf, 2019-06-12:
Please see https://github.com/keirf/FlashFloppy/commit/83186755a6da743b0f4dbe4bf7e230057e065354

And raise new issues as necessary.

closed 2019-06-12 by keirf