keirf/flashfloppy #238: Roland W30

roses2uk, 2019-06-01:
At the moment, I use a standard Gotek SFRM72-FU-DL with my Roland W30 sampler workstation. I convert my samples to *.IMG files, and load them using the b00-b99 system from a folder on the USB stick called IMG720. But I am limited to only 100 disk images, hence why I am considering upgrading to Keir's FF, and adding an OLED screen and rotary encoder. My question is: after flashing do I still use IMG files or another format such as the native Roland *.W30 or *.OUT ? Is there a limit on the number of files on the stick? What kind of *.CFG file would I need in the root directory? Any other W30 owners out there who have trodden this path before me?? Thx

keirf, 2019-06-01:
I think most Roland use PC drives and formats so: jumper at JC and at S1. Put IMG files on your stick and go. No configuration needed, probably. By the way Roland OUT is really just IMG. You should be able to rename OUT images to IMG suffix. I could even add direct OUT support.

roses2uk, 2019-06-01:
Thanks. Yes, I forgot to mention that that is what I do - just rename the W30 and OUT files to IMG. But of course with the Gotek they have to be 001.IMG, OO2.IMG, etc which is frustrating as it's hard to keep track of what's what. With FF you can have normal file names, yes (with an OLED display of course)? I use the normal standard config file, I think? Also, is there a quick and easy way to jumper pins 1-2 and 3-7 for the flashing process (I don't have a soldering iron and no skills to use one either). What is the limit to number of files on the stick that FF can deal with?"

roses2uk, 2019-06-01:
Other W30 owners reading this, please give me some feedback on how it worked for you with FF...

keirf, 2019-06-01:
Yes you can have normal filenames. A reasonable number of files per folder would be around 500, though you can have more. But FlashFloppy supports folders, so you can arrange larger numbers of images into subfolders, no problem. In that way, FlashFloppy supports as many files as you like.

Some people jumper the pins using a paper clip or other bent thick wire which can be wedged into the PCB holes.

roses2uk, 2019-06-01:
Folders?? Cool, the answer to all my prayers. Ok, I'll try the paper clip trick. I hope there's enough solder around the header pin holes to make contact. Can you leave links for where to buy a suitable OLED screen and rotary encoder? Also, what are those pre-made 10cm wires with square connectors that go onto the header pins called? Appreciate all your help...

keirf, 2019-06-01:
The holes are through-plated so bunging a wire in the hole should make contact.

There are pictures of suitable displays and encoders on the wiki. Look for a 128x32 or 128x64 oled, and pretty much any rotary encoder (guess you want one with pcb if you don't solder).

The jumper wires are called dupont cables/wires.

roses2uk, 2019-06-01:
Are these ok?

https://www.ebay.co.uk/itm/0-91-inch-128x32-IIC-I2C-Blue-OLED-LCD-Display-DIY-Module-DC3-3V-5V-For-PIC-A-YB/312636253425?epid=6019964429&hash=item48ca92a4f1%3Ag%3ApjAAAOSwGuxc8VW4&LH_BIN=1

https://www.ebay.co.uk/itm/KY-040-Rotary-Encoder-Module-Brick-Sensor-Development-For-Arduino-Moduleu-TPD/283478088170?epid=1888833304&hash=item42009c59ea%3Ag%3AXsQAAOSw5IJWglGc&LH_BIN=1%7C1

Female-Female dupont cables??

keirf, 2019-06-01:
Yes fine. Bear in mind if display header is soldered that way round, or supplied separate, you will have soldering to do.

roses2uk, 2019-06-01:
Yes, I already noticed that the header pins were separate. Drat! Unless I can find a company that sells displays pre-soldered. Or is it enough to hot glue the pins in place (like you already advised me to do for the paperclip trick as those holes are also through-plated).

keirf, 2019-06-01:
You won't get a reliable connection that way. The pins are smaller than the holes. The paperclip (or whatever) method is only reliable enough for a few seconds of programming time.

Consider doing some soldering. It's quite easy and fairly cheap to start off.

roses2uk, 2019-06-02:
OK, I'll try, but it just means more expense for me (iron, solder, solder sucker, etc) and I'm already on a very tight budget as it is. Especially as I'm only going to use it all once...Thanks again for all your help, I'll let you know how I get on.

roses2uk, 2019-06-02:
Regarding flashing via USB-A from Windows, the wiki says: "Note: The *.dfu file is now included in the FlashFloppy distribution; there is no need to create it from the *.hex file and those steps can be skipped." So basically I just follow Nick Lines' video from 3m00s when he starts with the DFusedemo program...?? (Sorry if I appear a bit dense about all this.)

keirf, 2019-06-02:
Sounds about right.

closed 2019-06-02 by keirf

roses2uk, 2019-06-03:
"I could even add direct OUT support." That would be very useful, if it's not too difficult or time consuming for you. What about *.W30 format - is that substantially different to OUT/IMG? (Have ordered all the components needed, should arrive week after next. Still contemplating about a soldering iron kit - can't see anything within my budget. Will look locally in DIY stores.) Thx to you, Keir.

keirf, 2019-06-03:
OUT is now added in latest version. W30 I don't know anything about

roses2uk, 2019-06-03:
[EGUITAR1.zip](https://github.com/keirf/FlashFloppy/files/3248302/EGUITAR1.zip)

Attached example of a W30 file. Does that help you?

roses2uk, 2019-06-03:
Also, I noticed in the download for FF v1.2, there are files in the reloader folder. Are they relevant to my situation? If not, don't bother explaining them.

And the FF.CFG in the example folder, I just put that in the root of my stick, and that is sufficient for my 'out of the box' Roland???

keirf, 2019-06-03:
Don't use FF v1.2, download FF v2.12 instead. And ignore the files in the alt/ folder of v2.12.

keirf, 2019-06-03:
And you don;t need FF.CFG at all.

keirf, 2019-06-03:
I think W30 is just another standard IMG format. Try renaming them to IMG suffix and see if they work. If so, I can add direct W30 support just as I added direct OUT support.

roses2uk, 2019-06-03:
v2.12 - okey-dokey. No CFG at all? Great. As I think I said in my second post, I just rename both W30 and OUT files to IMG, and they work fine. So if FF already has support for the latter, it's surely gonna work with the former. This looks like it's going to be a piece of cake (famous last words), notwithstanding the soldering I will have to do...

roses2uk, 2019-07-16:
I seem to have trashed my SFRM72-FU-DL when trying to install the rotary encoder because I accidentally pulled off about 2mm of copper trace from the two holes marked 'JK' next to the 4 pins for the display (lack of experience and poor soldering skills) and don't know how to rectify it. The Gotek still works - can boot up with system disk and load other images, but would really like the encoder to work. (But had to revert to the old 3-character 7-segment display as the new OLED screen stopped working. Not sure if the damaged traces are responsible. Maybe reflashing would solve the problem??) My Roland W30 only works with 720kB images, so is it necessary to buy EXACTLY the same model, or can I get a cheaper 1.44 version and after flashing with F-F, it will still work for me? Anyone like to chime in with who in UK or Europe is selling them the cheapest these days?? Thx

drdpj, 2019-07-16:
The 1.44 and 720kb versions are identical in hardware terms, it's just the
original firmware that's different. As you're replacing that with
flashfloppy, just get the cheap 1.44mb version. Personally I search ebay,
sort by lowest price+p&p then decide whether I can be bothered to wait for
the slow-boat from Hong Kong.

roses2uk, 2019-07-16:
Wow, that's great news. If I'd known that from the start, I wouldn't have paid over the odds getting the 720k version from AliExpress. (I was recommended that model by another W30 owner but he - AFAIK - was using it in his keyboard with orig f/w and no mods.) Keir, can you make this clear somewhere on your Wiki: that whatever your application (720k/1.44MB), FlashFloppy caters for both formats. Maybe I'm stupid, but I didn't realise that till 5 mins ago.

keirf, 2019-07-16:
I think it's clear on the Gotek Compatibility wiki page: https://github.com/keirf/FlashFloppy/wiki/Gotek-Compatibility

roses2uk, 2019-07-16:
Yes you're right, it is. I retract my comment. I read that page again and again over the last few weeks, but it still didn't sink in. Apologies.