keirf/flashfloppy #239: stuck in UPD mode after flashing 2.12

roses2uk, 2019-06-11:
My USB A-A cable arrived today. Added the two jumpers as described in Nick's video, flashed from DFuse using the FF_Gotek-v2.12.dfu file, removed the jumpers, reattached Gotek to my W30, powered on, display says UPD, not F-F as I expected. So I added the FF_Gotek-v2.12.upd to the root of my FAT-formatted stick (which is good cos I was using it before), powered on holding down both buttons, but it does not update, and still says UPD. Any advice please?

roses2uk, 2019-06-11:
Small donation on its way to you as a token of my appreciation for all your assistance and advice. It's not much, but is all I can manage at the moment.....

keirf, 2019-06-11:
Do you have a jumper at position JA at rear of Gotek?

roses2uk, 2019-06-11:
Thanks for speedy response. Yes, jumpers at JA and S0

keirf, 2019-06-11:
Remove the jumper at JA. Those pins are for a third button and the jumper acts as a permanently pressed button.

roses2uk, 2019-06-11:
OK, now I get E34 error message. Is there any sense reflashing with 1.12?? (I can live without the OUT support)

keirf, 2019-06-11:
E34 is because you have no valid image files on your USB stick

roses2uk, 2019-06-11:
But F-F if no stick is inserted...

keirf, 2019-06-11:
That's right. Put some image files on your USB stick and insert it 😉

roses2uk, 2019-06-11:
Success!! The W30 found the first image (system disk) and booted up. How do I now choose different images using only the buttons (no display or rotary encoder arrived or installed yet)? Do I still need to unplug/replug the USB stick like before?

roses2uk, 2019-06-11:
Thank you Keir. But it's not quite 'out of the box' - please add 'remove JA jumper' message for Roland devices. Can I forget about doing the upd upgrade now?

roses2uk, 2019-06-11:
And you said it supports folders, but with the the orig 7-segment display, that's not feasible I guess?

roses2uk, 2019-06-11:
Just to confirm: Type 1 sounds (patches) and Type 2 sounds&song disks with the extension *.W30 are not recognised. But the simple workaround is to rename them with the extension *.OUT (you can keep the original name in front of the extension).

keirf, 2019-06-11:
What jumpers do you have on the gotek now? I will document that.

Are disk changes (simply select a new image with gotek buttons) working OK?

roses2uk, 2019-06-11:
Only the S0 jumper. Yes, disk changes are working OK (but different to old Gotek firmware). Just press the up or down buttons to get to the disk number you want, FF loads it automatically, no need to unplug-reinsert USB stick (hurray!!!). But FF does not retain the last used image in its memory, so at the end of the session you need to go back to 000 (or wherever your system disk is), so that the W30 boots automatically.

roses2uk, 2019-06-11:
Anyway, my SCSI2SD v5.1 arrives soon, so it'll be interesting how it interacts with the new modded FDE.

roses2uk, 2019-06-11:
My sincere thanks to you.....

keirf, 2019-06-11:
You mean FF does remember last image, whereas you always want image 0 at power on?

roses2uk, 2019-06-11:
Yes, if I used image 001 before switching off, at the start of the next session it's still there, and I need to manually set it to 000 to boot the system disk. It's no biggie. But I wish I hadn't mentioned it now, as it's just the same as the old Gotek firmware, when I think about it.

keirf, 2019-06-11:
Create an FF.CFG file in root of USB stick containing the line `image-on-startup=init`

Or just press both buttons to immediately jump to image 0.

roses2uk, 2019-06-11:
"Or just press both buttons to immediately jump to image 0". Oh, that's a cool trick - that's what I'll do. Don't wanna start messing about creating CFG files.........

roses2uk, 2019-06-11:
I wrote this 'How to...' PDF file mainly for myself, but if you think others can benefit from it, feel free to put a link for it somewhere on your Wiki. If you have any comments, additions, deletions, constructive criticism of its content, just chime in.

[How to flash the firmware of a factory Gotek with FlashFloppy.pdf](https://github.com/keirf/FlashFloppy/files/3278122/How.to.flash.the.firmware.of.a.factory.Gotek.with.FlashFloppy.pdf)

keirf, 2019-06-11:
Removing JA is not Roland specific. Most Goteks do not ship with that jumper in position but I guess the odd one does. Or it may depend on the specific Gotek model number (there are many). But if its in position FF will not work properly, regardless of host (Roland or whatever).

roses2uk, 2019-06-11:
Yes, I think you're right, that my specific Gotek model ships with that jumper pre-installed. So I amended the PDF accordingly...

[how to flash the firmware of a factory Gotek with FlashFloppy v2.pdf](https://github.com/keirf/FlashFloppy/files/3278416/how.to.flash.the.firmware.of.a.factory.Gotek.with.FlashFloppy.v2.pdf)

keirf, 2019-06-12:
Added comment on S0 jumper only in Roland wiki instructions. Removing jumper JA was already included in the Troubleshooting FAQ.

closed 2019-06-12 by keirf

roses2uk, 2019-06-12:
Great - I saw that.

Also noticed that you had already documented the E34 message (no valid files on stick) on the Errors page - sorry to have wasted your time on that one.

Regarding E35 - path too deep. How many nested folders are permissible? I assume it's impossible to use folders if I only have the original 3-character 7-segment display??

keirf, 2019-06-13:
Yes folders are supported only with an LCD or OLED display. Folders are supported up to 20 levels deep.

roses2uk, 2019-06-13:
Super-duper stormtrooper!! My OLED screen should be arriving soon, and I think I'm confident enough to do that, even though I still don't have a soldering iron yet. (As for the rotary encoder,  we'll see, as that might be a technical bridge too far...). One question about your diagram for connecting the screen: the VCC & SCL go on the bottom-most pins closest to the PCB (L & R if looking from front of Gotek), while the GND & SDA go above them, L & R respectively?

keirf, 2019-06-13:
OLED display does not need a soldering iron, though you probably want to modify the case a bit and glue the display in place. Easy enough really.

Yes that is the correct pin arrangement.

roses2uk, 2019-07-06:
My OLED screen arrived today, but the 4-header pins were supplied separately so had to borrow a friend's soldering iron (4mm wide blade tip and 1.5mm diameter soldering wire - both not ideal I know). After a bit of hassle, got the screen working and made the hole in the case a bit bigger to accommodate the full width. But I decided to go one step further and try and fit the rotary encoder too, and that's when the problems started. The screen has now stopped working (although the synth still boots as it finds the system disk at position 000. So trying to sort that out, and will let you know. FYI only, no further action on your part required.