keirf/flashfloppy #240: Ensoniq EPS classic "Disk Drive not ready" via Gotek FF V2.13

zaxxon-hh, 2019-06-15:
Hi there, first of all a nice Project !

But i have a problem :-(

interface = shugart  host = ensoniq pin02 = nchg pin34 = rdy

I could boot the OS automatically. So far so good.  The first Instrument loads without problem. 

On second i could load the Name by pressing the load button after pressing enter-yes "disk drive not ready" :-(  where i normal should get "pick instrument button"

If i press the rotary knob i can eject the Image and once again it is mounted again...

Now it loads like ist should. All files are converted to HFE V.2

I have played around with pin 2 and 34 values nothing helps....

Any solution which that could bring me closer to a real Floppy ???

keirf, 2019-06-15:
Are those configuration options each on their own line in FF.CFG?
Did you try `pin02 = auto`?

keirf, 2019-06-15:
You shouldn't need to convert to HFE by the way. The `host = ensoniq` option is so that raw sector files get correctly detected to be Ensoniq disk format.

zaxxon-hh, 2019-06-15:
wow you are fast :-)

pin2 auto same result

zaxxon-hh, 2019-06-15:
## FF.CFG: Example FlashFloppy Configuration File

# Place in the root folder or FF/ subfolder of your USB drive.
# Uncommented lines below are the default settings.
# Uncommented options cannot be overridden by settings in other config files.

##
## DRIVE EMULATION

# Floppy-drive interface mode
# shugart: P2=DSKCHG, P34=RDY
# ibmpc: P2=unused, P34=DSKCHG
# ibmpc-hdout: P2=HD_OUT, P34=DSKCHG (not generally needed: prefer 'ibmpc')
# akai-s950: P2=HD_OUT, P34=RDY (Akai S950)
# amiga: P2=DSKCHG, P34=DRIVE_ID (not generally needed: prefer 'shugart')
# jc: JC closed: ibmpc, JC open: shugart
interface = shugart

# Host platform: Improves image-format detection for generic types such as IMG
# acorn: Acorn ADFS
# akai: Akai (S01, S20, S950), Korg, SC Prophet 3000
# casio: Casio (FZ-1)
# dec: DEC (RX33, RX50)
# ensoniq: Ensoniq (ASR, TS, etc)
# fluke: Fluke 9100
# gem: General Music (S2, S3)
# kaypro: Kaypro
# memotech: Memotech
# msx: MSX
# nascom: Nascom
# pc98: NEC PC-98
# pc-dos: PC DOS Format (geometry determined from Bios Parameter Block)
# tandy-coco: Tandy Color Computer (CoCo)
# ti99: TI-99/4A
# uknc: UKNC / DVK Soviet PDP-11
# unspecified: Common default geometries (including IBM PC)
host = ensoniq

# Pins 2 & 34 output (drive->host) manual configuration
# auto: Auto-configure from interface= setting
# nc: Unused/No Connection [eg. if pin is an input (host->drive) on your host]
# low: Always 0 volts (0v)
# high: Always 5 volts (5v)
# rdy: Drive ready (Ready = 0v)
# nrdy: Logical complement of above
# dens: Density mode (High Density = 0v)
# ndens: Logical complement of above
# chg: Disk changed (Changed = 0v)
# nchg: Logical complement of above
# Values: auto, nc, low, high, rdy, nrdy, dens, ndens, chg, nchg
pin02 = auto
pin34 = rdy

config just now from the usb stick copied...

keirf, 2019-06-15:
No difference in behaviour at all when pin02 setting is changed?

keirf, 2019-06-15:
That FF.CFG is hard to read. You actually only need the three setting lines in your config all the comments could be removed

keirf, 2019-06-15:
Also worth a try playing with jumpers. For example try adding an extra jumper at MO. Remove it if it doesn't help

zaxxon-hh, 2019-06-15:
i know the EPS is a diva :-) i have built a Sony MFP 520-1 real Floppy to get work with my EPS...

No , sorry to say noting has changed in behavior

playing around with pin 2 and 34 i remeber that i have had some like "Disk has changed...." message

zaxxon-hh, 2019-06-15:
MO Jumper i have tried and the Hardware mod i have tried PB15 1k Ohm as you pinned in your Hardware mods

zaxxon-hh, 2019-06-15:
i would put point on the pin 34 if i am right the rdy goes to 0V and after it goes stable to 5 V on real Floppy drive....
i am not sure on that but that could be a hint

keirf, 2019-06-15:
If you connected motor to pb15 did you also try `motor-delay` in FF.CFG?

Avoltmeter on pins 2 and 34 of a real working drive may be useful however bear in mind these pins also react to other pins such as select motor and step and thus results can be confusing or misleading. Logic analyser is better but not many people have them.

zaxxon-hh, 2019-06-15:
motor-delay is set to 200

zaxxon-hh, 2019-06-15:
ok sounds like there are no other EPS User out there :-(

keirf, 2019-06-15:
Low then high on pin 34 would sound a bit like disk change. In which case interface=ibmpc or ibmpc-hdout worth trying

keirf, 2019-06-15:
Ask on the Facebook group. Not many other than me look here, it's my bug tracker not a support forum. Eg Chris Poacher and others on that group have broad synth experience.

zaxxon-hh, 2019-06-15:
ibmpc-hdout 
pin02 = nchg
pin34 = rdy

same result "diskdrive not ready"

zaxxon-hh, 2019-06-15:
ok i will grab my Voltmeter....

keirf, 2019-06-15:
pin02 and pin34 settings are overriding the new interface setting. Please set them both to `auto`.

zaxxon-hh, 2019-06-15:
i have the original Floppy drive which i will use for measurment on pin 2 and 34 

stay tuned....

zaxxon-hh, 2019-06-15:
ok first result: 5 V stable on pin 34 . If its reading the directory it goes low 0 V , after stable 5 V . 

if i load the instrument it goes from 5V to low 0V and after the file is in memory it goes back to 5 V !

that looks like that the gotek FF does not that behavior, i am right on that ?

FF does not emulate that right ?

zaxxon-hh, 2019-06-15:
i am not on FB

keirf, 2019-06-15:
That is likely correct behaviour for READY signal. It goes to 0v only when drive is selected and motor is enabled. If you test on FlashFloppy with `motor-delay=200` in FF.CFG, you should see the same (interface=shugart, pin02=auto, pin34=auto). This is why checking one signal at a time gets confusing.

keirf, 2019-06-15:
It all sounds like classic Shugart really, so the problems you see are perplexing.

keirf, 2019-06-15:
Taking a longer view to solving this, get a Saleae logic analyser on Ebay (clones available for cheap) and some grabber probes, and get a trace of real drive on pins 2, 10, 16, 20, 34 (at a minimum). Not much outlay, pretty useful thing to have. Then you can see correct relationship between signals.

zaxxon-hh, 2019-06-15:
ok, if i automount the mage after 2 seconds and that does the same as manually  i don not know whats wrong. 
on the manually way eject and mount again and it work it is not complicated i would say....
that could be a software issue i would like to say...

keirf, 2019-06-15:
Why automount anyway. If you have the third button on your Gotek then disable auto select in FF.CFG

zaxxon-hh, 2019-06-15:
it is more comfortable i would say, by the way that should not be a difference for me on the interface....

but i am not sure on that .... that is something you can tell me...

keirf, 2019-06-15:
Shouldn't make any difference unless host has a minimum time between disk swaps below which it doesn't recognise the disk change? I'm sure there's no difference on the firmware side

zaxxon-hh, 2019-06-15:
mhm, 

autoselect-file-secs = 0

but i does grab the image...

any idea why ?

keirf, 2019-06-15:
Still autoselected after two seconds? Then the config line or even perhaps you'd whole config is not accepted for some reason.

zaxxon-hh, 2019-06-15:
right after 1 second...

keirf, 2019-06-15:
I will confirm the config option behaviour myself later.

zaxxon-hh, 2019-06-15:
i will try a new FF.CFG

keirf, 2019-06-15:
Confirmed `autoselect-file-secs = 0` works correctly in v2.13 (ie disables autoselect).

You need to get that sorted as obviously none of your other config settings may be working either. In which case all your config attempts this far have been time wasted. This is a config option with a nice obvious effect for you to get fixed.

zaxxon-hh, 2019-06-15:
OK! got it sorted Confirmed autoselect-file-secs = 0 works correctly now

Fault is the same ! after load the O.S. rotary pressing eject ok step over to the next instrument pressing rotary it mount and load... 

eject on rotary step over to the next instrument mount "disk drive not ready" again eject after mount and it loads....

zaxxon-hh, 2019-06-15:
Pin2 behave like Pin 34 no access 5V stable if reading the directory or loading the instrument it goes on low 0V 

is that what the FF output pin 2  is doing ?

keirf, 2019-06-15:
What size are the eps disk images?

zaxxon-hh, 2019-06-15:
800 kb

keirf, 2019-06-15:
Are we back to interface=shugart pin02=nchg pin34=rdy?

zaxxon-hh, 2019-06-15:
if i set pin 2 on 5 V via FF.CFG stable it just boot the OS an the EPS don not read again....nothing happens to disk

in fact every disk change will generate in real floppy a 5V to 0V change. 

that let the EPS read the Floppy again...

if that does not happen that could give an error like "disk drive not ready" i would say 

correct me if i am wrong on that...

zaxxon-hh, 2019-06-15:
yes that is what actual use 

interface=shugart pin02=nchg pin34=rdy

but i see pin 2 does not change like the real floppy does....

keirf, 2019-06-15:
Try `pin02=low`

zaxxon-hh, 2019-06-15:
btw the Pin 34 looks like as you told me,(shugart compliant) so i would say it is ok

keirf, 2019-06-15:
Going to 0v on disk eject would be `pin02=chg` behaviour but then it will go back to 5v on next head step (pulse on pin 20). It could be a weird disk-change signal on pin 2 but I see no documentation of that online for EPS

zaxxon-hh, 2019-06-15:
ok tested Pin2 = low just give me always an "disk drive not ready "

zaxxon-hh, 2019-06-15:
ok sorry the OS boots up forgotten to say

zaxxon-hh, 2019-06-15:
i will check that ! if PIN2 goes back to 5 V i do not think so...

i will use PIN2 = auto so the interface should do the work without using the FF.CFG

right ?

keirf, 2019-06-15:
That is right pin02=auto means the pin behaviour is driven by the interface setting. Which for shugart is same as pin02=chg

zaxxon-hh, 2019-06-15:
ok my result: OS boots correct 
PIN 2 = auto
Pin 2 holds 5V stable if i eject or not but it stays on 5V on reading the disk in message with "disk drive not ready" 

in my opinion that is not EPS compliant....

keirf, 2019-06-15:
Yes this is why a logic analyser trace really is needed, so we can see relationship between the signals. I could actually characterise EPS pin 2 behaviour then.

zaxxon-hh, 2019-06-15:
i have an service manual for the EPS but to get information about is horrible. 

you are right documentation is a mess about that...

keirf, 2019-06-15:
Service manual may or may not help. It can be as brief as "replace drive with such and such model".

zaxxon-hh, 2019-06-15:
haha right 

but why is that so weird ? i have an Sony Floppy moded to work in it like hell....

zaxxon-hh, 2019-06-15:
the only reason i need to change to an Gotek is the old disks are no more readable due age 

i have about 70% that are just damaged now :-(

keirf, 2019-06-15:
For example some drives have a disk change reset signal from host to drive at pin 1. Then pin 2 would not transition from 0 back to 5v until pin 1 is asserted. Its rare though. Um and I think some other drives have disk change which ignores whether the drive is selected or not (ie that pin is not gated). Again, rare. But if nothing else is working, and you believe the host and cable is OK (eg they work with real drive)...

keirf, 2019-06-15:
If I could get a real working drive I can work out what's going on. But I figure you are mainland Europe not UK?

zaxxon-hh, 2019-06-15:
i am in Germany yes

keirf, 2019-06-15:
I suggest get a logic analyser and trace the drive. Or send it to me and I will do so. Perhaps EPS needs further support in the firmware.

zaxxon-hh, 2019-06-15:
is it worth to help other user if i send you an real drive ?

or iam the only one who try to get it run ?

keirf, 2019-06-15:
I checked with my go to guy and he tells me he and a friend of his have no luck either. So I guess the wiki is wrong and this is a more general unsolved problem.

[50 more replies]

closed 2019-07-08 by keirf